MarlinFirmware/Marlin #15182: [BUG] I want to lift steps/mm  to 4000 over LCD controller, but is max about 3200 and going to 5 back.

Milu1980, 2019-09-06:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Milu1980, 2019-09-06:
with pc connection is possible, but i need it in the LCD controller...

gloomyandy, 2019-09-06:
How recent a version of Marlin are you using? In theory this was fix a short while ago...
https://github.com/MarlinFirmware/Marlin/issues/14821

Milu1980, 2019-09-06:
Hi,

i use malin 2.0 skr 1.3 board

Milu1980, 2019-09-06:
I tryed with pc i set 4000 its done, but on the lcd i see 3000.
i dont know now, that 3 or 4000 steps/mm i have :(

AnHardt, 2019-09-06:
The fix is no fix. INT = INT32. limit was and is 3276.7 .

AnHardt, 2019-09-06:
For now ,set with M92. Check with M503.

Milu1980, 2019-09-06:
hi, yes about 3200...
can i change it in marlin somewhere ?

gloomyandy, 2019-09-06:
@AnHardt So I guess that @thinkyhead proposed fix of making the menu only allow int values (https://github.com/MarlinFirmware/Marlin/issues/14821#issuecomment-518049811) did not happen?

AnHardt, 2019-09-06:
#14909

gloomyandy, 2019-09-07:
Though I'm always a little suspicious when I see folks using non integral steps/mm, often means something else is wrong with the mechanics of the machine (I suppose if you have very low steps/mm it may make sense).

Milu1980, 2019-09-07:
hi, because i have translate inside. 3d printer, laser cutter and milling

AnHardt, 2019-09-07:
It only needs imperial sized lead-screws to need fractions.

Milu1980, 2019-09-07:
i have just ball screws.

Patag, 2019-09-07:
@AnHardt AFAIK, extruder step/unit can't be an integer (unless rounding) , PI being used in the calculation

boelle, 2019-09-24:
@Milu1980 still having problems?

Milu1980, 2019-09-24:
yes...

boelle, 2019-09-24:
have you tried with latest 2.0.x ?

Milu1980, 2019-09-24:
no, i will try on the weekend...

Milu1980, 2019-09-24:
can you please tell me what was changed ? i have too much setting in my
version to change it

boelle, 2019-09-24:
no i just update my copy of marlin and copy over old settings when the 2 settings files change layout

you can use a tool like winmerge to compare files and copy your settings

boelle, 2019-10-21:
@Milu1980 did you try the latest 2.0.x ?

thinkyhead, 2019-10-22:
This should fix the issue… f00b48f9a8da77d0d603b74c843359b82ee306e3

closed 2019-10-22 by thinkyhead