MarlinFirmware/Marlin #15183: [BUG] 3DTouch doesn't working.

dav63300, 2019-09-06:
Hi,

I'm trying to build a new printer. I have a re-arm 32bits cards with a ramps 1.6.
I want to setup a 3dtouch.
I plugged the white and black cable in z_min pin, and the other three wires on the first servo plug (the nearest from reset button) (d11 I think)

When I power on the printer, the 3dtouch deploys and stows 2 times, then the LCD blinks a bit, and the 3d touch is blinking red.

If I try a reset, the 3d touch stop blinking.
I can deploy it.
But if I click Stow, it stows, and fast in the same time the LCD blinks, and the 3d touch blinks red too ....

I plugged the printer to a laptop, open up Repetier host.
A M119 command says open (when pin is deploy) and triggered when pin is stow (and blinking)

In the firmware, I put a 750ms delay for bltouch.

Can someone help me to make it work ? I spent almost 5 hours making changes in the firmware, upload new congifuration but nothing work

Thank you very much for your help.

shitcreek, 2019-09-06:
You need to follow the template and provide your configuration files for us to know what's wrong.
Did you follow any tutorial online? Are you sure this is a bug?

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

pjanco, 2019-09-06:
Hi,
it looks like you have a hardware problem and not bug in firmware. Because of the blinking of LCD backlight, maybe your 5V power is s not strong enough or you have weak connection on cable.

I was trying to add 3Dtouch to my Ender-3 just few days ago. I was using +5V and GND pins from ISP header for supplying 3D touch. After power up the printer, 3Dtouch stow/deploy only once and then 3Dtouch start to blink red (it mean self-test failure).

My solution was to solder wires for powering 3Dtouch to another pinheader closer to power supply. Now it is working great!

pjanco, 2019-09-06:
That is weird!
I moved from Bugfix-1.1 to Bugfix-2.2 right now and my 3DTouch is not working anymore. It is going to alarm state after reset of printer.

I tried to disconnect ribbon cable to LCD display to reduce power consumption from 5V power supply, it did not help.

I tried to disconnect SERVO0 signal from PCB board. GND and +5V is still connected to the board, and this help. Self-test is sucessfull after printer is turned ON.

I tried to connect SERV0 signal back to the PCB and use separate external 5V power supply. I apply then external power supply first. 3Dtouch turn on without a problem, self-set is successful. Then I turn ON rest of the printer and 3D is doing self-test again, probably is restarted.

So, definitely there is some software problem but maybe it i different problem then yours.

I have two theories about this problem.
1.
When Bugfix-2.0 booted, hot end FAN is turned ON for 1 second. Then  it is turned OFF.
When Bugfix-1.0 booted, hot end FAN is OFF all the time.
Maybe there is some interference from the FAN.
I have on both configurations activated `#define PROBING_FANS_OFF` function.

2.
Newer firmware is trying to do some "smart" initialization for BLTOUCH 3.1 after reset. This commands should be ignored by older versions of BLTOUCH. But maybe this this cheaper chineese clone have problem with "smart" commands.

There is my configurations form both firmwares:
[Configuration2.zip](https://github.com/MarlinFirmware/Marlin/files/3585967/Configuration2.zip)
[Configuration1.zip](https://github.com/MarlinFirmware/Marlin/files/3585968/Configuration1.zip)

dav63300, 2019-09-07:
Thank you.
It was a 5v power supply fault !!! Ramps 1.6 sucks with 5v...
I connect black and white cable from 3dtouch to z_min pin. Orange on d11 pin, and power 3dtouch with an external 5v power supply.
It works perfectly.
Once again thank you for your help.

I'm sorry, if the post must be deleted I can do it. I posted it here, and it's not the right place.

shitcreek, 2019-09-07:
@dav63300 good to hear. You can close the issue yourself next time.

@pjanco could you start another issue once you've determined it to be a bug. Thanks.

closed 2019-09-07 by shitcreek