MarlinFirmware/Marlin #15184: [BUG] BLTOUCH Doesn't Reset Alarm and Keeps Rising After Failed Stow Move

shahmir2797, 2019-09-07:
Hi everyone, I'm running an ender 3 with latest marlin 1.1.x-bugfix and I'm encountering an issue with my BLTouch v3 (I've removed the interfering capacitor from the mainboard). I've had this printer running reliably for many months, but since moving from the TH3D firmware to this, I've noticed some issues.

I usually run my printer with a spray on adhesive against the glass bed. This leaves a very sticky coating. Sometimes the tip of the bltouch gets stuck to this during the stow move part of the probe. This sends the bltouch blinking red and it tries to continue the G29 command but the printer head just keeps on rising higher and higher at every probe point. Never actually going down to touch the bed again.

When I don't have the glue on the bed, the bltouch works like a dream, but my bed adhesion greatly suffers. When I was running the TH3D firmware I noticed that when this alarm happens the printer head just moves up and lets the bltouch reset itself and then continue moving on to probe the rest of the bed. 

I've been trying to poke around the firmware files to pick apart the code and see if i can insert something like a _bltouch_reset(); somewhere but I'm extremely new to programming. 

If I'm just confused and this isn't a bug, please guide me to where I can find a way to fix my BLtouch's alarm not resetting. Thank you

shitcreek, 2019-09-09:
As you've described, this is a physical problem and not a bug. The BLTouch stows the probe with a small magnet. It can't function if it's overcomed by the adhesive.

As to where to insert your reset function, look inside `bltouch.cpp`. If you have `DEBUG_LEVELING_FEATURE` enabled, you'll get more descriptive readouts of the processes.

Have you tried just scuffing up your glass surface or use masking tape? Or get a better build surface like BuildTak or EZMat? https://www.matterhackers.com/news/choosing-the-right-3d-print-bed-surface

blueeagle69, 2019-09-13:
I had this problem once with my Bltouch 3.1, but thankfully knew what it was.
As I am using PVA on glass, I simply preheat the bed and wait for the glue to dry before probing.
Problem solved.

boelle, 2019-09-24:
will close this as its not a bug

closed 2019-09-24 by boelle