MarlinFirmware/Marlin #15185: [BUG] Firmware Retract - crash when changing value via LCD

Chr157i4n, 2019-09-07:
### Description

Today i activated the firmware retract.
When i change any value via the LCD-menu, i can choose a new value. But when i press the wheel to accept the value, the printer doesn't respond and it restarts after a couple of seconds.

On the other hand changing those values via PC with M207 and M208 works fine. 
Also the right values are displayed in the LCD-menu.

### Steps to Reproduce

1. go into the Configuration/Retract menu
2. change any value
3. press button to accept the new value

**Expected behavior:**
Changed value

**Actual behavior:** 
Crash

#### Additional Information

SKR Pro with Marlin Bugfix 2.0.x
[Marlin-Configfiles.zip](https://github.com/MarlinFirmware/Marlin/files/3586137/Marlin-Configfiles.zip)

sjasonsmith, 2019-09-21:
I just reproduced this behavior. I happened to have an SKR Pro with TMC2208 drivers sitting around, which matches your config. It took a bit to get a screen working with it, but I'll see if I can figure out what's going wrong.

Chr157i4n, 2019-09-21:
Thanks for looking into this issue.
I also have the Issue, that mixing with three extruders (same setup) doesnt work.
Could that be the same reason?

sjasonsmith, 2019-09-21:
This crash is related to unaligned access to data, caused by the `pragma pack` around `fwretract_settings_t`, in fwretract.h.

```
#pragma pack(push, 1) // No padding between fields
typedef struct {
  float retract_length,                     // M207 S - G10 Retract length
        retract_feedrate_mm_s,              // M207 F - G10 Retract feedrate
        retract_zraise,                     // M207 Z - G10 Retract hop size
        retract_recover_extra,              // M208 S - G11 Recover length
        retract_recover_feedrate_mm_s,      // M208 F - G11 Recover feedrate
        swap_retract_length,                // M207 W - G10 Swap Retract length
        swap_retract_recover_extra,         // M208 W - G11 Swap Recover length
        swap_retract_recover_feedrate_mm_s; // M208 R - G11 Swap Recover feedrate
} fwretract_settings_t;
#pragma pack(pop)
```

If you remove the pragmas, then the menu works fine. I suspect that it worked through the M207/M208 commands, because the compiler recognized the misalignment and added extra instructions to account for it. In the menu it is used through pointers which prevent the compiler from recognizing there is an issue.

In my tests removing the pragma increased data usage by 4-8 bytes, but decreased program size by 56 bytes. This is a sure sign that the compiler was generating a lot of extra instructions to access the unaligned data.

sjasonsmith, 2019-09-21:
> I also have the Issue, that mixing with three extruders (same setup) doesnt work.
I checked the code for other instances of pragmas like this, and there is only one other. It is in the EEPROM code, where I suspect it probably isn't causing issues.

I couldn't even compile with your three extruders, that's why I posted PR #15329.
You might want to start on one of the support forums for the mixing issues, until you know there is an actual issue to report.

sjasonsmith, 2019-09-25:
@Chr157i4n, the two merged PRs references above should fix your crashing. Could you try using the latest bugfix-2.0.x and close this issue if it is fixed?

Chr157i4n, 2019-09-25:
i cant get the latest bugfix-2.0.x to compile without an error.
But i tested your changes in my current build and it fixed the problem.

closed 2019-09-25 by Chr157i4n