MarlinFirmware/Marlin #15186: [BUG] Z homing not working properly

StKob, 2019-09-07:
After not using my printer for longer time I started to observe that it no longer home z axis properly. After that I updated marlin to newest on bugfix 2.0 but still I am seeing this problem. I'm using inductive probe. It looks like either something is triggering z endstop (I should see light on probe stop litting) or some king of bug.  

Hardware: 
Clone of Prusa i3 Mk2
DUE3DOM LE
Arduino Due

Configuration: https://pastebin.com/xZjKJSa6
Configuration_adv: https://pastebin.com/FAsmnd6g

shitcreek, 2019-09-09:
If Z wasn't homing properly all on its own, then wouldn't that suggest that it is a hardware failure? 

Please be certain it is indeed a bug before posting here.
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

boelle, 2019-09-24:
@StKob still having problems?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle