MarlinFirmware/Marlin #15198: BLtouch (clone) is probing in air

Ke5han, 2019-09-08:
I installed a bltouch on my delta, and I can run auto calibration, after that if I try to run UBL bed leveling my effector will go all the way down till the bltouch is triggered, but this only works for one or two probe points, and then the effector will rise to the height that the bltouch pin just touch the surface but won't trigger the pin.

I think somewhere in my config my calculation logic is wrong or it's a bug? because it does hit and trigger once or twice.

Thanks.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3587658/Marlin.zip)

boelle, 2019-10-12:
@Ke5han tried what @tvixen suggested?

boelle, 2019-10-21:
will close this one as it seems that @Ke5han dont have an issue anymore

we can always reopen it

closed 2019-10-21 by boelle