MarlinFirmware/Marlin #15199: [BUG] SKR 1.3 Fans not runing while printing

ArketypeDesign, 2019-09-08:
### Description

Is it a bug or is it me?
I am pretty much a Noob at this, so bear with me.
Any help would be appreciated.

Just upgraded my custom Core X-Y from a MKS GenL to a 32 bit Bigtree SKR 1.3 board.
(slightly modified V-King design https://vkingprinter.com/ ) 

I have full control of the fans (E0 and Part cooling) from the LCD control menus.
Everything seems normal, and works properly. (auto homing, etc. all work and fans work as expected)

But when I print a file from the SD card the fan behavior is not as expected.
E0 fan does not appear to run, allowing the extruder to overheat and clog.
The part cooling fan also does not run.

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3587931/Config.zip)
[image]

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

reloxx13, 2019-09-09:
post the latest gcode file u tried to print and had not fans.

Promite2501, 2019-09-10:
I'm not sure if your fan pin is configured correctly. mine works fine.

For the hot end heat sink cooling fan
In Configuration_adv.h you have ` #define E0_AUTO_FAN_PIN 7`
try 
` #define E0_AUTO_FAN_PIN -1`  (if plugged into the normal fan header I think)
or 
`#define E0_AUTO_FAN_PIN FAN1_PIN // -1`  should come on at 50c if you have it plunged into E1 heater.

ArketypeDesign, 2019-09-10:
Attached is a test cube... This printed fine with the MKS board.

[20mmTestCube-PLA.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/3595520/20mmTestCube-PLA.gcode.zip)

ArketypeDesign, 2019-09-10:
> I'm not sure if your fan pin is configured correctly. mine works fine.
>
> For the hot end heat sink cooling fan
> In Configuration_adv.h you have ` #define E0_AUTO_FAN_PIN 7`
> try
> ` #define [...]

Thanks I will look into this.

shitcreek, 2019-09-10:
Just to clarify, pin 7 (P2_07) is for hotend 0 (heater for E0). Pin 4 (P2_04) is the for Fan0 (PWM controllable fan) for the BigTreeTech SKR v1.3 - see `pins_BIGTREE_SKR_V1.3.h` for all the pinouts.
`E0_AUTO_FAN_PIN -1` should be kept at -1 to use the default pin.

tvixen, 2019-09-17:
Hi Mate,
I will suggest you to compile a newer version of Marlin, as the one that you got is **really** old.
I have the SKR 1.3 and it's running fine with the same fans as you.
Especially have a look at this line:
#define E0_AUTO_FAN_PIN P2_04   (not 7 as you have)

brasco901, 2019-09-17:
#define E0_AUTO_FAN_PIN P2_04 this fixed my fan issues

ArketypeDesign, 2019-09-17:
Thanks everyone for your help.
I have re-downloaded the latest 2.0 marlin and plan on testing the pin settings as discussed above very soon.
I am limping along right now with a hardwired manual on-off fan.  ; )

brasco901, 2019-09-18:
@ArketypeDesign hope everything works. Changing the fan pins definitely worked for me.  When you update the firmware you will no longer see fan 1 and fan 2 in temperature.  You'll just see fan. 👍🏾

DerAndere1, 2019-09-19:
Thanks for the helpful comments all. As this is an often reported issue, it would be great if someone could attach (drag and drop) his working config files and describe concisely which physical pins the part cooling- and extruder fans were connected to, so we have one reference.

ArketypeDesign, 2019-09-23:
Found the problem.
After a full day of troubleshooting and rewiring....

The default Pins file for the SKR 1.3 says
#if HOTENDS == 1
  #define FAN1_PIN         P2_04

But the Configuration.h file says
#define EXTRUDERS 1

So the  system was never recognizing that there was 1 extruder.

Changed the Pins file line 169 to read
#if EXTRUDERS == 1
  #define FAN1_PIN         P2_04

Now it seems to work.

ArketypeDesign, 2019-09-23:
How does one post this as a "bug" to be fixed? (sorry I am a github noob)

samnangor, 2019-09-24:
I have tried all day, it is still not working.
I can send the command to run the fan, but it will not turn on automatically.
Please help

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3644760/Configuration_adv.zip)

Promite2501, 2019-09-24:
samnangor, as out lined about, for auto fan you need to change in Configuration_adv.h

`#define E0_AUTO_FAN_PIN FAN1_PIN // -1
`

samnangor, 2019-09-24:
change to
#define E0_AUTO_FAN_PIN FAN1_PIN // -1
in Configuration_adv.h
and set nozzle heat to 70, temperature climbed to 90 still no fan.

ArketypeDesign, 2019-09-24:
samnangor, I had the same problem.
See my note above about the PINS file not using the correct code to recognize the number of extruders. This will prevent the fans from working properly. Once I found this and corrected it, things started working for me.

samnangor, 2019-09-24:
Thanks ArketyprDesign.
I think I did Change it. I will check again tomorrow, it is a little late for my bed time now.
Thanks YOU ALL, I will let you know tomorrow.

samnangor, 2019-09-25:
i did change 
#if EXTRUDERS == 1
  #define FAN1_PIN         P2_04
still no help. It will turn on and off manually when I send Gcode command M106 
and M107 .

samnangor, 2019-09-25:
Change back to

#if HOTENDS == 1
#define E0_AUTO_FAN_PIN FAN_PIN // -1

still have the same problem 
but here what it does

Gcode command M106 and M107 will turn on and off the fan duct.
Gcode command M106 P1 and M107 P1 will turn on and off the heater fan

Promite2501, 2019-09-26:
The fan needs to be connected to the E1 heater not the cnc fan, if using `#define E0_AUTO_FAN_PIN FAN1_PIN `
[image]

samnangor, 2019-09-26:
Found it
#define E0_AUTO_FAN_PIN P2_04
in Configuration_adv.h

boelle, 2019-10-12:
@ArketypeDesign is issue still there?

ArketypeDesign, 2019-10-12:
Seems to be working after updating to the latest Marlin.
Thnaks.

boelle, 2019-10-12:
oki, will close then

we can always reopen

closed 2019-10-12 by boelle

icke86, 2019-11-29:
Hi,
my fan only turns on when i start it via display. an m106 command only starts the partcooling fan but not the hotend fan. 
i cant find the P2_04 in my pins file...
i tried to follow the instruction from TeachingTech but all the new Marlin versions that i can find don´t have the part where it says "Heaters / Fans" and i cant find a result for a search either .
I dont have any experience with marlin or with coding so i dont know what i have to look for or what i am missing. 
thanks in advance

DerAndere1, 2019-12-02:
@icke86: Before discussing in this issue queue, please read [this info](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question). You will find several occurences of "fan" or "FAN" in your configuration.h or configuration_adv.h files. In the [official Marlin documentation](http://marlinfw.org/docs/configuration/configuration.html) it is most efficient to use the browser's search function (Strg + F in Mozilla Firefox). The relevant pin files are [pins_BTT_SKR_V1_3.h](https://github.com/MarlinFirmware/Marlin/blob/9ea75dca3eabca8d6a9c3d88a5f564d92d7a9c77/Marlin/src/pins/lpc1768/pins_BTT_SKR_V1_3.h#L312 ) and especially [pins_BTT_SKR.h](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/pins/lpc1768/pins_BTT_SKR.h).

icke86, 2019-12-03:
oh i didnt know that i have to look in the pins_BTT_SKR.h for that... now i found it thanks.
i changed my start g-code and added a m106 p1 command and it did the trick too

MoppelMat, 2020-01-14:
thank you all for this info!
just for future clarification this is all you need (in Configuration.h)

// @section extruder

// This defines the number of extruders
// :[1, 2, 3, 4, 5, 6]
#define EXTRUDERS 1

// This defines the number of hotends to the same value as extruders (for pins_BTT_SKR.h)
#define HOTENDS EXTRUDERS