MarlinFirmware/Marlin #15200: [BUG] Re-ARM mechanical runout sensor default pin not working. also not trigger M600

sl1pkn07, 2019-09-08:
### Description

Connect a mechanical (signal and GND) endstop for use as runout sensor not working if use a default pin set in pins_RAMPS_RE_ARM.h

the default runout pin is set in Servo 3 pin (D4 / P1_18)

according of this comment, https://github.com/MarlinFirmware/Marlin/issues/10231#issuecomment-439280303 Servo3 pin not work for this purpose

change the default pin to Servo2 (D5 / P1_19) do the trick

also i'm not sure why, but the runout sensor works without print if see what happen with M119, but durning print (from SD), the trigger never do

### Steps to Reproduce

1. enable runout sensor
2. setup the hardware
3. load and unload filament by hand
4. see what happen with M119
5. see what happen dunring print when cut the filament

**Expected behavior:** [What you expect to happen]

run M600 automagically

**Actual behavior:** [What actually happens]

nothing happen, the prints continue

when run M119 with filament:
~~~
M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
filament: open
~~~

when run M119 without filament
~~~
M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
filament: TRIGGERED
~~~

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
https://github.com/sl1pkn07/Marlin/commit/d8518f297a97bdd7f9740370a0fe53b9ed0ac629
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sl1pkn07, 2019-09-08:
also i notice the runout not work iself when print without filament.

change the logic do nothing

Squid116, 2019-09-11:
I am seeing exactly the same.

SKR 1.3, mechanical endstop (microswitch with 50nF capacitor) wired to z max.

Test with M119 and have the same results as mentioned above.
With filament not loaded:
`filament: TRIGGERED`

With filament loaded:
`filament: OPEN`

Start a print and it stops immediately (during homing) saying it has triggered.

I'll see if I can track down last working commit over the weekend if no one else works it out.

xeddog, 2019-09-16:
The default pin for SKR V1.3 is P1_28 which is X-max.

thinkyhead, 2019-09-17:
Can everyone confirm that connecting to the X_MAX pin fixes the issue?

thinkyhead, 2019-09-17:
Change your `FIL_RUNOUT_INVERTING` setting if the values are opposite of what you want.

Squid116, 2019-09-17:
Mine is working, following what @thinkyhead said - I swear I had checked that before posting, guess not.

sl1pkn07, 2019-09-17:
Mine no, connect were connect.(Servo3 NOK, Servo2 OK)

detect filament, but do nothing when trigger

xeddog, 2019-09-17:
> 
> 
> Mine no, connect were connect.(Servo3 NOK, Servo2 OK, Z_MAX OK, etc)
> 
> detect filament, but do nothing when trigger
> 
> My mechanical filament endstop is the ender 3 Z,X,Y axis stock endstops
> 
>
I cannot address the servo pin not working, But when I was working with a REARM a month ago or so, I had the same problem.  To work around the problem, I just changed the pin assignment for the runout sensor to "#define FIL_RUNOUT_PIN   P1_25" which is also X-Max.  Since you are running a Creality Ender 3, I would assume that this would be an acceptable solution since X-Max should be available.  I have a Creality CR-10S and it uses x-Max by default, and it came that way from Creality.

sl1pkn07, 2019-09-17:
Hi

is acceptable (not yet tested) but strange, because Servo2 and ZMax is tested. in both detect the filament, but do nothing. Servo3(default) is always trigged/open (depend of `FIL_RUNOUT_INVERTING`) , and also do nothing

just for the record, my tests:

With working tested pins (Servo2 and Zmax)
with `FIL_RUNOUT_INVERTING false`  open with filament, trigged without it in M119
- with filament, start print and extract/cut the filament = do nothing, just "print"
- without filament and start print = do nothing, just "print"

With `FIL_RUNOUT_INVERTING true` trigged with filament, open without it in M119
- with filament, start print and extract/cut the filament = do nothing, just "print"
- without filament and start print = do nothing, just "print"

With not working pin (Servo3) 
with `FIL_RUNOUT_INVERTING false` get always trigged in M119
- with filament, start print and extract/cut the filament = do nothing, just "print"
- without filament and start print = do nothing, just "print"

with `FIL_RUNOUT_INVERTING true` get always open in M119
- with filament, start print and extract/cut the filament = do nothing, just "print"
- without filament and start print = do nothing, just "print"

greetings

xeddog, 2019-09-18:
> 
> 
> Hi
> 
> is acceptable (not yet tested) but strange, because Servo2 and ZMax is tested. in both detect the filament, but do nothing. Servo3(default) is always trigged/open (depend of `FIL_RUNOUT_INVERTING`) , and also do nothing
> 
> just for the record, my tests:
>
This would take one of the developers or someone who knows more about the firmware, but I had some similar test results with the wrong pin assignment.  The M119 results were as they were supposed to be, but the sensor did not work.  I am **assuming** that the M119 was showing the way the hardware was working, but the firmware was looking elsewhere.  But as I said, You would need a Marlin expert for a definitive answer.

bullgun, 2019-09-26:
> Hi
>
> is acceptable (not yet tested) but strange, because Servo2 and ZMax is tested. in both detect the filament, but do nothing. Servo3(default) is always trigged/open (depend of [...]

i have the same problem, do you have solved this issue?

sl1pkn07, 2019-09-26:
Nope in this momment. I'm waiting the fix for the TMC(2130) SPI daisy chain feature.

Greetings

sl1pkn07, 2019-09-30:
ok. tested again. this time, without touch noting in the config except update marlin. the rounout sensor start running!

connected to Servo2 pin, with NC contact in the runout sensor, and `#define FIL_RUNOUT_INVERTING false`

Servo 3 still not working, Z_max or X_max not tested

as said in https://github.com/MarlinFirmware/Marlin/issues/10231#issuecomment-439280303, ReARM pin file need update the default runout pin to Servo2 instead of Servo3

is ok @thinkyhead ?

greetings

boelle, 2019-10-12:
@sl1pkn07 i guess this one is still waiting for a fix?

sl1pkn07, 2019-10-12:
seems working now for me. but i'm waiting to merge #15519

boelle, 2019-10-21:
#15519 has been merged so i will close this one

we can always reopen

closed 2019-10-21 by boelle

salfter, 2019-11-22:
This explains why I was having trouble getting a filament-runout sensor working on the default input a while back...ended up putting it on an unused endstop input and redefining the runout input to look there.