MarlinFirmware/Marlin #15201: [BUG] Re-Arm + TMC2208 in standalone mode. Motors noise.

uniksoid, 2019-09-08:
**Description**
I've upgraded my printer with Re-Arm board using same RAMPS shield from Mega with same TMC2208 drivers installed in standalone mode (not UART). During any movement at a reasonable speed (40-60mm/s) motors produce strange vibrating noise while printhead is moving. Sounds like StealthChop is not working. 
Didn't have this issue using Mega with Marlin 1.1.9 with exactly same set-up (Drivers+Motors). Same configuration for drivers, steps per mm in configuration.h

**Steps to Reproduce**
1. Install TMC2208 on Ramps in legacy mode
2. Set Vref at 0.8V
3. Set drivers to TMC2208_Standalone in Configuration.h

**Expected behavior**
TMC2208 running in StealthChop mode with low noise of motors.

**Actual behavior**
At speed around 40-60mm/s motors are producing noise and vibrating while printing. Wheareas same set-up in Marlin 1.1.9 gives expected result. 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3588044/Configuration.zip)

shitcreek, 2019-09-09:
Looks like you don't have `HYBRID_THRESHOLD` enabled - 40-60mm/s is too fast for StealthChop to operate silently.
Also, I have my vRefs set to:
v0.6 : X & Y
v1.0 : Z
v0.9 : E

Please determine that this is indeed a bug before submitting a report.

We're not a support group:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

uniksoid, 2019-09-09:
Dear shitcreek,

I've spent almost a week thrying to find a solution checking all avaliable resources (Mentioned above but not limited to) to find solution. And never had an intention to find support here. 
Hybrid_threshold setting doesn't change much which seems reasonable as the drivers are in legacy mode, but not connected via UART.

Checking vRef was the first thing I did. I even tried a different couple of motors (1.8 degree for a change). Surprisingly on Mega vRef around 0.8V works perfectly. But in order to get to the same noise level in Marlin 2.0 I had to raise vRef to 1.1V, which is not the best solution.

Seems like a bug to me.

shitcreek, 2019-09-10:
You are right, TMC2208 doesn't have spreadCycle in standalone mode.
These TMC drivers are rated up to 36v, so 1.1v isn't bad, and according to some, it is recommended to run them at 12v. They get really hot regardless if you're running at .7 or 1.1v. 
Alex Kenis has a great video on this subject: https://www.youtube.com/watch?v=GVs2d-TOims

tvixen, 2019-09-11:
TMC2208 in Stealthchop is running fine 100mm/s on 24V, very silent (try giving them some more juice)
Are you still using Arduino 1.8.9 as compiler ?  or have you switched to PlatformIO.

uniksoid, 2019-09-13:
> TMC2208 in Stealthchop is running fine 100mm/s on 24V, very silent UART(try giving them some more juice)
> Edit 13/9-2019:
> Did a test yesterday, on IOplatform (SKR 1.3), and I noticed the drivers now produce more noise with the latest update 2.0.x (14 days ago) that they did 2.0.x for a month ago.
> 
> I will try with the 2.1.x-dev version tonight.

After swithcing from Re-Arm to Mega and back I can tell that the difference in noise level (and the sound itself) is obvious. 

I swithced from standalone mode to UART. Still have the same issue. Tryed different currents, but can't get rid of the noise.

Found similar topic [https://github.com/MarlinFirmware/Marlin/issues/14183] but not sure if it is the same issue. But apparentely under Marlin 2.0 TMC2208 works different in StealthChop mode.

tvixen, 2019-09-17:
Did a test with the 2.1.x-dev and it's working perfect. No noise in StealthChop mode.
Please try it out.  (Ps. in UART you don't have to adjust vref, just set the current to 800mA approx. )

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

Pilson66, 2019-10-12:
@boelle So, what the final resolution? Why issue was closed without fixing?

boelle, 2019-10-12:
look at what @tvixen wrote ?? and then @uniksoid did not say if it worked or not

seems to me that issue has been fixed

Pilson66, 2019-10-12:
@boelle  @tvixen wrote that tested with 2.1-dev branch.
What about 2.0 ?

boelle, 2019-10-12:
i dont know i 2.1 is not the latest... look at when last updated

but noise from TMC steppers can be because they run on 12V and not 24V

Pilson66, 2019-10-12:
[image]

boelle, 2019-10-12:
that is a long time ago, a month plus 14 days, a lot has happend in that time

Pilson66, 2019-10-12:
So, the issue has been fixed in 2.0, or not fixed? If yes - by what commit? Can I have a look at this commit?

boelle, 2019-10-22:
@uniksoid is the issue still there?

uniksoid, 2019-10-22:
@boelle,
Sorry for a delay. I was waiting the delivery  of another Re-Arm.
I did a lot of different tests and eperiments. Tried replacing TMC2208 drivers, switching to UART mode, replacing ramps and motors. The only thing which remained the same - Re-Arm board.
While waiting for the new one I switched back to Mega+Ramps 1.5 + TMC2208 in standalone mode. All was working smooth.
Yesterday after receiving new Re-Arm board I switched my Mega with it (changed few pins due to difference in pinout and PWMs). Downloaded the latest Marlin 2.0. The problem remains. Actually it is much worse than it used to be. Not only the noise and vibrations from motors, but the actuall speed of printing and movement is increased a couple of times (or more). The firmware seems to ignore the speed from G-code. So i am trying to solve this at the moment. Did try to replace the new Re-Arm with the old one. The problem remains. Changing back to Mega with same Ramps set-up gives a smooth operation.

What's been done:
1. New drivers.
2. New RAMPS
3. New motors. 
4. New Re-Arm.

The problem remains. 

Will change the drivers to LV8729 to make sure that it is not a hardware issue. 
Will report in few hours

boelle, 2019-10-22:
@tvixen could you attach your working config? maybe @uniksoid could use that to compare and see if he can get it working

tvixen, 2019-10-22:
Well he can't use the information, as I'm runnig 24V on a SKR 1.3 now and not Re-Arm.
As I suggested earlier, please try them with 24 volts.

tvixen, 2019-10-22:
Hmm yes very strange ... Im running out of ideas.

uniksoid, 2019-10-25:
So... For some reason the issue is fixed.
Tried to replace drivers to L8729. It worked as expected. 
Returned TMC2208 and it worked also. 
Now running TMC2209. StealthChop works as expected. 
I think this issue can be closed.

closed 2019-10-25 by uniksoid