MarlinFirmware/Marlin #15205: [BUG] Termistor hotend 7 C everytime anet3d v1.7 motherboard

userosos, 2019-09-09:
Hi! I install marlin firmware on anet3d v1.7 and temperature extruder = 7 C and not reacts if i enable hotend. 7 C on the hotend everytime. I think that may be me pins not correct. I use pins for Anet3d 1.0 motherboard.

shitcreek, 2019-09-09:
Please follow the template and provide more information such as your configuration.

I'd wager that you haven't set the correct temp sensor. There are example configs of several Anet machines, start off with one of those if you aren't already.

But before continuing here, are you certain this is a bug and not a configuration error on your part? Have you exhausted other sources?

We're not a support group:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

userosos, 2019-09-09:
@shitcreek 
My config:
https://cloud.mail.ru/public/2dKg/4fKBXWjDA
>I'd wager that you haven't set the correct temp sensor. There are example configs of several Anet machines, start off with one of those if you aren't already.
No, temp bed is ok. And the sensor hotend does not react to heating in any way. If a switch hotbed resistor with hotedn resistor it work ok (but temp hotend is displayed in hotbed temp section).
I think that pinouts in anet3d v1.0 and pinouts in anet3d v1.7 not equally.

shitcreek, 2019-09-09:
Marlin 1.x is no longer being worked on. Please try 2.0 and report back if the issue still exists.

boelle, 2019-09-24:
@userosos still having problems?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle