MarlinFirmware/Marlin #15206: Bed leveling hig on the Right and low on the Left

muhammadelmogy, 2019-09-09:
Hi,
I have a DIY cartesian printer 300*300*400 with a heated bed and a closed chamber, I have been using this for more than a year now starting with marlin 1.1.8. I have upgraded to 1.1.9 recently and I am having a problem with Bed Auto Leveling.

I am using an inductive sensor for bed leveling and for z homing.
I am using safe z homing since my probe is always outside the boundaries of the bed.
I am using Bilinear Leveling Grid.

The problem is that after G29 one side of the bed is always lower than the other side which leads to uneven first layer adhesion!

I haven't really faced this issue before and I have been trying to fix this issue for some time now.
What exactly happing is that one side (left) has an under extrusion like form and the other side does not suffer from that.

I have been searching on the internet about this and found others facing exactly the same problem.

A workaround for this issue is really to decrease z offset (make it more negative) and increase the flow rate a bit for compensation, of course, this leads to ugly first layer at some places.

Bilinear Leveling Grid:
      0      1      2      3
 0 +0.163 -0.044 -0.093 -0.129
 1 +0.196 +0.038 -0.017 -0.128
 2 +0.195 -0.001 -0.065 -0.157
 3 +0.176 -0.065 -0.176 -0.285

AnHardt, 2019-09-09:
That's a relative commen problem. Please search around here a bit. 
[Visualisation of x-sled-rotation](https://github.com/AnHardt/Marlin/issues/84)

muhammadelmogy, 2019-09-09:
This issue came out after upgrading to 1.1.9 very shortly, I have not changed almost anything in my printer after the upgrade.

My printer is all metal parts nothing plastic in it and the hot end structure is custom made using e3d parts.

My machine is not a CR10 or Prusa printers form, it is a metal cube with bed moving up and down.
My machine does not suffer from any twist or issues with the z-axis, at least thats what I think!

AnHardt, 2019-09-09:
Said about everyone suffering from this.
Prove it. Is PROBE_OFFSET_FROM_EXTRUDER the same on the left and on the right?

muhammadelmogy, 2019-09-09:
What do you mean by " is the same", I suppose this totally irrelevant to this topic, even if it is not the same should the auto bed leveling compensate for the inequality?
Anyway, how should I "Prove" it ?

AnHardt, 2019-09-09:
Bed leveling has no clue about the nozzles position. It tries to keep the probe at the same height. If the offset between nozzle and probe is not constant the nozzle height varies.
How did you determine your offset? Exactly that way - but once on the left, once on the right.

This was discussed often enough. Please start reading.

muhammadelmogy, 2019-09-09:
"If the offset between nozzle and probe is not constant the nozzle height varies."

Well, the distance between the nozzle and the probe is always the same. in fact, it is always the same in all directions, what is changing is really the distance between the nozzle and the bed not between the nozzle and the probe since the bed is uneven surface!

I usually do not change my z offset, maybe -+ 0.01 at most since it is always printing perfectly, but after this happen recently, I started to compensate with -0.05 to overcome the lower side.

Roxy-3D, 2019-09-09:
> Said about everyone suffering from this.
> Prove it. Is PROBE_OFFSET_FROM_EXTRUDER the same on the left and on the right?

Home the machine.   Leave bed leveling inactive.     Go to each corner and lower the nozzle until it just touches the bed.    Do you get something very close to 0.0 mm in each corner?     If not, you need to square up your machine.   (And if the X-Gantry has a slight twist in it...   Even with a perfectly level bed, you will see discrepancies doing the corner test.)

muhammadelmogy, 2019-09-09:
> Home the machine. Leave bed leveling inactive. Go to each corner and lower the nozzle until it just touches the bed. Do you get something very close to 0.0 mm in each corner? If not, you need to [...]

It seems that I really do not understand well enough the auto bed leveling feature, I will read more about it though and try what you have said"I have already squared my machine before", even I have mentioned several times that this issue found its way after upgrading to 1.1.9 shortly!.

I will do this and mention the results later!

muhammadelmogy, 2019-09-09:
I remember seeing someone on youtube was talking about auto bed leveling and later in his video he intentionally raised one of the corners over the others and the print just worked fine, that's why I do not really understand why do I need to level the corners?

ManuelMcLure, 2019-09-09:
This is a writeup I did in another issue - although it refers to a Prusa-style X axis, it applies to any case of a twisted axis.

> Imagine the following with the standard 2-rod X axis (this is what my AM8 has):

> On the left side of the bed the lower rod is slightly more towards the back of the printer than the upper rod (in my case this is about 0.5mm)
On the right side of the bed, the upper and lower rods are exactly aligned (as they should be).
The rods are around 38mm apart - this makes the amount of twist asin(0.5 / 38) = 0.75 degrees.
This results in a twisted X axis. The nozzle on most i3-style machines is a few cm in front of the plane of the rods - in my case around 30mm. This means that the nozzle will be around sin(0.75) * 30mm = 0.4mm closer to the bed on the left side than on the right, assuming the bed is actually perfectly flat.

> Now in my case my sensor has a ``Y_PROBE_OFFSET_FROM_EXTRUDER`` of -40. So the probe is 40mm in front of the nozzle, or 70mm in front of the plane of the rods. Let's pretend my ``X_PROBE_OFFSET_FROM_EXTRUDER`` is 0. Then on the left side of the bed the sensor will be sin(0.75) * 70mm = 0.91mm closer to the bed - a difference of 0.5mm between the amount the nozzle changed vs the amount the sensor changed. So the bed leveling system thinks the bed is 0.5mm higher on the left side than the right side.

The TLDR is that if your ``Y_PROBE_OFFSET_FROM_EXTRUDER`` is not 0, any twist in your X axis will produce exactly the behavior you're seeing. The bigger your ``Y_PROBE_OFFSET_FROM_EXTRUDER``, the bigger the effect of the twist.

You have not provided your configuration files (as requested by the new issue template) so we don't know your configuration - there's a reason we always ask for those.

Roxy-3D, 2019-09-09:
The machines do not stay 'square'.   Crashing the nozzle can unsquare a number of different things.
Even printing things with curling edges can cause enough force as the nozzle plows through them to affect things.

Please do the corner test with bed leveling turned off and report what you see when you lower the nozzle to just touch the bed.      And we will help you figure things out. 

>  he intentionally raised one of the corners over the others and the print just worked fine, that's why I do not really understand why do I need to level the corners?

Yes...   And did he take out a calipers and measure the dimensional accuracy of the finished print for you?   The bed leveling can correct a whole number of bad and annoying problems.   But it does come at a cost.   You want the machine as square as you can get it before you turn the bed leveling on.

> The TLDR is that if your Y_PROBE_OFFSET_FROM_EXTRUDER is not 0, any twist in your X axis will produce exactly the behavior you're seeing. The bigger your Y_PROBE_OFFSET_FROM_EXTRUDER, the bigger the effect of the twist.

Yes.   This is true!

muhammadelmogy, 2019-09-09:
My config files, sorry forgot to upload!
[Config_1.1.9.zip](https://github.com/MarlinFirmware/Marlin/files/3592193/Config_1.1.9.zip)

muhammadelmogy, 2019-09-09:
I did the corners test and the difference between them was almost negligible (-+0.02, used feeler gauge)
 regarding my gantry, there is a very slight twist, I think less than 0.1 according to my square blocks and the feeler gauge!

boelle, 2019-09-10:
i wonder if this is the same with 2.0.x

ie start all over and manual by hand copy over old settings to configuraton.h and configuration_adv.h

muhammadelmogy, 2019-09-11:
After a lot of hassle and fiddling trying to measure and make sure everything is square and as accurate as possible, I have been able to bring it to a degree that I can call it accepted.

That was sure a hardware problem, not a firmware.

My understanding of the problem was based on some wrong assumption about auto bed leveling that it can fix this kind of misalignment but Nah, that wasn't the case!

Anyway, I am still having the same problem actually but it is now better than before and I can live with it for now.

I am closing this issue now.
Thanks

closed 2019-09-11 by muhammadelmogy