MarlinFirmware/Marlin #15211: [BUG] MINTEMP triggering, despite being set below room temperature

Gooberpatrol66, 2019-09-09:
### Description

Hotend heats as expected, then MINTEMP is triggered once it starts extruding. HEATER_0_MINTEMP is set to 5. The thermistor is functional and reports accurate temperature in Octopi graph.

### Steps to Reproduce

1. start a print job

**Expected behavior:** [What you expect to happen]
print job proceeds
**Actual behavior:** [What actually happens]
MINTEMP triggered, system stopped! Heater_ID: 0 - Printer halted. kill() called!
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
https://3dprinting.stackexchange.com/questions/10993/mintemp-hotend-drops-to-175c-once-extrusion-starts
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3592506/config.zip)

* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-09-09:
Are you certain this is not a hardware issue? Have you checked all your connections and cables?

Gooberpatrol66, 2019-09-10:
It reports normal temperatures until right after the plastic starts coming out.

shitcreek, 2019-09-10:
Marlin 1.x is no longer being worked on. Please try out Marlin 2.0 and report back if the problem still exists.

Gooberpatrol66, 2019-09-13:
It was a frayed wire.

closed 2019-09-13 by Gooberpatrol66