MarlinFirmware/Marlin #15212: Error compiling SKR PRO 1.1

davidcgu, 2019-09-10:
Good Morning

I'm trying to compile latest release downloaded today 9 Sep for an SKR Pro 1.1 but I'm having fail as follows:

`In file included from Marlin\src\module\stepper\trinamic.h:35:0,
                 from Marlin\src\module\stepper\trinamic.cpp:32:
Marlin\src\module\stepper\../../feature/tmc_util.h: In instantiation of 'TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(uint16_t, uint16_t, float, uint8_t, bool) [with char AXIS_LETTER = 'X'; char DRIVER_ID = '0'; AxisEnum AXIS_ID = (AxisEnum)0; uint16_t = short unsigned int; uint8_t = unsigned char]':
Marlin\src\module\stepper\trinamic.cpp:203:7:   required from here
Marlin\src\module\stepper\../../feature/tmc_util.h:213:38: error: no matching function for call to 'TMC2209Stepper::TMC2209Stepper(const uint16_t&, const uint16_t&, const float&, const uint8_t&)'
       TMC2209Stepper(RX, TX, RS, addr)
                                      ^
In file included from Marlin\src\module\stepper\trinamic.h:29:0,
                 from Marlin\src\module\stepper\trinamic.cpp:32:
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:993:3: note: candidate: TMC2209Stepper::TMC2209Stepper(Stream*, float, uint8_t)
   TMC2209Stepper(Stream * SerialPort, float RS, uint8_t addr) :
   ^~~~~~~~~~~~~~
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:993:3: note:   candidate expects 3 arguments, 4 provided
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note: candidate: constexpr TMC2209Stepper::TMC2209Stepper(const TMC2209Stepper&)
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\sd\SdBaseFile.cpp.o
 class TMC2209Stepper : public TMC2208Stepper {
       ^~~~~~~~~~~~~~
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\sd\SdFatUtil.cpp.o
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note:   candidate expects 1 argument, 4 provided
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\sd\SdFile.cpp.o
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note: candidate: constexpr TMC2209Stepper::TMC2209Stepper(TMC2209Stepper&&)
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note:   candidate expects 1 argument, 4 provided
In file included from Marlin\src\module\stepper\trinamic.h:35:0,
                 from Marlin\src\module\stepper\trinamic.cpp:32:
Marlin\src\module\stepper\../../feature/tmc_util.h: In instantiation of 'TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(uint16_t, uint16_t, float, uint8_t, bool) [with char AXIS_LETTER = 'Y'; char DRIVER_ID = '0'; AxisEnum AXIS_ID = (AxisEnum)1; uint16_t = short unsigned int; uint8_t = unsigned char]':
Marlin\src\module\stepper\trinamic.cpp:217:7:   required from here
Marlin\src\module\stepper\../../feature/tmc_util.h:213:38: error: no matching function for call to 'TMC2209Stepper::TMC2209Stepper(const uint16_t&, const uint16_t&, const float&, const uint8_t&)'
       TMC2209Stepper(RX, TX, RS, addr)
                                      ^
In file included from Marlin\src\module\stepper\trinamic.h:29:0,
                 from Marlin\src\module\stepper\trinamic.cpp:32:
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:993:3: note: candidate: TMC2209Stepper::TMC2209Stepper(Stream*, float, uint8_t)
   TMC2209Stepper(Stream * SerialPort, float RS, uint8_t addr) :
   ^~~~~~~~~~~~~~
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:993:3: note:   candidate expects 3 arguments, 4 provided
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note: candidate: constexpr TMC2209Stepper::TMC2209Stepper(const TMC2209Stepper&)
 class TMC2209Stepper : public TMC2208Stepper {
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\sd\SdVolume.cpp.o
       ^~~~~~~~~~~~~~
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note:   candidate expects 1 argument, 4 provided  
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\sd\cardreader.cpp.o
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note: candidate: constexpr TMC2209Stepper::TMC2209Stepper(TMC2209Stepper&&)
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note:   candidate expects 1 argument, 4 provided
In file included from Marlin\src\module\stepper\trinamic.h:35:0,
                 from Marlin\src\module\stepper\trinamic.cpp:32:
Marlin\src\module\stepper\../../feature/tmc_util.h: In instantiation of 'TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(uint16_t, uint16_t, float, uint8_t, bool) [with char AXIS_LETTER = 'Z'; char DRIVER_ID = '0'; AxisEnum AXIS_ID = (AxisEnum)2; uint16_t = short unsigned int; uint8_t = unsigned char]':
MCompiling .pio\build\BIGTREE_SKR_PRO\src\src\sd\usb_flashdrive\Sd2Card_FlashDrive.cpp.o
arlin\src\module\stepper\trinamic.cpp:231:7:   required from here
Marlin\src\module\stepper\../../feature/tmc_util.h:213:38: error: no matching function for call to 'TMC2209Stepper::TMC2209Stepper(const uint16_t&, const uint16_t&, const float&, const uint8_t&)'
       TMC2209Stepper(RX, TX, RS, addr)
                                      ^
In file included from Marlin\src\module\stepper\trinamic.h:29:0,
                 from Marlin\src\module\stepper\trinamic.cpp:32:
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:993:3: note: candidate: TMC2209Stepper::TMC2209Stepper(Stream*, float, uint8_t)
   TMC2209Stepper(Stream * SerialPort, float RS, uint8_t addr) :
   ^~~~~~~~~~~~~~
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:993:3: note:   candidate expects 3 arguments, 4 provided 
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note: candidate: constexpr TMC2209Stepper::TMC2209Stepper(const TMC2209Stepper&)
 class TMC2209Stepper : public TMC2208Stepper {
       ^~~~~~~~~~~~~~
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note:   candidate expects 1 argument, 4 provided
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note: candidate: constexpr TMC2209Stepper::TMC2209Stepper(TMC2209Stepper&&)
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note:   candidate expects 1 argument, 4 provided
In file included from Marlin\src\module\stepper\trinamic.h:35:0,
                 from Marlin\src\module\stepper\trinamic.cpp:32:
Marlin\src\module\stepper\../../feature/tmc_util.h: In instantiation of 'TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(uint16_t, uint16_t, float, uint8_t, bool) [with char AXIS_LETTER = 'Z'; char DRIVER_ID = '2'; AxisEnum AXIS_ID = (AxisEnum)2; uint16_t = short unsigned int; uint8_t = unsigned char]':
[... 16 lines cut ...]
In file included from Marlin\src\module\stepper\trinamic.h:35:0,
                 from Marlin\src\module\stepper\trinamic.cpp:32:
Marlin\src\module\stepper\../../feature/tmc_util.h: In instantiation of 'TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(uint16_t, uint16_t, float, uint8_t, bool) [with char AXIS_LETTER = 'E'; char DRIVER_ID = '0'; AxisEnum AXIS_ID = (AxisEnum)3; uint16_t = short unsigned int; uint8_t = unsigned char]':
Marlin\src\module\stepper\trinamic.cpp:252:7:   required from here
Marlin\src\module\stepper\../../feature/tmc_util.h:213:38: error: no matching function for call to 'TMC2209Stepper::TMC2209Stepper(const uint16_t&, const uint16_t&, const float&, const uint8_t&)'
       TMC2209Stepper(RX, TX, RS, addr)
                                      ^
In file included from Marlin\src\module\stepper\trinamic.h:29:0,
                 from Marlin\src\module\stepper\trinamic.cpp:32:
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:993:3: note: candidate: TMC2209Stepper::TMC2209Stepper(Stream*, float, uint8_t)
   TMC2209Stepper(Stream * SerialPort, float RS, uint8_t addr) :
   ^~~~~~~~~~~~~~
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:993:3: note:   candidate expects 3 arguments, 4 provided 
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note: candidate: constexpr TMC2209Stepper::TMC2209Stepper(const TMC2209Stepper&)
 class TMC2209Stepper : public TMC2208Stepper {
       ^~~~~~~~~~~~~~
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note:   candidate expects 1 argument, 4 provided  
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note: candidate: constexpr TMC2209Stepper::TMC2209Stepper(TMC2209Stepper&&)
C:\Users\davidcgu\.platformio\lib\TMCStepper_ID5513\src/TMCStepper.h:991:7: note:   candidate expects 1 argument, 4 provided
*** [.pio\build\BIGTREE_SKR_PRO\src\src\module\stepper\trinamic.cpp.o] Error 1
=============================================== [FAILED] Took 314.12 seconds ===============================================`

I have no clue where the issue could be comnig from, any advice is wellcome.

Best regards

RudyBenner, 2019-09-10:
With VSCode not running.
Delete
C:\Users\RudolfBenner\.platformio\.cache
C:\Users\RudolfBenner\Documents\GitHub\Marlin\.pio

Change Platforio.ini to

lib_deps =
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  LiquidCrystal@1.3.4
  #TMCStepper@<0.4.6
  https://github.com/bigtreetech/TMCStepper

I had thought this was fixed, but it is not.
I have no idea why this works.

Seems that TMCStepper 0.4.6 is the culprit.
TMCStepper@<0.4.6
or TMCStepper@0.4.5
do not work.

I can consistently compile the SKR Pro with this workaround.

RudyBenner, 2019-09-10:
Compiling your files......
1 succeeded in 00:09:30.069
slow computer

RudyBenner, 2019-09-10:
I have been working on this for several days, only to be thwarted by my lack of education.
This seems to be the only thing that works, and it has problems

[image]

As best as I can determine, TMCStepper 0.4.6 is the problem. It was supposedly fixed....

RudyBenner, 2019-09-10:
https://github.com/teemuatlut/TMCStepper/issues/56

davidcgu, 2019-09-10:
> I have been working on this for several days, only to be thwarted by my lack of education.
> This seems to be the only thing that works, and it has problems
> 
> [image]
> 
> As best as I can determine, TMCStepper 0.4.6 is the problem. It was supposedly fixed
> You may need to clear out
> C:\Users\RudolfBenner.platformio\.cache
> and
> C:\Users\RudolfBenner\Documents\GitHub\Marlin\.pio
> when the frustration factor rolls in.

Thanks Ruddy, that worked for me too. That means is some bug with the official TMC library.

Regards

RudyBenner, 2019-09-10:
I would be dangerous if I knew what I was doing.

davidcgu, 2019-09-10:
Theorically this is expected to be corrected on next official TMC library release 0.4.6... so as soon as it merges could be back to normal and won't need to use BTT library, from other side even if BTT merges another one that does not work you have already defined to use only 0.4.5, so no affection.

I you see on original ini was telling to use any version below 1.0.0 but you configured to be = to 0.4.5.

Regards

RudyBenner, 2019-09-10:
Yes, I was trying very hard to hook 0.4.5 but my efforts were lacking. Perhaps someone can bring me up to speed how to do this without having to get it from BTT, where the highest version is 0.4.5. 
I compared the 0.4.5 files, they are identical and they ultimately originate from the same address.

RudyBenner, 2019-09-10:
Just found this ....
https://github.com/teemuatlut/TMCStepper/tree/v0.4.5/src

davidcgu, 2019-09-10:
> Yes, I was trying very hard to hook 0.4.5 but my efforts were lacking. Perhaps someone can bring me up to speed how to do this without having to get it from BTT, where the highest version is 0.4.5.
> I compared the 0.4.5 files, they are identical and they ultimately originate from the same address.

As per error returned the problem is not the address, at certain point it receives arguments than expected  what causes the error during compiling... need to go deeper on library to investigate but I think it does not worth for me the efford.

See: candidate expects 3 arguments, 4 provided 

Regards

teemuatlut, 2019-09-10:
It's not a bug and will not be addressed in the next release.
The SKR Pro is an STM32 based board and the STM32 framework **does not include a software serial library**.
Neither do I want to add the library directly into TMCStepper because I most definitely do not want to then have to maintain it.
So this will not change until BTT or FYSETC or Marlin or whoever makes a standalone SW Serial library for the STM32 platform that I can then list as a dependency for PIO.

AnHardt, 2019-09-10:
The software serial is on its way for the (used with the PRO) official ST-kernel. Either https://github.com/stm32duino/Arduino_Core_STM32/pull/627 or https://github.com/stm32duino/Arduino_Core_STM32/pull/645 will find its way.

closed 2019-09-10 by thinkyhead

thinkyhead, 2019-09-11:
> See: candidate expects 3 arguments, 4 provided

Try telling `platformio.ini` to grab the "`master`" branch of `TMCStepper` and not the release version because sometimes the latest `TMCStepper` changes have not been given a version number yet.

davidcgu, 2019-09-11:
> > See: candidate expects 3 arguments, 4 provided
> 
> Try telling `platformio.ini` to grab the "`master`" branch of `TMCStepper` and not the release version because sometimes the latest `TMCStepper` changes have not been given a version number yet.

Taking bigtreetech it compiles, how should do exactly to take "master" branch of TMCStepper?  I will test.

Thanks in advance

sjasonsmith, 2019-09-11:
> Try telling platformio.ini to grab the "master" branch of TMCStepper and not the release version because sometimes the latest TMCStepper changes have not been given a version number yet.

I tried this. It didn't work for me, either because it wasn't compiling with C++2017, or there was no SoftwareSerial.h available. Based on the tooltips from VS Code and the comments above, I think it was probably both.

AnHardt, 2019-09-11:
When using a PRO, for now you need BTTs version of [TMCStepper](https://github.com/bigtreetech/TMCStepper). Not for some subtle version differences, but because of the [added software UART for F1/F4](https://github.com/bigtreetech/TMCStepper/commit/7fda12fd9575f893684cb4889015bf6162436950).

extesy, 2019-09-11:
https://github.com/stm32duino/Arduino_Core_STM32/pull/645 just got merged so STM32 core now has SoftwareSerial implementation.

RudyBenner, 2019-09-11:
too early to restore platformio.ini yet. tried to compile several times. no go.

richgain, 2019-10-27:
I just had the same error - failure to compile due to the missing software serial library.
I fixed it by using the following steps:
1. Downloaded the zip file from https://github.com/bigtreetech/TMCStepper
2. Located the Marlin-bugfix-2.0.x_SKR-PRO\.pio\libdeps\BIGTREE_SKR_PRO\TMCStepper_ID5513 folder
3. Replaced the contents of this folder with the equivalent contents from the bigtreetech zipfile
There are probably more efficient ways but I couldn't find them and this one worked for me.