MarlinFirmware/Marlin #15216: [FR] TMC driver - Different HOLD_MULTIPLIER by stepper

BenBeBoy, 2019-09-10:
### Description
This request is about the trinamics drivers. 
I am trying to add a feature in the "Configuration_adv.h" to let the user add different HOLD_MULTIPLIER for each motor.
The same way as with the current, microsteps, etc.

At the moment, it is only possible to use a general value for this setting.

#### Why
Each axis steppers does not work for the same task, at the same torque.
I have a DIY CoreXY printer, to move the X and the Y axis we need a combination of the α and β motor.
In some movements one of the motor shall not move to ensure correct position.
To avoid losing steps of this static motor we need to choose a high value for the general HOLD_MULTIPLIER.
But for example the Z motor doesn't need a high holding torque, or not as high as the X and Y.
The result is an excess heat generated for "nothing".

#### My work
I worked on the source code of marlin. I use **marlin 2.0** on an arduino **DUE board**
Because I am a mechanical engineer and not an expert programer, I have only basic and intermediate coding level.

The extra words I added are in comment for the example.

Here is what I've done on the "**Configuration_adv.h**"
[image]
I added this line on every tmc axis declaration  (##_HOLD_MULTIPLIER)

Here is what I've done on the "**trinamic.cpp**"
[image]
[image]
[image]

When I compile with the modifications, there is no errors.
I've made the modification by "copying" the structure. I don't really know what the code do.

When I upload the code on the due, no errors again.
BUT!
When starting pronterface and connecting the board, its starts to do crazy stuff...
Millions of lines are added and the app is stuck.
Here is an extract of the errors :
[image]
I have to erase the memory of the stm board to use it again.

What is the problem ? Is it my modification, or it needs something more ?
I would appreciate any support from the community

I attached you the "**trinamic.cpp**" file.
**[trinamic.zip](https://github.com/MarlinFirmware/Marlin/files/3596566/trinamic.zip)**

MarkMan0, 2019-09-10:
Not sure what you did there, but in conf_adv.h there are these lines:
```
  /**
   * You can set your own advanced settings by filling in predefined functions.
   * A list of available functions can be found on the library github page
   * https://github.com/teemuatlut/TMCStepper
   *
   * Example:
   * #define TMC_ADV() { \
   *   stepperX.diag0_temp_prewarn(1); \
   *   stepperY.interpolate(0); \
   * }
   */
```

Now I thin, you can do stuff like:
```
#define TMC_ADV() { \
    stepperZ.rms_current(Z_CURRENT, 0.3);\
}
```
And that would set the Z hold current to 0.3 times the nominal.

teemuatlut, 2019-09-11:
...or
```cpp
stepperZ.hold_multiplier(0.3)
```

BenBeBoy, 2019-09-11:
Ok, that makes sense.
It should erase the previous initialization values.

I added these lines :
```
#define TMC_ADV() { \
    stepperX.hold_multiplier(X_HOLD_MULTIPLIER); \
    stepperY.hold_multiplier(Y_HOLD_MULTIPLIER); \
    stepperZ.hold_multiplier(Z_HOLD_MULTIPLIER); \
    stepperE0.hold_multiplier(E0_HOLD_MULTIPLIER); \
}
```
Nothing crashes this time. Movements are correct. But it seems that the only value for the holding is the general one (HOLD_CURRENT).
I tried to use a very low value for only one motor and the torque is still the same as the others.

Even this doesn't work.
```
#define TMC_ADV() { \
    stepperX.rms_current(X_CURRENT, 0.1); \
}
```

Maybe the TMC_ADV() function is sent before the general initialization ?

teemuatlut, 2019-09-11:
`TMC_ADV` is executed after general init, but before reading values from EEPROM.
You can check the hold current with `M122`.

BenBeBoy, 2019-09-12:
I don't have any EEPROM on my board, an also it's not enabled. The problem is probably not here.

I tried to set the general `HOLD_MULTIPLIER` value to **0**
And different values in the `TMC_ADV()` function for the steppers.

Here is what `M122` returns :
```
Run current	18/31	18/31	18/31	18/31
Hold current	0/31	0/31	0/31	0/31
```
Seems like the function didn't work again.
But the weirdest thing is that my motor continue to hold even with a **0** value

I removed all of my modifications and set the `HOLD_MULTIPLIER` value to **0** 
Same problem again, motors are still holding.

thisiskeithb, 2022-04-09:
Added in https://github.com/MarlinFirmware/Marlin/pull/23345

closed 2022-04-09 by thisiskeithb