MarlinFirmware/Marlin #15217: [BUG] Z-Probe Pin randomly dropping

asko58, 2019-09-10:
Hi there,

Ever since I updated from Marlin 1 (MKS-Gen-L) to Marlin 2 (SKR 1.3) the Pin of my Z-Probe (3d touch clone) randomly drops during prints. 

This does not happen during every print but it does happen with all versions of Marlin 2 I have tried so far (latest as of yesterday). When it happens the BL-Touch clone that is in use blinks as if it has run into an error state.

What I can exclude as causes:
-wiring has been triple-checked
-vibrations: (hitting the probe hard does not trigger the error)
-slicer: used three different slicers
-gcode: it does not necessarily happen again printing the same gcode

Going back to Marlin 1 on the MKS Gen L however seems to fix the issue (as far as I have tested with several prints). I did not have a chance test Marlin 2 on Mks Gen L.

Am I missing something or could it be that Marlin sends some strange command to the probe that causes it to freak out? I'd really appreciate any help or hints.

asko

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3596859/config.zip)

tpruvot, 2019-09-10:
Be sure the pwm cable is not close/touching motor wires. EMI noise can do that...

Powerprobot, 2019-09-11:
On the SKR1.3 from Bigtreetech you must stabilize the onboard 5V Power Line. Or have you own a BL-Touch V3. The V3 is complicated... and need an extra 5V power supply. I think that is a Hardware issue, not Marlin.

blueeagle69, 2019-09-11:
I have a genuine  Bltouch 3.1 with an SKR 1.3, and I haven't had any issue with the pin randomly deploying. 
With the clones though it's anyone's guess half the time when they play up.

asko58, 2019-09-12:
Thank your for the answers. I have also suspected the hardware but could find anything so far...

> 
> Be sure the pwm cable is not close/touching motor wires. EMI noise can do that...

@tpuvot: The hint with the noise influencing the cable sounds interesting. The Probe wires run in parallel with the hotend and fan wiring, not the motor. Might that be a problem?

> 
> On the SKR1.3 from Bigtreetech you must stabilize the onboard 5V Power Line. Or have you own a BL-Touch V3. The V3 is complicated... and need an extra 5V power supply. I think that is a Hardware issue, not Marlin.

@Powerprobot: I will try to have a look at the 5v line with a scope to see how noisy it is. Have you stabilized the 5v line on one of these boards before? If so, what did you do? Connect an external supply or add a capacitor?

Powerprobot, 2019-09-12:
@asko58 : I installed on the SKR1.3 the FGD 12864 from BBT and the TFT. After installing the TFT the LCD Display show only scrambled characters. I had planned to install a BL-Touch. Thinking of all periphicals used the 5Volt onboard power, i need a solution. So i try to add a capacitor and it works. 
The only different is the BL-Touch. My Version ist a clone V1.2 and work fine. i hear from problems with the V3; they need an external 5Volt powerline.

asko58, 2019-09-12:
Thanks for the info. What capacity did you use? 22uF or more? Did you add a charging resistor or just the capacitor?

tpruvot, 2019-09-12:
the pwm cable is listening a frequency... If a motor stepper runs at a frequence near the pwm signals.. that can affect it... i dont think the 5V input is really related to that, what i seen is a change of received signal, not a reset of the bltouch (which do a self test on the antlabs v3..)

asko58, 2019-09-14:
To see if it is really the wiring I exchanged the whole cable going from the board to the BLtouch with a piece of foiled CAT5e ethernet cable. We will see if it helps. Thanks for your help again, I will post an update in case this thread can help someone else spot their problems. :)

Nuttavoot, 2019-09-15:
I have a genuine BLTOUCH v3.0 and a SKR v1.3. I also have randomly pin dropping during print. I also found that sometimes during power up the BLTOUCH does not do self-test and the red led blinks, I have to power off and on again several times to make it do self-test. Still finding for solution.

asko58, 2019-09-15:
If the probe does not perform a self-test during boot it is most certainly related to a positioning issue inside the probe. I have had the same problems over and over again while removing and reinstalling the probe. Remove the pin, clean the inside of the probe body (e.g. blow through) reinstall the pin and try to find out the correct positioning of the top screw. If it is off by only half a turn, the probe won't self test. I tightened it until lightly seated and went outward in quarter turns until the probe self tested again.

asko58, 2019-09-15:
By the way: five prints (two hours each) now and no drops anymore :) So far so good

ema987654, 2019-09-16:
I had that issue with the REARM board, I thought it was a power issue, I mean, the LCD lowered brightness every time the pin moved, but turns out there was a problem with the 5v signal to the servo, figured the REARM servo 3 output had 5v instead of the 3.3 on servo 0. (just to be clear, I'm also powering bltouch separately, with an exclusive 5v regulator circuit). To make it work I had to change the number of servos to 4: #define NUM_SERVOS 4, change the delay to have 4 elements in an array like so: #define SERVO_DELAY {0,0,0,0} and select the bltouch servo_pin_3: #define Z_PROBE_SERVO_NR 3 (and obviously plug it correctly on servo pin3 of the board)

closed 2019-09-17 by thinkyhead