MarlinFirmware/Marlin #15219: MARLIN 2.0 NOT COMPILING

cmdima, 2019-09-11:
Hello everybody, I´m new with this, and this is my first time configuring Marlin. I can´t compile Marlin 2.0 bugfix, it´s impossible!

I´m working with arduino mega 2560 and ramps 1.4. I set the configuration.h and tried with arduino IDE 1.8.9 without succes, then with the beta version of IDE 1.9, impossible, then i installed platformIO and I have some errors that can´t compile.

Please, can anyone can help me or maybe suggest a way to compile the firmware?

thanks for everything!

shitcreek, 2019-09-11:
You must provide more information to get help. Was there any error message? Did you follow any tutorial? Are you sure this is a bug?

We're not a support group
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

boelle, 2019-09-11:
@shitcreek maybe its time to have that bot octoprint has that tests if the issues are formatted correctly etc??

shitcreek, 2019-09-11:
I am all for it.

boelle, 2019-09-11:
@shitcreek now we just need to convince @thinkyhead that its a good idea to be done 

source is here: https://github.com/foosel/GitIssueBot

also look me up on discord if you want to have a discussion about it

boubouexpress, 2019-09-14:
a tu suivi un tuto ici http://marlinfw.org/docs/basics/install_platformio_atom.html
il faut installer GIT pour qui puisse télécharger les source

cinealfa, 2019-09-21:
why not just make it a form? Is that possible?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle