MarlinFirmware/Marlin #15220: [BUG] (Random Dementia bugfix-2.0.x)

rflulling, 2019-09-11:
### Marlin Dementia

**Steps to Reproduce:**
This is a random issue that happens when no one is looking. As soon as I can observe anything that allows me to recreate on demand, I will happily acquire as much extra data as I can. 

**Expected behavior:**
I expect the print to run until end of file, uninterrupted. If there is a crash or interrupt I expect to see a error or some indication on screen of the issue. 

**Actual behavior:**

Machine wanders off. Becomes unresponsive. Keeps extruding and moving though not directed by gCode. After a period the machine wakes up and begins to execute where it left off, only the axis are lost and the print must be restarted. During this, the LCD is still responsive, though commands are ignored. 

Machine may also just stop. No error, no crash. It as if S3D and Marlin just stopped talking to each other. There are no errors on screen and the Printer interface is otherwise fully responsive. 

Possibly related. After end of print, over USB. "Printing..." is still shown on the LCD. Interface is active as in printing mode not standby. 

#### Additional Information

Custom Build dual direct large format
PSU_NAME "Non-ATX 360W 24V 15A"
Controller: Makerbase MKS Gen 1.4
Driver: Todd Grace A4988
Firmware: Marlin BugFix-2.0.x (09/04/2019 20:44)

64bit Windows 7pro
Gigabyte GA-H55M-52H
Intel 650 i5-3.2Ghz, 8GB XMP DDR3, 1TB SATA3
XFX ATI R7 260 w/2GB DDR5

Link to printer is USB2.0 Intel chip set I believe. 

* I have included a zip with the full nightly build just in case it contains a bug particular to that build. 
[MarlinBugFix-20X-201909042044.zip](https://github.com/MarlinFirmware/Marlin/files/3598165/MarlinBugFix-20X-201909042044.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
[image]

MarkMan0, 2019-09-15:
Have you tried a lower baud rate, or a different cable?

rflulling, 2019-09-16:
The cable is a heavy duty and brand new from Cricut... More than robust
enough for heavy use. The controller is a MKS 1.4 so USB is all it does.
Not sure why but 250000is the rate specified as default. Have not tried to
tweak out a lower. I do know that without altering firmware, nothing will
connect at a lower rate.

rflulling, 2019-09-19:
Oddly I still have little to no data concerning this issue other than the randomness of it all. Perhaps I need to put Marlin in Developer mode? Maybe I can get some logs??

I had the machine running via SD card. Just done with a very successful overnight print. I have a new print running. 1hrs and 22min according to the LCD has passed and it's just done the first several layers. I am cleaning the previous part and then nothing. It just stopped. Twilight zone. As if it was never running. The bed is still warm, the fans are still on, the head is hot and in place. The end scripts did not run. The axis are blinking. It just came to a full stop with warning and without reason. Now, it just sits there looking at me. Me looking at it wondering what happened. 

At this time the machine was running on a file from SD card. I also disconnected the USB cable just in case there might be an unexpected interruption. So I can 100% remove poor quality USB cable quality, baud rate, Windows USB buggy behavior, S3D controller, USB chipset/driver issues. All eliminated as suspect. 

What remains. Issue in the code. Issue in firmware. Unknown fault in Controller card. 

I read that if the card becomes overwhelmed the firmware will reset the card. No reset has happened. 

Perhaps that the LCD is acting like the print flat out ended. Maybe there was an interruption to the SD card and the file was dropped? No persistence? No recovery? Maybe some switches I need to test and flip on...

For now. All I can do now is lower the build plate, clean it and start again, and Keep looking for something to report that might help Identify the issue, and resolve it.

boelle, 2019-10-12:
@rflulling any news?

rflulling, 2019-10-15:
I think this was some kind of screwy time out. Some how something on the PC hogged the CPU for a few seconds to long causing the wait to time out. Oddly I have had stuff connected for hours, even days without issue. 

As for the Machine randomly wandering off... I haven't seen that in several weeks. It might have been related to another bug. I am also working on switching up my controller from am MKS GEN V1.4 to an SKR PRO V1.1, if I can get it to work the way I want.

In the mean time, since no one else seems to have seen this bug, or at least they did not post their stories here. I will close the ticket. If, I see something new that is useful to all of you making and fixing the code, I will then reopen the ticket and update that information.

closed 2019-10-15 by rflulling