MarlinFirmware/Marlin #15221: [BUG] (short description)

rflulling, 2019-09-11:
### Marlin Dementia

**Steps to Reproduce:**
un-comment "#define HOMING_BACKOFF_MM { 10, 10, 0.1 }  // (mm) Move away from the endstops after homing" in a current version of the nightly build. 

_Note: have not downloaded again since 9/4/2019 to see if there is a change. It is possible this a bug already addressed. Less it be a conflict with another option unknown. 

No Files other than Configuration and Configuration_Adv have been edited._ 

**Expected behavior:**
With this uncommitted, in previous iterations, the bed will raise and perform normal XYZ zero and self calibration. Then move away from Zero. 

**Actual behavior:**
As of this build, un-commenting this line will fail the build. 

> sketch\src\module\motion.cpp: In function 'void homeaxis(AxisEnum)':
> sketch\src\module\motion.cpp:1661:32: error: 'Z_PROBE_SPEED_FAST' was not declared in this scope
>        line_to_current_position(Z_PROBE_SPEED_FAST);
>                                              ^
> exit status 1

#### Additional Information

Custom Build dual direct large format
PSU_NAME "Non-ATX 360W 24V 15A"
Controller: Makerbase MKS Gen 1.4
Driver: Todd Grace A4988
Firmware: Marlin BugFix-2.0.x (09/04/2019 20:44)

64bit Windows 7pro
Gigabyte GA-H55M-52H
Intel 650 i5-3.2Ghz, 8GB XMP DDR3, 1TB SATA3
XFX ATI R7 260 w/2GB DDR5

Link to printer is USB2.0 Intel chip set I believe. 

* I have included a zip with the full nightly build just in case it contains a bug particular to that build. 
[MarlinBugFix-20X-201909042044.zip](https://github.com/MarlinFirmware/Marlin/files/3598165/MarlinBugFix-20X-201909042044.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
[image]

sjasonsmith, 2019-09-11:
I think it changed in #14972. You should be able to build by uncommenting the following line in Configuration.h:

`//#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)`

rflulling, 2019-09-11:
I operate without a probe. I try to keep all everything turned off that it will not complain about being turned off. But testing to see if it will compile. 

Nope. Sorry. Failed compile. 

Same error.

rflulling, 2019-09-11:
Why is this required now? I don't even use a Z-Probe. 

_Heck lots of stuff seems to be broken, like manual LCD bed leveling. So I keep those functions turned off. No point cluttering my control menu with junk I can't or don't use._ 

---

**So it seems that some one... bound these functions:**
> `#define` Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z
to 
> `#define` HOMING_BACKOFF_MM { 10, 10, 0.1 }

So now homing backoff cannot be used without the Z Probe Fast Home option enabled. But Z Probe Fast Home, has no ties to Homeing backoff and works regardless. 

I still feel this is a bug as the firmware should be using whatever has been defined to detect home, be it an endstop or a probe and not suddenly requiring a probe. If this function now requires a probe. Some one needs to add those notes either in the Wiki, or in the config file as a comment.

rflulling, 2019-09-14:
in my setup (and you can see it from the uploaded firmware), I have the bumps and the back-off active. Off the top I don't recall what the bumps do. But in going home the machine will fine home quickly, then more slowly do it again, to get a very precise setup. If I had a very accurate Z-Probe, it would be good to be able to use that on the second bump. But I don't so it should be using the switch, which seems to be working as far as I can tell. Now, once the second bump is set and the machine should be Zero to within 0.05mm of each home, the machine will now Backoff, 10mm from each edge as a safety. I have clips on my bed and the head is able to run past the clips. The Z should lower slightly to take any tension off the head mount if any at all. Plus this avoids contact with the build plate during initial warm up. All in all, backoff is not a requirement as I can script the behaviors. But it's a great tool to have to simplify the mess. 

All in all the point of the post was to say, "Hey! Something changed and it was unexpected!" 

Thank you SJasonSmith, you helped identify how to get my setup going again, and as such further highlighted the bug. From my point of view this is resolved. But as a bug it is not. Unless some one with greater insight into the code would rather us believe this is a feature... (yes I deal with programmers at work, and bugs are often features so they don't have to be fixed...)

rflulling, 2019-09-14:
I should give the nightly build a download again tonight and see if this has changed. Update as needed here. -More to come.

sjasonsmith, 2019-09-18:
> I think that change will also cause the X/Y backoff to use Z_PROBE_SPEED_FAST as well, which doesn't seem correct. Backoff is normally only a couple millimeters, but if anyone uses a large backoff it would be noticeable. For many printers that XY backoff would run at about 4 mm/s rather than the old bump speed, which would probably usually be 25-50mm/s.

I finally got around to trying this out. There is a real problem, although it isn't what I had expected.
The backoff speed is actually _too fast_ for X and Y. This is because Z_PROBE_SPEED_FAST is defined in mm/m, and this is passing it to a function expecting mm/s.

I'm testing a fix for this and the original issue right now. I'll post a pull-request soon.
One risk is that the old fix was effectively raising Z at it's maximum axis speed, rather than the intended Z_PROBE_SPEED_FAST. It is possible that changing this to use Z_PROBE_SPEED_FAST correctly will cause the original BLTouch issue to return.

sjasonsmith, 2019-09-18:
In my experiments, it seems that the changes in #14972 have no real effect if your machine is freshly booted, because the feedrate defaulted to 1500mm/m. It was not using the bump speed as the pull request suggested, it was using feedrate_mm_s, which seems to be unchagned by the prior fast/slow homing moves.

If you execute a slow move prior to G28, _then_ you get normal-speed homing, but slow backoffs matching the prior move.

rflulling, 2019-09-18:
Happy to help identify an issue. Not wanting to break the software...

Just feels like binding the zprobe should be an if enabled function that
can be disabled not required. That way no changes to the values are needed
at all. Though notes warning what value might break it, would also be nice.
:)

sjasonsmith, 2019-09-19:
@rflulling, my pull request was just merged into bugfix-2.0.x to address this. Can you try it out and see if it resolves your concerns?

rflulling, 2019-09-19:
Am running on a copy from Sat? Though did not try to disable Z probe yet.

So far it has been running and stable with the probe function active though
I have no probe and have no other related functions active.

Will download a fresh copy and work in the settings tonight. Thank you.

rflulling, 2019-09-20:
Downloaded fresh 2.0.x version........ 09/19/2019.....

I wonder if I am the only one to use WinMerge to expedite Config setup? No it is not automated. Least not the editing... Only the compare. 
-Lets me snoop in all the other changes, even if I don't understand what they do. I can compare the full code of every document in the firmware in seconds. ;)
 
Transferring settings. 
Disabling "#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z"

Have a print running. But, Verify compile, looks good, no errors.
Unless I can break it, I will say fixed. Upload after print.

sjasonsmith, 2019-09-21:
> Have a print running. But, Verify compile, looks good, no errors.
> Unless I can break it, I will say fixed. Upload after print.

Did you have a chance to upload and try the new firmware? If it is ok, can you go ahead and close this?

rflulling, 2019-09-21:
will add to if anything new pops up. Otherwise, yes looks good right now. Thank You for the fix.

closed 2019-09-21 by rflulling

rflulling, 2019-09-22:
Just discovered one more lurking that I did not catch. Is related. 

`//#define Z_PROBE_LOW_POINT  -2` // Farthest distance below the trigger-point to go before stopping

[image]