MarlinFirmware/Marlin #15223: Default steps for TMC2208 ?

tvixen, 2019-09-11:
Not so much a request/bug as a simple question.

In the Configuration_adv.h, the standard setting for a TMC2208 Z axis is set to 3,
and the #define E0_HYBRID_THRESHOLD  is set to  30 ?
But i wounder why.  I know Z its not supposed to run fast, but why not 30mm/s ?
And Extruder make a lot of noise when it's only 30, so why not 100mm/s  like X and Y ?

 
/**
   * TMC2130, TMC2160, TMC2208, TMC2209, TMC5130 and TMC5160 only
   * The driver will switch to spreadCycle when stepper speed is over HYBRID_THRESHOLD.
   * This mode allows for faster movements at the expense of higher noise levels.
   * STEALTHCHOP_(XY|Z|E) must be enabled to use HYBRID_THRESHOLD.
   * M913 X/Y/Z/E to live tune the setting
   */
  //#define HYBRID_THRESHOLD

  #define X_HYBRID_THRESHOLD     100  // [mm/s]
  #define X2_HYBRID_THRESHOLD    100
  #define Y_HYBRID_THRESHOLD     100
  #define Y2_HYBRID_THRESHOLD    100
  #define Z_HYBRID_THRESHOLD       3
  #define Z2_HYBRID_THRESHOLD      3
  #define Z3_HYBRID_THRESHOLD      3
  #define E0_HYBRID_THRESHOLD     30
  #define E1_HYBRID_THRESHOLD     30
  #define E2_HYBRID_THRESHOLD     30
  #define E3_HYBRID_THRESHOLD     30
  #define E4_HYBRID_THRESHOLD     30
  #define E5_HYBRID_THRESHOLD     30
  /**

shitcreek, 2019-09-12:
First off, you are talking about speed and not steps. 

The threshold setting is to switch between spreadCycle and stealthChop - it is not your set speeds. StealthChop cannot operate silently at certain thresholds and does not provide enough torque which will result in skipped steps.

In order for a leadscrew with a pitch of 5mm to go 100mm/s | (100/5*60), it must spin 1200rpm or 20rps. Although it can match the speed, the issue is with acceleration and deceleration. Since Z only has to move up once every layer and by a tiny margin, there is no need to go faster and it wouldn't be all that noticeable.

As for the extruder, the extruder has to push filament which has a bit of resistance. If you tried to extrude too fast, it will skip.

tvixen, 2019-09-13:
No skipsteps at all on 80mm/s  (Ender 3 Pro)

Thanks mate.

closed 2019-09-13 by tvixen