MarlinFirmware/Marlin #15224: [BUG] CoreXY single reversed axis

soimy, 2019-09-11:
### Description

I've set up a printer with CoreXY structure with Marlin-bugfix-2.0.x newest firmware. Everythings work fine except there will always be one axis running at the reversed direction.
I tried to `INVERT_X_DIR` / `INVERT_Y_DIR` but this is for motor direction, not hybrid CoreXY A/B axis. if I set
```
INVERT_X_DIR true;
INVERT_Y_DIR true;
```
It simply changes the reversed axis from X->Y. I've looked closely on motor behavior:
```
X+ movement: MotorA CW, MotorB CW
X- movement: MotorA CCW, MotorB CCW
Y+ movement: MotorA CCW, MotorB CW
Y- movement: MotorA CW, MotorB CCW
```
So I wanna reverse a single axis, I can't achieve it with simple motor direction invertion.

So my question is: Is there any way to reverse a single axis in CoreXY structure?

#### Additional Information

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3601162/Configuration.zip)

shitcreek, 2019-09-11:
Have you tried swapping the motors ie swapping their plugs?
Here is a good write up on coreXY: https://drmrehorst.blogspot.com/2017/08/setting-up-corexy-printers-origin-and.html

vivian-ng, 2020-01-24:
> Here is a good write up on coreXY: [https://drmrehorst.blogspot.com/2017/08/setting-up-corexy-printers-origin-and.html](https://drmrehorst.blogspot.com/2017/08/setting-up-corexy-printers-origin-and.html)

This should be pinned somewhere... to help everyone who has issues trying to get their coreXY printers going in the right direction.