MarlinFirmware/Marlin #15225: [BUG] Bed level tilt

arvydas, 2019-09-11:
### Description

Using 2.0.x-bugfix latest.
Printer: Ender 2
Sensor: BLTouch V3
Bed leveling: Bilinear 3x3 grid or linear 3x3 grid produces same results

There is a visible bed tilt not compensated by BLTouch. It is much more apparent the further away X axis travels, but it is slightly diagonal.

[image]

[image]

#### Additional Information

Startup script for Simplify3D:

```
M117 Heating bed...
M140 S[bed0_temperature] ; heat bed
M190 S[bed0_temperature] ; wait for bed to reach temperature
G28 ; home all axes
M117 Autolevel...
G29; autolevel
G1 X0 Y0 Z5.0 F3000 ; Move to extruder heating position 0:0
M117 Heating extruder...
M104 S[extruder0_temperature] T0 ; set extruder temperature
M109 S[extruder0_temperature] T0 ; wait for extruder to reach temperature
M117 Purge extruder
G92 E0 ; reset extruder
; G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
G1 X3 Y0 Z0.3 F5000.0 ; move to start-line position
G1 E40 F200 ; extrude 40mm of filament
G1 X3 Y120.0 Z0.3 F1500.0 E55 ; draw 1st line
G1 X3.3 Y120.0 Z0.3 F5000.0 ; move to side a little
G1 X3.3 Y60 Z0.3 F1500.0 E58 ; draw 2nd line
G1 X3.3 Y50 F180 ; Wait for ooze
; G1 E27 F3000 ; retract filament 3mm
G92 E0 ; reset extruder
; done purging extruder
G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
M117 Printing...
```

Config files attached:

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3601241/Marlin.zip)

I have been googling about this issue and the only explanation I could find is the answer to the stackoverflow question that the problem is caused by Hysteresis.

https://3dprinting.stackexchange.com/questions/9956/bl-touch-bed-leveling-seems-to-produce-tilted-bed-level-around-y-axis/9960

Ender 2 uses a single Z axis stepper with one threaded rod to drive it. The issue is consistent and reproducable every time.

On Prusa printers this could be caused by X rods being twisted, however Ender 2 X axis moves on wheels on top of aluminium extrusion. Could this be a mechanical issue when the X axis moves farther away from the Z axis it bends the X axis aluminium extrusion due to weight thus changing the offset of the probe?

One morething which is a bit strange: if I use bilinear bed leveling, my nozzle has to be -0.75, however with linear it has to be -1.00 to produce same results.

muhammadelmogy, 2019-09-11:
Take a look at this issue over there, it may give you a hint .
`https://github.com/MarlinFirmware/Marlin/issues/15206`

Roxy-3D, 2019-09-12:
Printing a G26 mesh validation pattern and posting the results will give us a lot of information about what is going on.

TwoToneEddy, 2019-12-19:
> Apparently ABL can't compensate for an uneven bed

Isn't this the point of ABL? I don't understand.

Vertabreak, 2019-12-19:
If the frame isn't square it will have issues or if there are belt issues. The idea that its auto bed leveling is not correct it's assisted it assumes there are no other issues present to function.

arvydas, 2019-12-19:
I'd like to add that one of the causes for the issue was related to z coupler. It was too springy causing to compress while moving z axis and then release the tension at a random time. It put the BLTouch measurements completely off.

UBL works magic. It only fails when mechanical issues occur.