MarlinFirmware/Marlin #15226: [BUG] SKR Mini E3 DIP (STM32) with TMC2130 booting issues with TMC_DEBUG enabled

AmiSMB, 2019-09-11:
### Description

I have replaced my stock Creality Ender 3 main board with the SKR Mini E3 DIP with TMC2130 drivers and a BLTOUCH. Using the latest commits from bugfix-2.0 branch I have found that if TMC_DEBUG is enabled then the firmware is flashed on the first boot with the new firmware but after a power down and a reboot the firmware does not load even though there is a FIRMWARE.CUR on the SD Card which matches the firmware.bin file that was added to the card. I did compile the firmware before with the same configuration and was able to have TMC_DEBUG and use the M122 command without an issue. The only difference that I can see is that with the latest commits I now see the SD card mounted when I plugin the USB cable to my computer whereas I did not before. The PR that relates to this is #15180

### Steps to Reproduce

1. Write firmware.bin to SD card and insert into board
2. Power up board and wait just over 10 seconds for firmware to flash and load
3. Power off board
4. Power on board firmware flashes again and starts up correctly
5. Power off board
6. Power on and nothing happens on the LCD and unable to connect via serial port
7. Recompile with TMC_DEBUG disabled
8. Write firmare.bin to SD card and insert into board
9. Board flashes correctly and is fine after power cycling

**Expected behavior:** 
I expected the board to start with the firmware that is now flashed to the board via the bootloader.

**Actual behavior:**
Board does not start until I put firmware.bin back on the SD card and repeat the process and if I do not comment out the TMC_DEBUG line then I end up with a non booting board after power cycling more than twice.

#### Additional Information

I have attached my configuration files

[Ender3_SKR_E3_DIP_TMC2130_BLTOUCH.zip](https://github.com/MarlinFirmware/Marlin/files/3601454/Ender3_SKR_E3_DIP_TMC2130_BLTOUCH.zip)

brew99, 2019-09-11:
I can't even compile the latest version with my SKR mini E3 (non-DIP version)  and TMC_DEBUG enabled.  Error says something in regards to "won't fit on ROM".   Comment out TMC_DEBUG and I'm good at 26.9% DATA, and 86.2% PROGRAM.

I ran into the issue you were having about a week ago, then on Sep6 version it was OK.   Today it won't fit.

AmiSMB, 2019-09-11:
I had the same problem but the config I posted here compiles fine as I made sure I uncommented SLIM_LCD_MENUS which helps a lot.

thisiskeithb, 2019-09-11:
It's probably safe to assume that `TMC_DEBUG` and emulated `EEPROM` are not compatible on the SKR Mini line of boards.

@tpruvot: Do you have other `STM32F1` hardware with emulated `EEPROM` & `TMC_DEBUG` enabled to test if there are similar issues?

tpruvot, 2019-09-11:
TMC_DEBUG on skr mini, no issue
DATA:    [===       ]  25.5% (used 12536 bytes from 49152 bytes)
PROGRAM: [========  ]  78.4% (used 205564 bytes from 262144 bytes)

thisiskeithb, 2019-09-11:
> TMC_DEBUG on skr mini, no issue (TMC2208 here)
> DATA: [=== ] 25.5% (used 12536 bytes from 49152 bytes)
> PROGRAM: [======== ] 78.4% (used 205564 bytes from 262144 bytes)

Have you tried emulated `EEPROM` on the Longer3D board?

tpruvot, 2019-09-12:
I use both here on it, to easily swap Marlin and a more "Stable/tested" firmware.

thinkyhead, 2019-09-13:
> SD card insertion is no longer detected

Is `SD_DETECT_PIN` being set as `-1`?

salacpavel, 2019-09-13:
> 
> 
> > SD card insertion is no longer detected
> 
> Is `SD_DETECT_PIN` being set as `-1`?

No, its set as PC4

tpruvot, 2019-09-14:
try #15255 works well on the mini if you ignore the possible hangs caused by the usb composite (unrelated to the PR)

boelle, 2019-09-24:
@AmiSMB still having issues?

AmiSMB, 2019-09-25:
I am going to try a pull tonight and enable TMC_DEBUG and see if the same thing happens

AmiSMB, 2019-09-27:
So after testing with different configurations this is my findings with the latest branch of bugfix-2.0.x at commit 62e4e05a1947b006dfbaa33e33a75dcd30cf993b

platformio.ini
default_envs = STM32F103R_bigtree_USB
   If TMC_DEBUG is enabled this requires SLIM_LCD_MENUS to be enabled so that the firmware is not too big. The firmware is uploaded correctly on the first power up. After powering down and turning back on again the firmware does not load and just the LCD backlight is lit. Looking at the SD card FIRMWARE.CUR is a zero byte sized file.
  I also found that if you mount the SD card under an OS then the serial connection is stopped and you cannot reconnect unless you unplug and replug in the USB cable.

platformio.ini
default_envs = STM32F103R_bigtree
   As the USB Composite is not included then the TMC_DEBUG can be enabled as well as SLIM_LCD_MENUS can be disabled as this will fit the flash size. Everything works fine apart from the fact you cannot mount the SD card under an OS, which is to be expected as USB Composite is not enabled. Firmware is uploaded on first boot and starts without a problem after every power cycle.

TimoBre, 2019-09-28:
Hi, I'm new at the GitHub, so :-)
I had the same problem -  the the progmem space is to less 256K. But it has nothing to do with the TMC2130 :-( Maybe it has to be divided to a new one.

I found an workaround which helped me for the lack of memspace.

First, i read that the actual Marlin does not talk rigth to the SKR e3 DIP.
It takes a wrong CPU/MCU. The rigth one STM32F103R**E**T6 has 512K !.
If you do change it with the work around from - _Gustavo Alvarez sl1pkn07 at gitmemory.com 
https://www.gitmemory.com/sl1pkn07_, and it will be ok.
Atom-PlatformIO will compile it perfectly, also with all memory expensive optiones enabled. 

Im my view there is a bug which declared the MCU wrong. It is not the C modell! It is the E modell. Also verified on my CPU/MCU number writton on my processor.

Compare it here:

C modell 256 k
https://www.mouser.de/ProductDetail/STMicroelectronics/STM32F103RCT6?qs=%252BB84zevwoRA6TYzZIgOIoA%3D%3D
E model 512 k 
https://www.mouser.de/ProductDetail/STMicroelectronics/STM32F103RET6?qs=tZXS5FmuAdnGL83bu1Xcew%3D%3D

[image]

Also i get an compiling error if i past the complete STM32F1RE_SKR_MINI.id from him. I only ad the part
 
 MEMORY
{
  ram (rwx) : ORIGIN = 0x20000000, LENGTH = 48K - 40
  rom (rx)  : ORIGIN = 0x08007000, LENGTH = 512K - 28K
} 

than i get the green huck and 512 k . But my work around for the STM32F1RE_SKR_MINI.id caused in a lag of knowledge. Maybe he is rigth and i missed someting

Thank you to the community for your outstanding support!

boelle, 2019-10-24:
will close this one as it seems to be working

we can always reopen if its not the case

reloxx13, 2019-10-25:
> will close this one as it seems to be working
> 
> we can always reopen if its not the case

its not "working". its a space issue, if its not possible to detect it by the build script, this can be closed.
devs have to answer if they want to catch this error.
if not, we will get more "issues" asking why its not working cause it does not show its out of space.

boelle, 2019-10-25:
hehe, forgot to hit close :-D

but since its not working 100% we should keep it open

boelle, 2019-11-06:
@thinkyhead @shitcreek time to have a look here before the issue gets lost and not fixed

AmiSMB, 2019-11-26:
Have been playing around with different firmware files and I am compiling for the 256k chip but when the firmware file gets around 230k I experience issues. I have tried 512k chip builds but I think that my board has a bootloader which refuses to support this so will have to check with an ST Flasher when it arrives.

closed 2019-11-26 by thinkyhead