MarlinFirmware/Marlin #15228: Can't disable STEALTHCHOP on TMC2208 MKS_GEN_L in Marlin 2.0 bugfix

porlock, 2019-09-11:
Commenting flowing lines
```
  #define STEALTHCHOP_XY
  #define STEALTHCHOP_Z
  #define STEALTHCHOP_E
```
does not disable STEALTHCHOP, even settings 
`  #define TMC_ADV() { stepperX.en_spreadCycle(true); }`

does noting

M122 shows STEALTHCHOP set to on, and motors works ulta silently, like on STEALTHCHOP, i test it on MKG_GEN_L

### Steps to Reproduce

1. Comment out this lines
```
  //#define STEALTHCHOP_XY
  //#define STEALTHCHOP_Z
  //#define STEALTHCHOP_E
```
2. add this line 
`#define TMC_ADV() { stepperX.en_spreadCycle(true); }`
3. Compile and install Marlin 2.0 bugfix on MKS_GEN_L with TMC2208
4. Test is  STEALTHCHOP is disabled on X axis or any other

boelle, 2019-09-11:
attach your 2 config files as a zip, just as the issue template told you to

porlock, 2019-09-11:
sure, there you go
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3602365/Marlin.zip)

shitcreek, 2019-09-11:
Did you make sure to run M502 and M500 after flashing?

porlock, 2019-09-11:
Nope, i did not, I thought that disabling STEALTHCHOP for axis is not stored in user eprom configuration, because i don't see any gcode that could change that setting with ouch recompiling firmware.

shitcreek, 2019-09-11:
I had an issue when I was setting up the EZBoard with StealthChop until I ran M502.

reloxx13, 2019-09-11:
its stored

http://marlinfw.org/docs/gcode/M569.html

porlock, 2019-09-11:
Thanks, will test it and get back, thanks.