MarlinFirmware/Marlin #15231: [BUG] Pancake motor randomly stops after few minutes on TMC2208 Marlin 2.0 bugfix

porlock, 2019-09-11:
### Description
After i moved from marlin 1.1 to 2.0 I facing an issue that pancake motors like 17HS4023 driver by TMC2208 randomly turns off, when this happens i can freely move shaft. I happens on continuous moves, it is easy to observe when pancake motor is installed as extruder that is because it works all time when print goes, without any breaks. When motors stops for few seconds (for example when i pause a print)  and new G1 codes comes it start works as normal. I special bought another pancake motor to test it, and it is the same situation. What is real strange that this not affect a bigger Nema17 motors.  I make some other tests and this is what i discovered:

1) On Marlin 1.1 everything works fine no issues at all, I use it for years without any problem. For a test I downgraded for 1.1 from 2.0 and everything  get back to normal, so it is definitely related to 2.0
2) On Marlin 2.0 in TMC_Standalone mode everything works fine too.
3) For test i tried  to lower the values of current to 400mA RMS, motors and steps have room temperature but motors stops anyway.
4) Downgrading TMC LIB to https://github.com/teemuatlut/TMCStepper/tree/revert_uart 
does not help.

### Steps to Reproduce

1) Use MKS_GEN_L, TMC2208, 17HS4023 motor as extruder, connect all in UART.
2) Compile and install Marlin 2.0 bugfix
3) Run a  print that will take approx 2h
4) Check if extruder motor stops without any reason.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3602931/Marlin.zip)

shitcreek, 2019-09-11:
Not sure what is the cause of your problem, but 600mA seems quite low for the extruder. Since the extruder has to work harder than X and Y motors, it typically need more power than they do. I have it set at 900 for my extruder using TMC2208. Z motors need more power as well since they have to move a heavy bed against gravity - I have mine at 1000mA.

porlock, 2019-09-11:
I works fine for 0.85V in Standalone mode witch Marlin 2.0, and at 600mA RMS over UART in marlin 1.1.9 bugfix

shitcreek, 2019-09-11:
We've got a bunch laying around, but not used on any of our machines. I might test one out if this goes further. Mind posting your old config for side by side comparison?

porlock, 2019-09-11:
Thanks i will be appreciated, this is my old config from Malrin 1.1.9 bugfix, that works fine.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3603098/Marlin.zip)

mak0t0san, 2019-09-13:
Have you tried with a regular sized stepper? I had this problem with my extruder motor where it would turn off a few minutes into the print if I had Stealthchop on but turning it off, or using Hybrid mode and setting the value to something low like 30 fixed the issue.  The problem went away when I turned off LIN_ADVANCE, but sounds like there might be an issue with the way Marlin handles stealtchop.  I have this problem with TMC2208s in UART mode and with TMC5161 drivers.

porlock, 2019-09-13:
Hej mak0t0san , yes it is fine on bigger motors, and i don't use LIN_ADVANCE, please check info in initial post there is a lots of details regarding conditions and test that i have made so fare

tvixen, 2019-09-17:
I think you should try the new 2.1.x-dev.  My experience says there has been a few changes with the TMC drivers.
I did note the the noise from the steppers was louder in StelthChopmode in the version 2.0.x for 14 days ago. But this is now gone in 2.1.x-dev.

shitcreek, 2019-09-18:
I've just tested with a pancaked motor (17HS4023) running freely (not attached to the extruder part) and did not have any issue. I also held the spindle and resisted the motor and it kept turning without any issue.

LMF5000, 2019-09-19:
To the OP - could it be drivers overheating and doing a protective shutdown?

To test, lower your extruder driver current, say to half of its current value. Put the stepper on the table again and run a long print. If it doesn't stop it means that your problem was likely caused by drivers overheating. Try fitting bigger heatsinks and using forced air flow from a fan blowing on the drivers.

stefi01, 2019-10-07:
i get the nsame thing but with a rambo board, works fine with marlin 1.1.9 but the latest bugfix 2.0 i can start to print and then after a few minutes the printer just stops

boelle, 2019-10-12:
@porlock still having issues?

porlock, 2019-10-12:
@boelle  yes.

porlock, 2019-10-12:
I swap MOBO to SKR 1.3 will see how it goes, same currents, same firmware version, same steppers, same motors, same cables.

boelle, 2019-10-12:
not sure if the question has been asked before, but TMC's need active cooling, a 80mm fan will do fine

porlock, 2019-10-13:
@boelle Yes I have a 40x40 fan on mobo, stepers are cold, I made a test when I drop a current to low value, unmounted motor from extruder, put it freely on the table and run print "on dry". Motor and step drivers haved a room temperature and it stops anyways. I test it in skr 1.3 and this issue doesn't occurred. So it is affecting only MKS s gen l and Rambo (like @stefi01 reported)

boelle, 2019-10-24:
@porlock could it be a hardware issue? ie a error by the board maker?

porlock, 2019-10-24:
Don't think so, on marlin 1.x it works fine

boelle, 2019-10-24:
what if there is no filament loaded? but you still print

porlock, 2019-10-24:
The same, even if I unmount motor from extruder

boelle, 2019-11-06:
@thinkyhead @shitcreek time to have a look here before the issue gets lost and not fixed

shitcreek, 2019-11-06:
I was not able to reproduce this issue using a pancake motor. Perhaps I didn't test it long enough? I'll give it a try again.

porlock, 2019-11-06:
> I was not able to reproduce this issue using a pancake motor. Perhaps I didn't test it long enough? I'll give it a try again.

Did you use MKS Gen motherboard ?

shitcreek, 2019-11-06:
No. We do have a couple of them laying around I could test out.

porlock, 2019-11-06:
please do, i test it on SKR 1.3 and everything was fine, issue was only on mks gen l

sjasonsmith, 2019-11-09:
I just received an MKS Gen L and will see if I can reproduce this. I just dealt with some similar 2208 issues working with an SKR Pro, perhaps something similar is happening here.

sjasonsmith, 2019-11-10:
@porlock, can you post updated configs which build against the current state of bugfix-2.0.x, and reproduce the problem?

It looks like the currently posted configs are configured for SpreadCycle, even though the issue involves StealthChop?

sjasonsmith, 2019-11-11:
@porlock, I've set up a system to try to reproduce this, but so far have not seen the issue.

I don't want to spend any more time on it, unless you can confirm you still have a problem, and can provide updated configs for the latest bugfix-2.0.x.

It would also be helpful to have the STL you are trying to print, along with the sliced gcode you are printing. The sliced output might be relevant, because speeds and acceleration are often overridden by your slicer.

porlock, 2019-11-11:
i can't test it right now on current branch sorry :(

sjasonsmith, 2019-11-11:
> i can't test it right now on current branch sorry :(

Do you know when you might be able to try it?

boelle, 2019-11-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-11-24 by boelle