MarlinFirmware/Marlin #15234: [BUG] SKR Pro with MMU2 compile issue

TB1405, 2019-09-12:
Hi i tried to compile marlin 2.0.x (10.09.2019) 
for the SKR Pro with MMU2 feature. but at the end of the compiling i got the following error message. 
I use VS Code and PlatformIO

`.pio/build/BIGTREE_SKR_PRO/src/src/Marlin.cpp.o: In function `manage_inactivity(bool)':
c:\Users\Tim\Documents\3D Drucker\Marlin_Download\hourlyBuild\Marlin-bugfix-2.0.x/Marlin\src/Marlin.cpp:494: undefined reference to `Serial2'
.pio/build/BIGTREE_SKR_PRO/src/src/Marlin.cpp.o: In function `setup':
c:\Users\Tim\Documents\3D Drucker\Marlin_Download\hourlyBuild\Marlin-bugfix-2.0.x/Marlin\src/Marlin.cpp:1160: undefined reference to `Serial2'
.pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o: In function `serialprintPGM(char const*)':
c:\Users\Tim\Documents\3D Drucker\Marlin_Download\hourlyBuild\Marlin-bugfix-2.0.x/Marlin\src\core/serial.cpp:38: undefined reference to `Serial2'
.pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o: In function `serial_echopair_PGM(char const*, char const*)':
c:\Users\Tim\Documents\3D Drucker\Marlin_Download\hourlyBuild\Marlin-bugfix-2.0.x/Marlin\src\core/serial.cpp:42: undefined reference to `Serial2'
.pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o: In function `serial_echopair_PGM(char const*, int)':
c:\Users\Tim\Documents\3D Drucker\Marlin_Download\hourlyBuild\Marlin-bugfix-2.0.x/Marlin\src\core/serial.cpp:44: undefined reference to `Serial2'
.pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o:c:\Users\Tim\Documents\3D Drucker\Marlin_Download\hourlyBuild\Marlin-bugfix-2.0.x/Marlin\src\core/serial.cpp:45: more undefined references to `Serial2' follow
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\BIGTREE_SKR_PRO\firmware.elf] Error 1
`

#### Additional Information
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 2

  // Serial port used for communication with MMU2.
  // For AVR enable the UART port used for the MMU. (e.g., internalSerial)
  // For 32-bit boards check your HAL for available serial ports. (e.g., Serial2)
  #define INTERNAL_SERIAL_PORT 2
  #define MMU2_SERIAL Serial2

TB1405, 2019-09-13:
Done.

boelle, 2019-09-24:
@TB1405 and this is still an issue?

TB1405, 2019-10-01:
Yes

TB1405, 2019-10-09:
With the newest marlin the mmu2 succesfully communicates with the SKR PRO.

#define SERIAL_PORT 0

//#define SERIAL_PORT_2 -1 //Do not use!

**Configuration_adv.h**

  // Serial port used for communication with MMU2.
  // For AVR enable the UART port used for the MMU. (e.g., internalSerial)
  // For 32-bit boards check your HAL for available serial ports. (e.g., Serial2)
 // #define INTERNAL_SERIAL_PORT 2
  #define MMU2_SERIAL Serial1

Connect the MMU2 to the TFT port on the SKR PRO
with Serial3 (UART PORT) it wont compile.

BastR, 2019-10-09:
I will receive a SKR PRO with TFT35 3.0 LCD which uses the EXP1, EXP2 and TFT port, so I guess I can't use serial port 1 for MMU2, I need Serial port 3 or other port maybe?

Do you try to compile with serial port 6 which is located on ESP8266 pins ?

[image]

boelle, 2019-10-12:
@TB1405 issue fixed?

boelle, 2019-10-12:
@TB1405 so you have both lcd and mmu2 working?

TB1405, 2019-10-12:
> @TB1405 so you have both lcd and mmu2 working?

Yes it works but  while filament change the Display freezes.
And it works only with Serial1

boelle, 2019-10-12:
i would call that a workaround at best

let's keep this open, if you know what is wrong we can maybe cook up a real fix and submit a PR

TB1405, 2019-10-12:
Yes that Sounds good. If i know something i will also Report here. And some Info for people they would also use the mmu2 on a mk3s clone. You should know that actual the mmu2 support in marlin doesnt support the mmu2s SW so you have to calibrate the the ptfe Tube length

boelle, 2019-10-24:
@TB1405 any idea how to fix this one?

TB1405, 2019-10-25:
> @TB1405 any idea how to fix this one?

Wich problem do you mean? 
that Serial3 doesnt work or mmu2s support or display freezes ?

boelle, 2019-10-25:
all 3

TB1405, 2019-10-25:
Not really am i not really firm with programming. 
But the main problem with the compiling is solved. (it only works with Serial1)
So if u want you could close the this. 
For the display freez i could open a new issue
and for mmu2s support i opened a FR.

boelle, 2019-10-25:
opening a new issue would be ideal as it would be more focused and fresh :-D
but check that an issue has not been created byt another user

sounds like a plan that is ok with you?

TB1405, 2019-10-25:
Yes Sounds good

closed 2019-10-25 by TB1405

Minims, 2020-01-18:
> I will receive a SKR PRO with TFT35 3.0 LCD which uses the EXP1, EXP2 and TFT port, so I guess I can't use serial port 1 for MMU2, I need Serial port 3 or other port maybe?
> 
> Do you try to compile with serial port 6 which is located on ESP8266 pins ?
> 
> [image]

@BastR Did you succeed with mmu2 on skr pro on serial 3 ou 6 ?  Many Thanks.

BastR, 2020-01-19:
@Minims Non je n'ai pas essayé. Attention toutefois, si tu utilises Octoprint + le TFT35, je ne suis pas sûr que tu puisses déclarer un 3ième port série, il me semble avoir lu que Marlin limitait l'utilisation à 2 ports, par contre si tu n'utilises pas Octoprint alors tu es tranquille et tu pourras surement compiler avec en utilisant le RX et TX de l'ESP

Minims, 2020-01-19:
@BastR J'ai finalement réussi cet après midi. J'utilise le RX/TX3 de l'UART pour le MMU. Le TFT en RX/TX1 , l'ESP  connecté sur le TFT. J'ai donc Serial 1 et -1. Donc j'arrive a avoir MMU2 + TFT + ESP + USB de fonctionnel. J'ai pas pu mettre l'ESP en RX/TX6 car effectivement pas de possibilité de 3 e port.
Il me reste le BUG de freeze LCD précisé ici, mais ca fonctionne :-) 
https://github.com/MarlinFirmware/Marlin/issues/16495#issuecomment-576042471