MarlinFirmware/Marlin #15247: [BUG] (#define AUTO_BED_LEVELING_LINEAR -> error: 'Z_VALUES' was not declared in this scope)

wolfgangmauer, 2019-09-13:
Hi,
i tried bugfix-2.0 with my DIY printer
I have a very flat bed(Glas) and tryed:

#define AUTO_BED_LEVELING_LINEAR

and got this at compile time

Marlin\src\lcd\menu\menu_bed_leveling.cpp:216:67: error: 'Z_VALUES' was not declared in this scope

tvixen, 2019-09-13:
Upload your config files, and we have a chance to help you.

wolfgangmauer, 2019-09-13:
Sorry, PlatformIO with a SKR 1.3

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3609985/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3609986/Configuration_adv.h.txt)

tvixen, 2019-09-17:
Hmm there's SO much wrong in these config files, that i suggest you start all over again. 
Grab a new version of Marlin, and please try find your 3Dprinter (or a close match) in the example files provided by the Marlin team.

Open the both versions and compare them in two windows. And copy only you pid's and the config for the SKR 1.3 board. 
When adding Bed_Leveling, you have to check ALL the parameters for your sensor. Not just add the 
#define AUTO_BED_LEVELING_LINEAR
Its a complex setup, and it has to fit ...perfect.

closed 2019-09-19 by wolfgangmauer