MarlinFirmware/Marlin #15250: Marlin fails to compile will EEPROM_Settings enabled

Astr0kreep, 2019-09-13:
### Description

Marlin fails to compile will EEPROM_Settings enabled. Failure points to configuration_store.cpp and the points to blacklash.distance_mm as being the cause of the error. Added a snip of the errors in VS Code. For what it's worth I'm compiling this for a SKR1.1 board with TCM2208 drivers.

### Steps to Reproduce

1. Uncomment #define EEPROM_SETTINGS
2. Compile and wait
3. Fail

**Expected behavior:** [What you expect to happen]

I expected Marlin to compile and give me EEPROM functionality

**Actual behavior:** [What actually happens]
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3610923/Configuration_adv.zip)

It failed to compile the firmware

#### Additional Information

Commenting out #define BACKLASH_COMPENSATION seems to work as a workaround. In my attached files you should see it commented out. Uncomment for glorious failure. I have tried this on multiple machines too.

[image]

rflulling, 2019-09-15:
I use the EEPROM function and so far it's able to compile fine.  Not to say that there is not something unseen changed on a more recent build...

I did locate this for you. 
`#define E0_DRIVER_TYPE TTMC2208_STANDALONE.` / / Seems to contain a typo.

You use a lot of custom set up functions even adding some code that is not in the config by default. I wonder if that is some how conflicting with backlash code. Keep in mind that it may require several things turned on and configured to work. 

I used Backlash on M3D printers and it worked great. But it also has side effects. So use if you require. But expect that there is a trade off for it's function. -Its been a couple years and I dont recall the trade off... But it does help make better circles. Really the backlash needs to be something built into our slicers so that it only effects specific behavior. Backlash in the Firmware effects ALL behavior.

Astr0kreep, 2019-09-15:
Thanks for having a look. I think the only custom code is commented out I think. It comes from a version of marlin that somebody else put together for the printer I have. I have since upgraded the board.

I turned backlash off as you can probably see and that allowed me to compile it. I wasn't really even aware of it being there, to be honest, so it's of no loss to me. Well spotted on the typo. I just corrected it thank you.

tvixen, 2019-09-17:
Oki, so its not really a bug.
If it solve your problem, please close.
Thanks in advance.

Astr0kreep, 2019-09-17:
Well no, tvixen, it is still a bug. If you picked up the firmware and enabled EEPROM_Settings your compile WILL fail. The issue has not actually been resolved and the bug is still present (unless stated otherwise by developers). I had to use a workaround to get it to work. That is not the same as fixing the bug. This is especially problematic if you had no idea the function that breaks it even exists in the first place.

tvixen, 2019-09-17:
You are right. Even in 2.1.x dev, enabling this function will give a compile error.
[image]

rflulling, 2019-09-17:
I will offer to port your config to fresh file tonight. I will try to I'd
the error and add to this if I can.

Astr0kreep, 2019-09-17:
Thanks for taking the time to look at this issue. That's pretty much what I saw when I went to compile the firmware. Auto bed leveling has been a huge pain from the word go, starting with this issue. But that's a story for a different forum.

rflulling, 2019-09-18:
Carried it over to a clean build. Got more errors than I can shake a stick
at. I need to compile in 32bit.
Got VS and PIO installed. Still haven't figured out how to run the test
compile... lol
-If I can get over a learning hump. I will keep digging.

Astr0kreep, 2019-09-18:
Can I assist you in some way rflulling? I'm relatively new to all of this but I have VS Code and PIO installed and working. I can compile the firmware and all of that.

rflulling, 2019-09-18:
I just need to get the hand of using VS to compile so I can test the config files in 32bit. My own hardware is AT Mega 2560, so Arduino IDE works just fine.

sjasonsmith, 2019-09-19:
You can compile if you enable BACKLASH_GCODE along with BACKLASH_COMPENSATION. There are some errors in the EEPROM logic when they are not enabled together.

sjasonsmith, 2019-09-20:
@Astr0kreep, can you try this with the latest bugfix-2.0.x now that #15307  has been merged?

Astr0kreep, 2019-09-20:
Apologies for the late reply. Got a little tied up yesterday. I can try that this afternoon. I need to recompile anyway so I may as well do it with fresh files.

Astr0kreep, 2019-09-23:
So I've finally been able to test this again. I downloaded a fresh copy of Marlin and enabled all the same stuff I previously did and didn't comment out backlash. It successfully compiled first time. It looks like it's been fixed to me. Happy compiling and thanks for the assistance everyone!

tvixen, 2019-09-23:
Great.

sjasonsmith, 2019-09-26:
Thanks for trying it out @Astr0kreep! Could you close the issue?

Astr0kreep, 2019-09-26:
Done! Thanks once more.

closed 2019-09-26 by Astr0kreep