MarlinFirmware/Marlin #15259: skr 1.3 sd won't mount on mac

cinealfa, 2019-09-14:
I am unable to get the SD card to mount I can see the card on the display and I can release it but it doesn't appear on my mac.  I am not sure what commit broke it because I was dealing with multple extrusion is and the ability to mount the SD card was not uppermost in my mind
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3612529/Archive.zip)
I have read this thread:
#14318 But I didn't get that a solution had been arrived at.

tvixen, 2019-09-17:
I suggest you try it on a Windows pc, as there can be issues with the drivers for Mac.

cinealfa, 2019-09-18:
that doesn't really solve the problem. I can check it on a my pc  but it used to work on a mac and SHOULD work on a mac.

gloomyandy, 2019-09-18:
So what has changed between it working in the past and now? Go back to whatever you had and see if it still works. If it does check to see if there have been any configuration changes (the shared SD card config was changed some time ago, your old working system may have been before then and the config may need updating). Then work your way forward until you narrow down what change makes it stop working. 

Trying a PC would also help as it would indicate if it is a Mac specific issue, or a more general one (though if it was a general problem I suspect we would have been seeing a lot of reports about it).

cinealfa, 2019-09-19:
see that's the thing, Im not sure when it worked last . I went down the rabbit hole due to so cascade failures on the board and while I was trying to fix other things it stopped working. I can hook up one of the pcs but I prefer to have the ender on the mac in my office and not have yet another thing hooked up to the perpetually buggy and enormously heavy lapzilla. Better yet one of the octopis. I have a mac mini 4core that I do all build on and sole for cad and slicing and then send everything to the pi. The only for the sd access is new builds and troubleshooting octoprint. I have tried to keep my configs current with the builds but that is a time consuming process to manually up date. (unless you know a trick I don't)

tvixen, 2019-09-19:
> that doesn't really solve the problem. I can check it on a my pc but it used to work on a mac and SHOULD work on a mac.

No, but you will see if the error is within the code, or it's on your Mac. 

(Btw, you didn't mentioned that it was working before. That is pretty vital info, and one of the reasons why you should use the Q/A form)

cinealfa, 2019-09-20:
> No, but you will see if the error is within the code, or it's on your Mac.
>
> (Btw, you didn't mentioned that it was working before. That is pretty vital info, and one of the reasons why you should [...]

Point taken, I will do that in the future. As far as whether or not it works on a PC it in fact does not. It sort of mounts. That is to say the generic removable drive icon appears but when you click on it windows exhorts you to insert a drive. To further slow down matters, the no the build environment doesn't recognize the sd card when doing a build. As documented in another bug report.

ktgrach, 2019-09-21:
Remove the SD card from the SKR board and put it in a reader. See if the computer can read the SD. I have been chasing a problem on my build so I have been doing alot of writes to the SD card and the one that came with mine SKR died. I also had the problem with a card getting corrupted had to reformat the card (remember FAT 32) and manually copy a firmware file to the SD card then reboot. That also fixed my problem.

gloomyandy, 2019-09-21:
You should also try a new SD card. Some cards just don't seem to work very well. In particular the cards that come "free" with the board often seem to cause problems. When you switch from Marlin to using USB access the card is reinitialised (at a different speed) and this seems to cause problems with that "free" small/slow card.

tvixen, 2019-09-21:
Yes, as many has pointed out, the SD card's from China is pure junk.  Grap one from you camera or pi  and try with that one first.

cinealfa, 2019-09-21:
ok Ill give it a try and fwiw it does mount fine in a reader and it did work on previous builds (I just don't know which one)

cinealfa, 2019-09-21:
Well once again the hive made has humbled me. Put a new card in and it worked, mystery solved! Even though it used to mount just fine obviously cheep memory cells probably wore out. San Disk saves the day. Mabye all the writing to the card due to the power outage recovery borked it idk. But for now Im shutting it off.

closed 2019-09-21 by cinealfa