MarlinFirmware/Marlin #15261: [BUG] LED Control Menu

kaisertheone, 2019-09-14:
It is my first bug on Marlin 2.0 and Github ever.

got a compile error if i use the LED menu.
..\lcd\menu\menu_led row 75 there is something written about a MSG_SET_LEDS_DEFAULT but it was never declared anywehre.

I found it in the leds.h but something seems to not work correctly.
uncommenting the row solved the error.

shitcreek, 2019-09-16:
`MSG_SET_LEDS_DEFAULT` should be in the language file. Which language are you using?

kaisertheone, 2019-09-18:
Just the normal english language.

shitcreek, 2019-09-18:
Which version of Marlin are you using? Check your `language_en.h` file if it's missing `MSG_SET_LEDS_DEFAULT`. I was able to compile the latest Marlin vanilla with `LED_CONTROL_MENU` without any issue.

boelle, 2019-10-12:
@kaisertheone solved this one?

kaisertheone, 2019-10-12:
Well i Found a workaround which is ok for me.(just commented the Line)
But the default entry is still not correct in the source.

boelle, 2019-10-12:
Which version of Marlin are you using?

boelle, 2019-10-21:
will close this one as it seems @kaisertheone dont have an issue anymore

if you have a fix for the issue please tell in detail

if its simple enough i could make a PR that fixes it

closed 2019-10-21 by boelle