MarlinFirmware/Marlin #15262: [BUG] Dual Extruder Retraction

smg201smg, 2019-09-14:
I use dual extruder (2 in 2 out). Only i have a problem.

For example:  second extruder retracts 7mm, and first extruder prints at a layer.
New turn for second extruder: It extrudes 5mm before printing.  (Yes i cannot tell good)

*So it retracts 7mm but it extrudes 5mm. It causes deformation at print*

How can i solve this problem? Please help me

shitcreek, 2019-09-16:
My guess based on limited information: check your slicer retraction settings for the two extruders. For example with slic3r, there is a setting called `Extra length on restart` which if supplied with a negative amount would un-retract the difference.
That said, you must provide more information such as your configuration files otherwise we don't have much else to go off on. And please follow the template. But before you do that, are you sure this is a bug and not a configuration issue as we do not provide tech support here.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

smg201smg, 2019-09-17:
OK thank you

boelle, 2019-10-12:
@smg201smg solved this one?

boelle, 2019-10-21:
i will close this one as solved

we can always reopen if that is not the case

closed 2019-10-21 by boelle