MarlinFirmware/Marlin #15263: [FR] Send data like Coordinates to Serial Port #2

TheWeakBerlin, 2019-09-14:
Hello

i want to set up a TFT Display with an Custom GUI, i do this in the Firmware of the Display but i want to show Temps, Coordinates, Print States and stuff like that on the Display. 

I dont want to Control the Printer with the Display, i still want to use Ocotprint for that. 

Actually i can set up the Display and do some Graphical stuff on it like print strings and bmp´s, but im unable to display any other data than Temps. I search around the Code of Marlin but i dont find a possebility to get additionial Data on the Display.

Whats about to send Data via the secondary serial port to the Display to maka Custom Print Screens like the Temp response Message.

Thank You

MarkMan0, 2019-09-15:
[M114](http://marlinfw.org/docs/gcode/M114.html)?

TheWeakBerlin, 2019-09-15:
Correct with M114 u will get the coordinates, but it will send to Serial port 1 but i want them on Serial port 2 and i dont think its a good idea to send permanently M114 during printing.

MarkMan0, 2019-09-15:
Oh. I assumed it sent back the data to where it came from. So you would like something similar to Temperature Auto-report, but for position, right?

TheWeakBerlin, 2019-09-15:
Yes, i found the point where the Temp oute-report is build but, but i dont want to play to much with printer firmware. When the Display brake i go ahead with old one, shit happens. But i dont want to print with my old mainboard so im carefull with marlin.

TheWeakBerlin, 2019-09-16:
I think a made it, but its a dirty solution i think.

in temperature.cpp i added "report_current_position()" at the auto report temperatures section
and in stepper.cpp i added "PORT_REDIRECT(SERIAL_BOTH)" in the Stepper::report_position() function

in the display firmware where the temp is parsed i just look for the "X:" and get the values, so i get them on my Display. 

I have to test now if this will affect the printer when hes working, and if not will look if i can reduce the intervall the coordinates will send.

but for now it works for me and im happy, but i will look for an better solution ........