MarlinFirmware/Marlin #15265: [BUG] SD card insertion is no longer auto-detected

salacpavel, 2019-09-14:
## Description

Since aboat a week ago, SD card insert event is no longer autodetected and card must be manually initialized from Marlin menu, in order to be able to access files on it.

### Steps to Reproduce

Insert SD valid card with printable conents on it.

**Expected behavior:** "Media inserted" message should appear on screen and SD card conetnts should be directly accessible form "Browse media files" menu.

**Actual behavior:** No notification on screen. "Browse media files" menu is empty with only "Refresh" item available, untill I select "Init. media" from main menu manually.

#### Additional Information

HW: Bigtreetech SKR mini /DIP with 4xTMC2209 UART, CR10 stock display

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3613037/Configuration_adv.zip)

tpruvot, 2019-09-15:
related to #15255 explanations inside...

salacpavel, 2019-09-15:
Awesome! I have read the discussion bellow, but I am a bit lost here - has your commit been merged yet, please?

salacpavel, 2019-09-15:
Oh by the way, I just had an idea: would not it be cool to have (configurable) option to automatically jump to "Print from media" submenu and display file list on display, when inserted card is recognized and mounted?

thinkyhead, 2019-09-15:
> when inserted card is recognized and mounted?

Yes! But only when manually inserted.

closed 2019-09-15 by thinkyhead