MarlinFirmware/Marlin #15268: [BUG] BigtreeTech SKR 1.3 3DTouch (Geeetech) not working

john4242, 2019-09-15:
### Description

I have an SKR 1.3 board and a 3DTouch (Geeetech "original") sensor. I have the same type of sensor working with Marlin on a GT2560 A+ board, although not the same piece.

What works:
- The servo part and the 3D Touch control is working well: I can stow and unstow with M commands, pins move, it reacts to sensing (pulls the pin back). The wiring for the servo is easy: there is a servo connector on the SKR 1.3 board, I just had to make sure the pin order was correct (+5V, GND, Servo).

- However, I have just struggled with the actual sensing. It's not getting triggered. I compiled Marlin with the PIN debug option, and there is actually one constellation which "almost" works:
-- I plugged the touch sensors into Z MIN (black to GND, middle pin, white to signal pin)
-- Z_MIN_ENDSTOP_INVERTING = false // as default
-- ENDSTOPPULLUP_ZMIN_PROBE defined
-- ENDSTOPPULLUP_ZMIN defined

 I have tried inverting as true and also as false (the true is tricky because BLTOUCH define overrides it in another file, but I removed that override).

I got as far as:
- If inverting = true, then M43 S shows:
```
Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PIN: 57
. Z_MIN_ENDSTOP_INVERTING: true
. Check for BLTOUCH
. Deploy & stow 4 times
WARNING: INVERTING setting probably backwards.
** Please trigger probe within 30 sec **
. Pulse width (+/- 4ms): 6
FAIL: Noise detected - please re-run test
ok
```

If inverting = false, then M43 S returns:
```
Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PIN: 57
. Z_MIN_ENDSTOP_INVERTING: false
. Check for BLTOUCH
= BLTouch Classic 1.2, 1.3, Smart 1.0, 2.0, 2.2, 3.0, 3.1 detected.
** Please trigger probe within 30 sec **
. Pulse width (+/- 4ms): 4
FAIL: Noise detected - please re-run test
ok
```

 The interesting bit as that it is not noise what it's detecting, that's my actual touch detected.
 I'm starting to be out of ideas, spent hours on the net googling, got some tips, none worked.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3613796/Configuration.zip)

john4242, 2019-09-15:
I also tried to move it to the ZMAX pin, and repeat the tests.
It is close to working, but with that "noise detected" error.
```
Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PROBE_PIN: 1336
. Z_MIN_PROBE_ENDSTOP_INVERTING: true
. Check for BLTOUCH
. Deploy & stow 4 times
WARNING: INVERTING setting probably backwards.
** Please trigger probe within 30 sec **
. Pulse width (+/- 4ms): 6
FAIL: Noise detected - please re-run test
ok
Reporting endstop status
x_min: open
y_min: open
z_min: TRIGGERED
z_probe: open
ok
Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PROBE_PIN: 1336
. Z_MIN_PROBE_ENDSTOP_INVERTING: true
. Check for BLTOUCH
. Deploy & stow 4 times
WARNING: INVERTING setting probably backwards.
** Please trigger probe within 30 sec **
. Pulse width (+/- 4ms): 4
FAIL: Noise detected - please re-run test
ok
PIN: 0.00        E1_DIR_PIN                             Output = 1
PIN: 0.01        E1_STEP_PIN                            Input  = 1
PIN: 0.02 (A 7)  <unused/unknown>   Analog in =  1023   Input  = 1                       
PIN: 0.03 (A 6)  <unused/unknown>   Analog in =     0   Input  = 1                       
PIN: 0.04        <unused/unknown>                       Input  = 1                       
PIN: 0.05        <unused/unknown>                       Input  = 1                       
PIN: 0.06        <unused/unknown>                       Output = 1                       
PIN: 0.07        <unused/unknown>                       Output = 0                       
PIN: 0.08        <unused/unknown>                       Input  = 1                       
PIN: 0.09        <unused/unknown>                       Output = 1                       
PIN: 0.10        E1_ENABLE_PIN                          Output = 1
PIN: 0.11        E0_DIR_PIN                             protected 
PIN: 0.15        SCK_PIN                                Input  = 1
PIN: 0.16        SDSS                                   Output = 1
.                SS_PIN                                 Output = 1
PIN: 0.17        MISO_PIN                               Input  = 1
PIN: 0.18        MOSI_PIN                               Input  = 1
PIN: 0.19        Y_STEP_PIN                             protected 
PIN: 0.20        Y_DIR_PIN                              protected 
PIN: 0.21        Z_ENABLE_PIN                           protected 
PIN: 0.22        Z_STEP_PIN                             protected 
PIN: 0.23 (A 0)  <unused/unknown>   Analog in =     0   Input  = 1                       
PIN: 0.24 (A 1)  <unused/unknown>   Analog in =  1023   Input  = 0                       
PIN: 0.25 (A 2)  <unused/unknown>   Analog in =     0   Input  = 1                       
PIN: 0.26 (A 3)  <unused/unknown>   Analog in =     0   Input  = 1                       
PIN: 0.27        <unused/unknown>                       Input  = 0                       
PIN: 0.28        <unused/unknown>                       Input  = 1                       
PIN: 0.29        <unused/unknown>                       Input  = 1                       
PIN: 0.30        <unused/unknown>                       Input  = 0                       
PIN: 1.00        <unused/unknown>                       Input  = 1                       
PIN: 1.01        E1_CS_PIN                              Input  = 1
.                Z2_SERIAL_RX_PIN                       Input  = 1
.                E1_SERIAL_RX_PIN                       Input  = 1
PIN: 1.04        Z2_SERIAL_TX_PIN                       Input  = 1
.                E1_SERIAL_TX_PIN                       Input  = 1
PIN: 1.08        E0_CS_PIN                              Input  = 1
.                E0_SERIAL_RX_PIN                       Input  = 1
PIN: 1.09        E0_SERIAL_TX_PIN                       Output = 1
PIN: 1.10        Z_CS_PIN                               Input  = 1
[... 19 lines cut ...]
PIN: 1.28        FIL_RUNOUT_PIN                         Input  = 1
PIN: 1.29        X_MIN_PIN                              protected 
.                X_STOP_PIN                             protected 
PIN: 1.30 (A 4)  <unused/unknown>   Analog in =     0   Input  = 1                       
PIN: 1.31 (A 5)  <unused/unknown>   Analog in =     0   Input  = 1                       
PIN: 2.00        SERVO0_PIN                             Output = 0
PIN: 2.01        X_ENABLE_PIN                           protected 
PIN: 2.02        X_STEP_PIN                             protected 
PIN: 2.03        FAN_PIN                                protected 
PIN: 2.04        FAN1_PIN                               protected 
PIN: 2.05        HEATER_BED_PIN                         protected 
PIN: 2.06        X_DIR_PIN                              protected 
PIN: 2.07        HEATER_0_PIN                           protected 
PIN: 2.08        Y_ENABLE_PIN                           protected 
PIN: 2.09        <unused/unknown>                       Input  = 0                       
PIN: 2.10        <unused/unknown>                       Input  = 1                       
PIN: 2.11        Z_DIR_PIN                              protected 
PIN: 2.12        E0_ENABLE_PIN                          protected 
PIN: 2.13        E0_STEP_PIN                            protected 
PIN: 3.25        <unused/unknown>                       Input  = 1                       
PIN: 3.26        <unused/unknown>                       Input  = 1                       
PIN: 4.28        <unused/unknown>                       Output = 0                       
PIN: 4.29        X_SERIAL_TX_PIN                        Output = 1
ok
```

asko58, 2019-09-15:
> However, I have just struggled with the actual sensing. It's not getting triggered. I compiled Marlin with the PIN debug option, and there is actually one constellation which "almost" works:
-- I plugged the touch sensors into Z MIN (black to GND, middle pin, white to signal pin)
-- Z_MIN_ENDSTOP_INVERTING = false // as default
-- ENDSTOPPULLUP_ZMIN_PROBE defined
-- ENDSTOPPULLUP_ZMIN defined

I have the same board and probe and for me the configuration is:
-- ENDSTOPPULLUP_ZMIN_PROBE defined //pullups for all endstops generally defined
-- ENDSTOPPULLUP_ZMIN defined
-- Z_MIN_ENDSTOP_INVERTING = true // !!
-- Black wire to ground, white to switch. (it is pulled high, so ground when endstop hit)

I saw that you redefine the pullups manually while pullups have already been enabled for all your endstops. I don't think it hurts but I would get rid of it.

wolfgangmauer, 2019-09-15:
I have the same board and probe.
Works perfect for me!
I think this is not a marlin issue ;-)

john4242, 2019-09-15:
For those of you where it works, if you have a bit of time, could you do a test?
Enable in Configuration_adv.h
`#define PINS_DEBUGGING`

Then reflash firmware, and do the tests:
`M43 S`
and
`M43`

I am almost suspecting that it's just the test now that's failing for me? Since it does react to a signal during the test, just says "noise detected". But that's impossible.
I am using the 2.0 bugfix version, ID=e604f76703165027a22dc33c7875f4858e9f1c80 date 2019-08-22, maybe I will try with a newer/older version.

wolfgangmauer, 2019-09-15:
Sorry, a have a 3-Day print running…

john4242, 2019-09-16:
Okay, I have actually another piece of Geeetech 3DTouch. I wired it up, and... got the same results:
(Keep in mind this other one is working every day in another printer, with a GT2560 board)
```
Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PROBE_PIN: 1336
. Z_MIN_PROBE_ENDSTOP_INVERTING: true
. Check for BLTOUCH
. Deploy & stow 4 times
WARNING: INVERTING setting probably backwards.
** Please trigger probe within 30 sec **
. Pulse width (+/- 4ms): 4
FAIL: Noise detected - please re-run test
ok
```

john4242, 2019-09-16:
From the source code (src/gcode/config/M43.cpp) it seems that the pulse width is really that short (about 4 msec), instead of the expected 15ms or more. I wonder if this is because of 3.3V (?).

john4242, 2019-09-19:
Okay, I think this is not a Marlin issue, so closing.
I was able to use Arduino (nano) to create a separate test circuit, and it confirmed that the sensor sends a HIGH (logical 1) signal for 4-6 milliseconds, so it's much shorter than the original 3D Touch. I guess that's the reason for the problem. I am guessing Marlin doesn't want to support such short signals because it considers it to be noise and unreliable. I might add some electronics to make the 1 signal longer.

closed 2019-09-19 by john4242

john4242, 2019-09-19:
Just one last comment: I created a simple Arduino circuit to 'make the pulse longer', that is, 20 milliseconds, and it appears to be accepted.
```
Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PROBE_PIN: 1336
. Z_MIN_PROBE_ENDSTOP_INVERTING: true
. Check for BLTOUCH
. Deploy & stow 4 times
WARNING: INVERTING setting probably backwards.
** Please trigger probe within 30 sec **
. Pulse width (+/- 4ms): 20
= BLTouch pre V3.1 (or compatible) detected.
ok
```

john4242, 2019-09-22:
Okay, it is working now. Note that "M43 S" servo test command is misleading. 3DTouch from Geeetech fires a 5ms short pulse, this is what Antclabs did long ago (until they realized it's too short), but the clones copied it. I used a separate PROBE pin - Z Max, so config and tests with happened with that.
Note that if you do an "M119" - endstop test, it will show the probe as TRIGGERED - when stowed, normally. That is OK! It might be misleading, but it's okay. Only when the probe is deployed and set to sense will it display "open".

So the correct settings:
Configuration.h
```
#define BLTOUCH
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // No need to invert, leave this as config default
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN - commented out because I am using ZMAX pin
#define Z_MIN_PROBE_PIN P1_24 // the z max pin - SKR1.3 board
//#define ENDSTOPPULLUP_ZMIN_PROBE - this is to be commented out, though maybe not important
#define AUTO_BED_LEVELING_BILINEAR - I use this one
#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command - 3DTouch
```

Configuration_adv.h
```
#define BLTOUCH_DELAY 1000 // probably needed, so we don't get errors
//#define BLTOUCH_FORCE_SW_MODE - leave this commented out, or you will have errors
// generally leave all other BLTOUCH bits commented out
```

wolfgangmauer, 2019-09-22:
You don’t have to define the Z_MIN_PROBE_PIN, its correct by default!

 

pins_BIGTREE_SKR_V1.3.h

//

// Z Probe (when not Z_MIN_PIN)

//

#ifndef Z_MIN_PROBE_PIN

  #define Z_MIN_PROBE_PIN  P1_24

#endif

anm8tor, 2019-09-24:
So I am having what appears to be the same issue on an SKR Mini V1.1
After making the changes @john4242 suggested, this is what I get on an "M43 S" command:

21:07:29.336 : Servo probe test
21:07:29.336 : . using index:  0, deploy angle: 10, stow angle:   90
21:07:29.336 : . Probe Z_MIN_PROBE_PIN: 2
21:07:29.336 : . Z_MIN_PROBE_ENDSTOP_INVERTING: false
21:07:29.336 : . Check for BLTOUCH
21:07:30.687 : . Deploy & stow 4 times
21:07:35.638 : ** Please trigger probe within 30 sec **
21:07:35.650 : . Pulse width (+/- 4ms): 2
21:07:35.650 : FAIL: Noise detected - please re-run test

I also noticed that before and after the changes, the "Z_MIN_PROBE_ENDSTOP_INVERTING" line stated false (meaning that maybe it is being overridden somewhere else?) Any ideas?

john4242, 2019-09-24:
The "FAIL: Noise detected - please re-run test" doesn't mean anything for clones.
The original ANTClabs BLTouch emits a 15 millisecond long signal, whereas clones only emit a shorter one (usually 5 milliseconds). It will still work, take this just as a warning (might be less reliable of course).
Is your touch probe from Geeetech? If it's Geeetech then you should not touch Z_MIN_PROBE_ENDSTOP_INVERTING (leave it as false).

Even if you try to set Z_MIN_PROBE_ENDSTOP_INVERTING to true, it will be overridden in another file, forgot which one, but you can do a search in all *.h files. If "BLTOUCH" is set, that file will override it to false always.

anm8tor, 2019-09-24:
@john4242 any thoughts on why I would experience a successful safe homing, but a failed probing after the first touch after making the adjustments then?

john4242, 2019-09-24:
> @john4242 any thoughts on why I would experience a successful safe homing, but a failed probing after the first touch after making the adjustments then?

Yes. Probably because you didn't set the servo timeout. That is:
#define SERVO_DELAY { 750 }

If you enable the debug trace, you will see that after the first touch it tries to stow (and physically succeeds), but the firmware still thinks the "stow" failed.

To enable trace:
- #define DEBUG_LEVELING_FEATURE
- Issue the command: M111 S32
  This will make Marlin output a lot of trace data about probing. That might help.

anm8tor, 2019-09-25:
@john4242 , Here's what M111 S32 reported:
22:01:18.445 : current_position= X-28.00 Y20.00 Z10.00 : >>> G29
22:01:18.445 : Machine Type: Cartesian
22:01:18.445 : Probe: BLTOUCH
22:01:18.445 : Probe Offset X:58 Y:10 Z:0.00 (Right-Back & Same Z as Nozzle)
22:01:18.445 : Auto Bed Leveling: BILINEAR
22:01:18.445 : (disabled)
22:01:18.445 : current_position= X-28.00 Y20.00 Z10.00 : set_probe_deployed
22:01:18.445 : deploy: 1
22:01:18.445 : do_probe_raise(10.00)
22:01:18.445 : >>> do_blocking_move_to X-28.00 Y20_blocking_move_to
22:01:18.450 : reset_bed_level
22:01:18.450 : >>> probe_pt(25.00, 15.00, raise, 0, probe_relative)
22:01:18.450 : current_position= X-28.00 Y20.00 Z10.00 :
22:01:18.450 : >>> do_blocking_move_to X-33.00 Y5.00 Z10.00
22:01:18.450 : <<< do_blocking_move_to
22:01:18.795 : current_position= X-33.00 Y5.00 Z10.00 : set_probe_deployed
22:01:18.795 : deploy: 1
22:01:18.795 : current_position= X-33.00 Y5.00 Z10.00 : >>> run_z_probe
22:01:18.795 : current_position= X-33.00 Y5.00 Z10.00 : >>> do_probe_move
22:01:18.795 : BLTouch DEPLOY requested
22:01:18.795 : BLTouch Command :10
22:01:19.595 : bltouch.deploy_proc() end
22:01:19.795 : >>> do_blocking_move_to X-33.00 Y5.00 Z-2.00
22:01:19.795 : <<< do_blocking_move_to
22:01:22.345 : current_position= X-33.00 Y5.00 Z-2.00 : sync_plan_position
22:01:22.345 : current_position= X-33.00 Y5.00 Z-2.00 : <<< do_probe_move
22:01:22.345 : SLOW Probe fail!
22:01:22.345 : current_position= X-33.00 Y5.00 Z-2.00 : <<< run_z_probe
22:01:22.345 : >>> do_blocking_move_to X-33.00 Y5.00 Z3.00
22:01:22.345 : <<< do_blocking_move_to
22:01:23.140 : current_position= X-33.00 Y5.00 Z3.00 : set_probe_deployed
22:01:23.140 : deploy: 0
22:01:23.140 : do_probe_raise(10.00)
22:01:23.140 : >>> do_blocking_move_to X-33.00 Y5.00 Z10.00
22:01:23.140 : <<< do_blocking_move_to
22:01:24.185 : >>> do_blocking_move_to X-33.00 Y5.00 Z10.00
22:01:24.185 : <<< do_blocking_move_to
22:01:24.185 : Error:Probing Failed
22:01:24.185 : <<< probe_pt
22:01:24.190 : current_position= X-33.00 Y5.00 Z10.00 : set_probe_deployed
22:01:24.190 : deploy: 0
22:01:24.190 : current_position= X-33.00 Y5.00 Z10.00 : > probing complete
22:01:24.190 : <<< G29
22:01:24.190 : X:-33.00 Y:5.00 Z:10.00 E:0.00 Count X:-3300 Y:500 Z:4000

dardan, 2019-10-30:
@john4242 Hi there, is your 3dtouch works on SKR1.3, if its working please can you share your Marlin firmware with me?
Thank you

wolfgangmauer, 2019-10-30:
I have the same hardware…

It works like a charm..

Well,

my configuration is very special -> DIY CoreXY.

My changes regarding 3dtouch are:

 

#define USE_ZMIN_PLUG

#define BLTOUCH

 

#define X_PROBE_OFFSET_FROM_EXTRUDER

#define Y_PROBE_OFFSET_FROM_EXTRUDER

#define Z_PROBE_OFFSET_FROM_EXTRUDER

 

#define XY_PROBE_SPEED 40*60

#define Z_PROBE_SPEED_FAST 2*60

 

#define AUTO_BED_LEVELING_BILINEAR

dardan, 2019-10-30:
@wolfgangmauer Thank you i have those what about your servo delay mine is #define SERVO_DELAY { 300 }