MarlinFirmware/Marlin #15269: [BUG] Can't compile with skr e3 dip

Bansh1990, 2019-09-15:
### Probably not a realy BUG

### May be I do something wrong?

BLtouch + USB multiple device + S Curve acc  + eeprom
not enough storage for firmware

AmiSMB, 2019-09-15:
Try enable SLIM_LCD_MENUS

delafer, 2019-09-16:
>> "Try enable SLIM_LCD_MENUS"
Are you joking? Because of Marlin developers have added a bunch of useless functions and libraries and other crappy stuff in the latest marlin versions,and don't even try to optimize project compiling size and it constantly growing without visible improvements you are recommending to disable useful features? I recommend replace rather Marlin 2.xx developers or take the older version. it has less shit and it is more stable.

delafer, 2019-09-16:
P.S. you are doing all in a right way,marlin developers doing something wrong. I have same issues with latest marlin versions.

AmiSMB, 2019-09-16:
I did a pull last night and tried disabling SLIM_LCD_MENUS and it compiled fine for my SKR E3 DIP. My branch head was at commit 20fbb751f1edd31f33335ce2d4aa26e44c40d524

sjasonsmith, 2019-09-16:
> what can I turn off to save space besides ARCsupport?

Do you need ARC support? Most slicers aren't capable of using it.

sjasonsmith, 2019-09-16:
> P.S. you are doing all in a right way,marlin developers doing something wrong. I have same issues with latest marlin versions. They are playing with joysticks,menus,animations,supports for 6,7,8,9,10 drivers instead to Fix open issues or to add anything really useful.

It's not fair to accuse contributors of working on unimportant details. They might not be important to you, but they are important for other people.

When work is done by volunteers, they work on the features that most interest them. I'm sure they would welcome you learning how to contribute to the areas which most interest you.

I see commits every day for these SKR E3 boards. They are very new and people are figuring out the best way to work with them. They are certainly not being ignored.

When it comes down to it, these have half the usable flash of an SKR 1.3, so you can't expect to be able to fit every possible feature into them.

Bansh1990, 2019-09-16:
> > what can I turn off to save space besides ARCsupport?
> 
> Do you need ARC support? Most slicers aren't capable of using it.

no? it already turned off. with enable SLIM_LCD_MENUS firmware is compiling, but i usefull thing for me. I'm want find another way)

sjasonsmith, 2019-09-16:
> > Do you need ARC support? Most slicers aren't capable of using it.
> 
> no. it already turned off. with enable SLIM_LCD_MENUS firmware is compiling, but it usefull thing for me. I'm want find another way)

Your configuration compiled fine for me. I actually had to _enable_ ARC support for it to fail and require slim menus.

Bansh1990, 2019-09-17:
> > > Do you need ARC support? Most slicers aren't capable of using it.
> > 
> > 
> > no. it already turned off. with enable SLIM_LCD_MENUS firmware is compiling, but it usefull thing for me. I'm want find another way)
> 
> Your configuration compiled fine for me. I actually had to _enable_ ARC support for it to fail and require slim menus.
> 
> I don't know exactly which commit I applied your configuration on top of, but it would have been the very latest state of bugfix-2.0.x whenever I tried it yesterday.

do you enable -DUSE_USB_COMPOSITE ?

AmiSMB, 2019-09-17:
I have -DUSE_USB_COMPOSITE in platformio.ini. I did a pull last night and compiled without SLIM_LCD_MENUS enabled. Here is my config with platformio.ini. I am unable to enable TMC_DEBUG but I have issue #15226 open as if I leave this enabled and manage to slim the firmware down enough to fit I have problems with the firmware booting up after the third power cycle.

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3619765/Archive.zip)

tpruvot, 2019-09-17:
consider USE_USB_COMPOSITE unstable and experimental, wont get a fix in the hour... i say that after a night of tries

Bansh1990, 2019-09-17:
> consider USE_USB_COMPOSITE unstable and experimental, wont get a fix in the hour... i say that after a night of tries. Even the simple usb direct serial is not really stable... for now

one more bug: when i store eeprom settings with enabled USE_USB_COMPOSITE, firmware dont want boot, but before, all boot fine

sjasonsmith, 2019-09-18:
> do you enable -DUSE_USB_COMPOSITE in platformio.ini ?
@Bansh1990, no I did not realize that was configured in the platform.io file. I had only selected the proper environment for the board, didn't change anything else.

TimoBre, 2019-09-29:
Hai Bansh,
does you fixed it?
Maybe the reason is an wrong MCU. There are C and E models out. And they are different.
I had also too less progmemory. I was able to double it, because i have the E MCU modell.
If you have the C MCU you have to make memory free, like in former days with the oringin Melzi boards.
Compare with https://github.com/MarlinFirmware/Marlin/issues/15254 maybe you have the same issue and you are able to solve it

boelle, 2019-10-12:
@Bansh1990 still having issues?

boelle, 2019-10-24:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-24 by boelle