MarlinFirmware/Marlin #15270: [Question] DEACTIVATE_SERVOS_AFTER_MOVE and BLTOUCH_DELAY

GMagician, 2019-09-15:
Not a real problem has been raised but I'm wondering if have been ever considered what happens to bltouch when both above define are defined.
BLTOUCH_DELAY will never act as expected or not?

shitcreek, 2019-09-16:
Defining `BLTOUCH_DELAY` in `configuration_adv.h` only overrides the default value of 500 as per `conditionals_post.h`:
```
#if ENABLED(BLTOUCH) && !defined(BLTOUCH_DELAY)
    #define BLTOUCH_DELAY 500
  #endif
```
In other words, `BLTOUCH_DELAY` is always in use when using `BLTOUCH`.
`DEACTIVATE_SERVOS_AFTER_MOVE` happens after the probing process and should have no effect on it.

closed 2019-09-16 by shitcreek

GMagician, 2019-09-16:
Are you sure? In stepper. Move is detached

shitcreek, 2019-09-16:
Upon further inspection, `DEACTIVATE_SERVOS_AFTER_MOVE` is disabled if `BLTOUCH` is used as per  `Conditionals_LCD.h`:
```
#if ENABLED(BLTOUCH)
  ...
  #undef DEACTIVATE_SERVOS_AFTER_MOVE
...
  #endif
```

GMagician, 2019-09-16:
I missed that...ok, thanks