MarlinFirmware/Marlin #15272: [BUG] BIGTREETECH SKR1.3 & TMS2130 , can't get read out MISO

KDU300, 2019-09-15:
Hi,
Have just made a Marlin2.0 update again. I noticed that I get a "TMS Connection Error". I then checked all the settings again, but it's all right. My printer is running properly, the current can also be adjusted. I looked at the SPI signals on the scope. CLK and MOSI are exelent, MISO is broken. For me, it looks like the output is popping on output. Also, I can not find in the code the place where the SPI interface is written. Here are the pictures. ... What can I do to continue debugging?

GESENDET: M122
LESEN: 		X	Y	Z	Z2
LESEN: Enabled		false	false	false	false
LESEN: Set current	1100	1100	1000	1000
LESEN: RMS current	581	581	550	550
LESEN: MAX current	819	819	776	776
LESEN: Run current	18/31	18/31	17/31	17/31
LESEN: Hold current	5/31	5/31	5/31	5/31
LESEN: CS actual	31/31	31/31	31/31	31/31
LESEN: PWM scale	255	255	255	255
LESEN: vsense		1=.18	1=.18	1=.18	1=.18
LESEN: stealthChop	true	true	true	true
LESEN: msteps		0	0	0	0
LESEN: tstep		4294967295	4294967295	4294967295	4294967295
LESEN: pwm
LESEN: threshold				
LESEN: [mm/s]					
LESEN: OT prewarn	true	true	true	true
LESEN: OT prewarn has
LESEN: been triggered	false	false	false	false
LESEN: off time	15	15	15	15
LESEN: blank time	54	54	54	54
LESEN: hysteresis
LESEN: -end		12	12	12	12
LESEN: -start		8	8	8	8
LESEN: Stallguard thrs	0	0	0	0
LESEN: DRVSTATUS	X	Y	Z	Z2
LESEN: stallguard		*	*	*	*
LESEN: sg_result	1023	1023	1023	1023
LESEN: fsactive		*	*	*	*
LESEN: stst		*	*	*	*
LESEN: olb		*	*	*	*
LESEN: ola		*	*	*	*
LESEN: s2gb		*	*	*	*
LESEN: s2ga		*	*	*	*
LESEN: otpw		*	*	*	*
LESEN: ot		*	*	*	*
LESEN: Driver registers:
LESEN: 		X	0xFF:FF:FF:FF	 Bad response!
LESEN: 		Y	0xFF:FF:FF:FF	 Bad response!
LESEN: 		Z	0xFF:FF:FF:FF	 Bad response!
LESEN: 		Z2	0xFF:FF:FF:FF	 Bad response!

[image]

Here, MISO is not connected to the SKR board. Here the signal looks good. There are also no pulls on the board, the signal goes through a jumper directly into the LPC1768. I also changed the Borad itself, the SPI error remained.

[image]

Regards 
Karsten

gloomyandy, 2019-09-15:
So did this work for you with a previous version of Marlin? If it did the most useful thing you can do is to step back and try and find the commit at which it stopped working.

teemuatlut, 2019-09-16:
Try downgrading the `TMCStepper` library version back to `0.4.6`.

thinkyhead, 2019-09-16:
> I can not find in the code the place where the SPI interface is written

You're not alone there. I need to Google it again every morning.

ReadieS, 2019-09-16:
So I have been having the same issue. I was under the impression it was me as this is my first install of Marlin 2.0 despite having years of installing with previous versions.
I have been getting TMC CONNECTION ERROR on startup and it will not allow me to do anything.
M122 reports
		X	Y	Z	E
Enabled		false	false	false	false
Set current	700	700	700	700
RMS current	673	673	673	673
MAX current	949	949	949	949
Run current	21/31	21/31	21/31	21/31
Hold current	10/31	10/31	10/31	10/31
CS actual	31/31	31/31	31/31	31/31
PWM scale	255	255	255	255
vsense		1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true
msteps		0	0	0	0
tstep		4294967295	4294967295	4294967295	4294967295
pwm
threshold	0	0	0	0
[mm/s]		-	-	-	-
OT prewarn	true	true	true	true
OT prewarn has
been triggered	false	false	false	false
off time	15	15	15	15
blank time	54	54	54	54
hysteresis
-end		12	12	12	12
-start		8	8	8	8
Stallguard thrs	0	0	0	0
DRVSTATUS	X	Y	Z	E
stallguard		*	*	*	*
sg_result	1023	1023	1023	1023
fsactive		*	*	*	*
stst		*	*	*	*
olb		*	*	*	*
ola		*	*	*	*
s2gb		*	*	*	*
s2ga		*	*	*	*
otpw		*	*	*	*
ot		*	*	*	*
Driver registers:
		X	0xFF:FF:FF:FF	 Bad response!
		Y	0xFF:FF:FF:FF	 Bad response!
		Z	0xFF:FF:FF:FF	 Bad response!
		E	0xFF:FF:FF:FF	 Bad response!
Testing X connection... Error: All HIGH
Testing Y connection... Error: All HIGH
Testing Z connection... Error: All HIGH
Testing E connection... Error: All HIGH

I will try the TMC library downgrade if I can find out where to change it and report back when I have a few minutes if no one else has done so already.

Thank you OP for the post and thank you for all the support above

ReadieS, 2019-09-16:
So I am not that familiar with the platformio environment yet (please forgive me if I am making rookie errors).
Using Atom I have gone to platformio home > Libraries > Installed Teamuatlut TMC 0.4.6 library and uninstalled the 0.5.0 library rebuilt and uploaded. (I presume this is the correct way to change the library?)
Its made no difference on my issue I also tried 0.4.5 nut this made no difference either.

Unfortunately I don't have the expertise or experience to test the SPI connection as the OP has done.
I am will to test anything that anyone suggests if it is within my capabilities.
I would really appreciate it if anyone who does come across a solution would keep this up to date.
Thanks.

teemuatlut, 2019-09-16:
The library and framework versions are listed at the start when you compile.

You need to edit the platformio.ini file as Marlin downloads the dependencies into the .pilibdeps folder. I think it does this even if you have a dependency installed already.
You should also delete this folder (or just TMCStepper_5513) to force re-downloading.

KDU300, 2019-09-16:
@teemuatlut 
So, I had a few minutes now. I tested the libs 0.4.5, 0.4.6 and 0.50, it was always the same behavior on the pin.

ReadieS, 2019-09-16:
> The library and framework versions are listed at the start when you compile.
>
> You need to edit the platformio.ini file as Marlin downloads the dependencies into the .pilibdeps folder. I think it [...]

I removed the TMCStepper_5513 folder and looked in the platformio.ini file but I wasn't confident on the correct change to make ( I presume it is todo with the "TMCStepper@<1.0.0" entry under common and LPC1768). On the dependencies list at the top of the build it is listing as 0.4.6 though so it seems to be using the library I installed locally.

As above still no effect on the issue.

I attach the build and upload log for reference:

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/
master.zip; framework: arduino; board: nxp_lpc1768)
-------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc176
8.html
PLATFORM: NXP Arduino LPC176x 0.0.2 > NXP LPC1768

HARDWARE: LPC1768 100MHz, 31.80KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: framework-arduino-lpc176x 0.1.3, toolchain-gccarmnoneeabi 1.80201.190214 (8.2.
1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.4.6

|-- <Adafruit NeoPixel> 1.2.4
|-- <SailfishLCD>
upload disk:  E:
Checking size .pio\build\LPC1768\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  42.9% (used 13960 bytes from 32568 bytes)
PROGRAM: [====      ]  40.4% (used 191976 bytes from 475136 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed
CURRENT: upload_protocol = mbed
Looking for upload disk...
Use manually specified: E:
Uploading .pio\build\LPC1768\firmware.bin
Firmware has been successfully uploaded.
(Some boards may require manual hard reset)
========================= [SUCCESS] Took 5.21 seconds =========================

Thanks,

thinkyhead, 2019-09-17:
These comment boxes interpret your input as MarkDown syntax. Certain characters are stripped out and treated like formatting, which makes your post unreadable. There is a little "M" next to the edit box that takes you to the documentation page. Meanwhile, just remember to wrap pre-formatted text by putting three back-ticks ` ``` ` on a line by itself both before and after the formatted text:
```
This is 
preformatted
text that    won't get screwed up!
```

thinkyhead, 2019-09-17:
> I removed the TMCStepper_5513 folder 

It can also help to delete your `$HOME/.platformio/.cache` folder so a new download will be done.

ReadieS, 2019-09-17:
> These comment boxes interpret your input as MarkDown syntax. Certain characters are stripped out and treated like formatting, which makes your post unreadable. There is a little "M" next to the edit [...]

Thanks for the correction. I was looking for an escape character to place before the formatting but i couldn't find the right one yesterday. Thank you for pointing out the "M" I will know where to look next time.

Unfortunately forcing the re-download doesn't seem to have helped at all either at my end.

Thanks

tvixen, 2019-09-17:
Did you try the Dev-2.1.x version. It seems more stable with the TMC drivers.

teemuatlut, 2019-09-17:
The library version was just to rule out the recent changes in SPI write methods. I didn't think they'd fix your issue but it was worth exploring as a first measure.

> My printer is running properly

What does this mean in more detail?
The current register in the driver is write only so reading back the values is not a good indicator of something working.

ReadieS, 2019-09-17:
> Did you try the Dev-2.1.x version. It seems more stable with the TMC drivers.

I can confirm Dev 2.1.x is not working also

> > My printer is running properly
> 
> What does this mean in more detail?
> The current register in the driver is write only so reading back the values is not a good indicator of something working.

I cannot answer for the OP who is obviously much more qualified than me to comment but my printer will not move at all, it will heat, menus are accessible and it responds to terminal commands except for movements.

KDU300, 2019-09-17:
>My printer is running properly

@teemuatlut 
Of course I have Clk u. MOSI tested. The signals look alright. The board talks through SPI with the drivers. If that was not the case, the printer would not move. Small power of the printer does not move (eg 300mA). If the current is set correctly, everything is ok and the printer is working. It looks to me like the outpin is not turned inpin. Look at the pictures again. Here you can see the drivers on the MISO line answers.

gloomyandy, 2019-09-17:
@KDU300 Has this hardware ever worked for you (with an older version of Marlin)? I ask because the setup you have is very popular and yet we have seen very few problem reports like yours. So either this problem is very new (so an older version works fine), or there is something very specific to your system.

KDU300, 2019-09-17:
Yes, in the past I was able to read the registers. I had 'address values' for all 4 drivers. I have nothing special, just the usual stuff.  Attached my configs.

[Configuration(20190917).zip](https://github.com/MarlinFirmware/Marlin/files/3622775/Configuration.20190917.zip)
[Configuration_adv(20190917).zip](https://github.com/MarlinFirmware/Marlin/files/3622776/Configuration_adv.20190917.zip)

ReadieS, 2019-09-17:
Thanks for everyone comments above. It has become apparent that we seem to be experiencing different errors. In the interests of everyone sanity I am going to withdraw and possible try new drivers. Good luck KDU300 and sorry for butting in and complicating issues.

gloomyandy, 2019-09-17:
@KDU300 So what happens if you go back to the older version of Marlin? Does it still work?

You might also want to try and remove all but one of the drivers (and possibly switch that driver around), it could be that something has happened to one of your drivers and that is driving the SPI bus and stopping the other "good" drivers from working correctly.

KDU300, 2019-09-17:
Everything back to the beginning. I now have the third card in operation. The MISO input pin works on it. 
Two broken Bigtree boards. I should look for another dealer. 
Thanks to all who helped me with words and deeds.

tvixen, 2019-09-17:
Super... thanks for info

teemuatlut, 2019-09-20:
Please remember to close issues after they've been resolved.

closed 2019-09-20 by KDU300