MarlinFirmware/Marlin #15277: 1d8ad7c breaks Re-ARM builds: "Marlin\src\core\utility.h:36:5: error: 'UNUSED' was not declared in this scope"

salfter, 2019-09-16:
I pulled in the latest round of updates from bugfix-2.0.x last week, but when I went to build it for my Re-ARM-equipped A8, compilation stopped here:

```
Compiling .pio\build\LPC1768\src\src\feature\I2CPositionEncoder.cpp.o
In file included from Marlin\src\core\utility.cpp:23:
Marlin\src\core\utility.h: In function 'void serial_delay(millis_t)':
Marlin\src\core\utility.h:36:5: error: 'UNUSED' was not declared in this scope
     UNUSED(ms);
     ^~~~~~
Marlin\src\core\utility.h:36:5: note: the macro 'UNUSED' had not yet been defined
In file included from e:\tmp\marlin\marlin\src\hal\hal_lpc1768\hal.h:39,
                 from e:\tmp\marlin\marlin\src\hal\hal.h:26,
                 from Marlin\src\core\../inc/MarlinConfig.h:30,
                 from Marlin\src\core\../Marlin.h:24,
                 from Marlin\src\core\utility.cpp:25:
e:\tmp\marlin\marlin\src\hal\shared\marduino.h:84: note: it was later defined here
   #define UNUSED(x) ((void)(x))
```

It looks like commit 1d8ad7c removed the definition of UNUSED in Marlin/src/core/macros.h.  Reversing that commit's change to that file should fix this regression.

thinkyhead, 2019-09-16:
Oops, it should include `macros.h` to get that defined.

thinkyhead, 2019-09-16:
Oops, doing some reshuffling to get dependencies sorted. Looks like one was missed.

salfter, 2019-09-16:
Undid my fix, applied the most recent commit...now I'm getting this:

```
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o
In file included from C:\users\salfter\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:33,
                 from e:\tmp\marlin\marlin\src\hal\shared\marduino.h:36,
                 from e:\tmp\marlin\marlin\src\inc\marlinconfigpre.h:33,
                 from Marlin\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c:28:
C:\users\salfter\.platformio\packages\framework-arduino-lpc176x\cores\arduino/const_functions.h:3:10: fatal error: cstdint: No such file or directory
 #include <cstdint>
          ^~~~~~~~~
compilation terminated.
```

thinkyhead, 2019-09-16:
If you include configs in your reports it helps a lot. Without exact user configs, we find it impossible to reproduce some issues.

closed 2019-09-16 by thinkyhead

monterman, 2019-09-17:
here is my error Using LPC1769 from the SKR v1.3.  my configs attached.
`In file included from Marlin\src\core\utility.cpp:23:
Marlin\src\core\utility.h: In function 'void serial_delay(millis_t)':
Marlin\src\core\utility.h:36:5: error: 'UNUSED' was not declared in this scope
     UNUSED(ms);

     ^~~~~~
Marlin\src\core\utility.h:36:5: note: the macro 'UNUSED' had not yet been defined
In file included from Marlin\src\core\../inc/../HAL/./HAL_LPC1768/HAL.h:39,
                 from Marlin\src\core\../inc/../HAL/HAL.h:26,
                 from Marlin\src\core\../inc/MarlinConfig.h:30,
                 from Marlin\src\core\../Marlin.h:24,
                 from Marlin\src\core\utility.cpp:25:
Marlin\src\core\../inc/../HAL/./HAL_LPC1768/../shared/Marduino.h:84: note: it was later defined here
   #define UNUSED(x) ((void)(x))
 
Compiling .pio\build\LPC1768\src\src\feature\Max7219_Debug_LEDs.cpp.o
*** [.pio\build\LPC1768\src\src\core\utility.cpp.o] Error 1
 [FAILED] Took 32.50 seconds`

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3619267/Marlin.zip)

thinkyhead, 2019-09-17:
@monterman — Are you defining your own `CS_PIN` somewhere? After patching `UNUSED` the next error is that `TOUCH_BUTTONS` can't find that pin.

salfter, 2019-09-17:
This is my LPC1768 configuration:

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/3622222/configs.zip)

thinkyhead, 2019-09-18:
Thanks! I think we got it settled now.

monterman, 2019-09-18:
Yes we did, thank you Scott!  
Now a new issue,  getting the tft24 screen lcd working over serial.  
User https://github.com/dianlight made a version that works, but it's not ready to upstream.  

Sincerely,

Andres Montero