MarlinFirmware/Marlin #15279: [FR] (Obesity+needless features.v2.x useless for <=256K ROM boards. constant compiled size growth.)

delafer, 2019-09-16:
### Description
Only for a short period from September 5 to 15 (with the same settings) compiled project size increased by 10k. in my case 233 kb and this is without event enabled BL-Touch option. I have LA, S-Curve activated and SKR E3 DIP.
with not-so-sophisticated settings, the project size is at its limits for 256 K devices. (90% of total devices)
if I want to turn on BL-Touch I should disable other useful features. 
just because the developers have recently stuffed a bunch of useless code and libraries into the project.
no one of developers thinks about size limits, turning a project into a heavy monster suitable only for > 512 KB rom boards.
It looks like sabotage or a complete misunderstanding of what Marlin 2.xx users need.
For BIG Tree SKR devices you have added USE_USB_COMPOSITE libs, without the ability to deactivate this option, etc.
Added joystick and other frank garbage....
Since 5 Sep 2019 auto detection of inserted SD card doesn't work properly. More size, more BUGS, no useful improvements.

Of course I can take the old version and use it and i will do it for sure.
I have only one question to developers: are you sure you are moving in the right direction?
turning a project into a monster of a patient with severe obesity?
forgetting to optimize the code and think about its compactness?

delafer, 2019-09-16:
P.S. Marlin developers have added a bunch of useless functions and libraries and other crappy stuff in the latest marlin versions,and don't even try to optimize project compiling size and it constantly growing without visible improvements. If you are trying to compile your project for 256K board and have activated BLTouch, S-Curve, Stepper Drivers, etc and it doesn't fit - I recommend to take an older version. it has less shit and it is more stable or replace rather Marlin 2.xx developers or block / blacklist some of them.. 

P.S. IMHO marlin developers do their best to kill the project, turning marlin 2.xx into a fat obese monster.

thinkyhead, 2019-09-16:
Thank you for your comments. We are well aware of the enlarged size and will deal with it as we figure out the causes.

closed 2019-09-16 by thinkyhead

delafer, 2019-09-16:
> Thank you for your comments. We are well aware of the enlarged size and will deal with it as we figure out the causes.

I know why, but you don't even to try to discuss and trying to check and solve this problem. You have closed my request immediately. You don't want to hear any critic about marlin and marlin developers.

thinkyhead, 2019-09-16:
We don't do this kind of thing here. If you want to rant, you can rant directly to me on Discord.

thinkyhead, 2019-09-16:
> Of course I can take the old version and use it and i will do it for sure.

That is the recommended course, as this branch is in heavy development and will definitely be unstable and broken from time to time.

Roxy-3D, 2019-09-16:
@delafer   Thinky closed this thread before I got to it.   So, I'll leave it at that.   But your tone is very unappreciated and I was going to ban you from even having read access to the repository.

thinkyhead, 2019-09-16:
My MarlinBot script (work in progress) closed this due to failure to follow instructions in the issue template. But, I would have closed it manually if the script had not.

thinkyhead, 2019-09-17:
For a smaller build remove `-DUSE_USB_COMPOSITE` from `platformio.ini`.

thinkyhead, 2019-09-17:
I've patched Marlin so the `USE_USB_COMPOSITE` flag is now only set in a separate environment that developers can continue to use, and is no longer set in the default one.

delafer, 2019-09-17:
> I've patched Marlin so the `USE_USB_COMPOSITE` flag is now only set in a separate environment that developers can continue to use, and is no longer set in the default one.

thanks. It should help other people. As for me, I would take an older version and invest time and integrate ony features I really need.

delafer, 2019-09-17:
I’m a developer myself and I understand what opensource is and therefore I think that developers do not owe anything to anyone. They (You) have no obligations and duties. This is voluntary not commercial participation without financial benefits.
I just wanted to say that it would be a shame that an initially good project will grow and grow and grow with rarely used functionality and will turn into a monstrous project. Many admire e.G. Lergde's boards (and their firmware) not because they have support for rarely used functions, a laser engraving, joystics, brick and tetris games, etc. - but simply because the basic functionality that they offer works well, stable, fast, and they perform exactly those main basic tasks that 99% of users really needs.

Сhinese clones of BL/3D Touch cost only about 10 $ and it's relates cheap useful features, which most will want to try (now or in the feature)
Linear Advance is something really cool, S Curve Acceleration sounds interesting too.
SPI driver control , Autoshutdown, etc. 

This is a base reasonable functional which IMHO makes sense. 
and if I want to activate it - it would be logical that this functionality would fit in ROM on the most common boards available on the market. 

The main criticism of the Marlin 2.x project is that it is still raw, sometimes unstable and buggy .
Most people update marlin version with hope that old bugs will be fixed, without a new ones.
And it’s really sad when you update the version, build it and notice that it no longer fits in the ROM.

delafer, 2019-09-17:
> @delafer Thinky closed this thread before I got to it. So, I'll leave it at that. But your tone is very unappreciated and I was going to ban you from even having read access to the repository.

I have answered above, nobody likes criticism and I didn’t mean anyone personally. this is by no means -> any kind of demands to anyone, but rather frustration that the efforts and huge efforts of the people who created this project will go to ashes.

regarding ban. there is a proverb - "trying to scare a hedgehog with a naked booty". Any anauthorised github user has a guest readonly access to public opensource REPOs. and this is the main philosophy of opensource projects -> it should be opensource and freely available to any person, even if you do not like someone personally.

InsanityAutomation, 2019-09-17:
And remember the opposing argument, the users who want Marlin to be more like Smoothie and RRF where everything is compiled in and changed around by gcode. It needs to be balanced. We cant support old hardware forever. At some point the gains justify it. 
Most of us do our best to keep new functionality behind #define code that will only include it when explicitly called for. Some changes do enlarge specific features. Sometimes in the development builds we get very verbose safe code in then back it out to more compact stuff without debug output later. There will come a point when that MCU is just too small for anything but klipper which splits the tasks. Its inevitable. Question is when.
A large percentage of users want more verbose output, guided screens for tasks, and fancy functionality. Things will grow to extend features to support this. We are well aware a full boat build no longer fits on an atmega2560. Remember, a few years ago the same argument was made for atmega1284p. We still fit functional builds on these boards today!
When dealing with hardware there has to be a point that you only support newer stuff. You cant support everything forever or everything else will just leave you behind to drop into obsolescence with it. There will come a day when we have to announce new versions no longer officially support Sanguino / Atmega1284p and even new versions no longer support 8-bit AVR. Its awhile off to be certain but we need to be mindful of the constraints it places vs competitors who only support 32-bit platforms!

delafer, 2019-09-17:
> And remember the opposing argument, the users who want Marlin to be more like Smoothie and RRF where everything is compiled in and changed around by gcode. It needs to be balanced. We cant support [...]

we are not talking about an old hardware / motherborads. My SKR E3 DIP for Ender I have bought only a few weeks ago. Of course it doesn't makes any sense to support old hardware, but you should not. Where is a solution. Branches. If someday one of the developers notice, that firmware is too large for 256 KB ROM Boards -> special for such cases a new GIT Branch could be created. e.G. Marlin 2.2 for 512 ROM Boards or Marlin 3.xx for 2 MB ROM or Marlin 10.xx for 1 GB ROM boards )

Roxy-3D, 2019-09-17:
@delafer    You obviously have all the answer and are much wiser than us.

You would be much happier using the RepRap or Smoothie firmware.     Why don't you join their community.   I'm sure they will be much more willing to take orders from you than we are.

delafer, 2019-09-17:
> @delafer You obviously have all the answer and are much wiser than us.
> 
> You would be much happier using the RepRap or Smoothie firmware. Why don't you join their community. I'm sure they will be much more willing to take orders from you than we are.

Orders ? Really ? Have you had a real chef and real job in your life ? Simply to compare how real orders looks like. I have written already: my own problems I can solve by myself and I doing that already, and I have also written ->"(marlin/opensource) developers do not owe anything to anyone. They have no obligations and duties. " (and shouldn't and must not response to anyone) , its all completely voluntary and optional.

Roxy-3D, 2019-09-17:
Wow!    Time to end this...    This clown is 'Blocked'.

InsanityAutomation, 2019-09-17:
A budget board designed for a budget machine. That gets it up to about the same as an atmega2560 but with a 32bit math processor, and its far better than the stock 1284p in the machine its designed for. Itll be a long time before Marlin doesnt work there. But dont expect every bell and whistle when the boards Marlin 2.0 is intended to compete with all support 512k or more as the basic memory footprint. Even RRF removed newer features for the older generation hardware. For the Duet 3 boards theyre even recommending host printing over local sd printing to offload stuff.

Patag, 2019-09-17:
@delafer : If you are a senior developer as you pretend in your signature : just do it !

alantoy, 2019-09-18:
> @delafer : If you are a senior developer as you pretend in your signature : just do it ! And take care to your brain size. It looks like it's becoming as big as a water melon.

in your case it’s too late to think about the size of the brain, it’s extremely smart to write to someone who is blocked and cannot answer in any form. There is a Russian proverb - “it’s better to be silent and seem like a fool than to say (write) something and to prove it definitely"
But your care and concern with regard to other people brains - is self-consistent , when you have no own (brains), all that remains is to write and take care of other people's brains.

you probably don't know - but there are many programming languages and being a developer and an expert in one area does not automatically mean knowledge of any other programming languages and technologies, although it does not contradict first statement.

alantoy, 2019-09-18:
> Wow! Time to end this... This clown is 'Blocked'.

a clown could be blocked only in one case - if you block yourself. Don’t worry, criticism of marlin was not in your address, according to your commits - your "active" participation in the development of the marlin 2.x project is in the deep past, it's a history, there is is there something you love much more - 
in psychiatry, this is called the “Napoleon Complex”, when a person is not outstanding from any point of view - but positioning itself like a big boss who in his small "virtual" world and can allow someone to speak/write someone or block / blacklist, only because you don’t like other people's points of view.