MarlinFirmware/Marlin #15282: SKR 1.3 E0 Invert_E0_DIR does not effect anything

kurtisane, 2019-09-16:
Hi guys,
I'm running Marlin 2.0 on SKR 1.3 with DRV Stepper Driver. I need to invert my Extruder Stepper. I know I could unpin and flip the Pins, but I don't really want to and this is anyway a bug or a misconfig from my side.

Changing the Invert_E0_Dir to true or false does not change anything. Also tried with the other Extruders E1, E2 etc. 
Tried this :  https://github.com/MarlinFirmware/Marlin/issues/13775
Did not change anything either. 

Regards !

tvixen, 2019-09-17:
Hmm seems you didn't do the M502, M501, M500 
If any further help, please upload your config files.
Also remember to do two reset of the SKR 1.3 board, if there's new firmware on the SD card.
1.st reset reads the firmware, 2.nd reset loads the new firmware.

kurtisane, 2019-09-17:
i just deleted the old firmware.cur from the sd card and put the new .bin on it. 
Started it up, waited for 20 sec., then reset with the button on the board two times, executed M502 then M501, then M500.
The Stepper still turns in the same direction. 
Just to note I'm testing it with M302 P1 then G1 E200

The zip contains the config.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3622050/Marlin.zip)

tvixen, 2019-09-17:
Make these changes in Configuration.h
//#define DISABLE_INACTIVE_EXTRUDER 

#define INVERT_E0_DIR true
#define INVERT_E1_DIR false
#define INVERT_E2_DIR false
#define INVERT_E3_DIR false
#define INVERT_E4_DIR false
#define INVERT_E5_DIR false

//#define DEBUG_LEVELING_FEATURE

in Conf_adv.h
//#define MINIMUM_STEPPER_PULSE 1 (for A4988 and A5984 stepper drivers you use 8825)

If this don't do it, switch your X driver pcb to the E driver slot, and try again with these settings.

kurtisane, 2019-09-18:
Alright that did it.

Thanks a lot!

boelle, 2019-09-18:
@kurtisane if "bug" is fixed please close the issue :-)

tvixen, 2019-09-19:
NP ...glad to help

boelle, 2019-09-19:
closing this one as it seems fixed, we can reopen if its not the case

closed 2019-09-19 by boelle