MarlinFirmware/Marlin #15288: [BUG] toolchain-gccarmnoneeabi Not working on Mac

ReadieS, 2019-09-17:
### Description

Bug fix 2.0 wont build on Macs due to an error with toolchain-gccarmnoneeabi compatibility.
Issue can be rectified by editing ~/.platformio/platforms/nxplpc-arduino-lpc176x folder, edit the manifest.json
[Please see discussion here](https://community.platformio.org/t/can-not-compile-arm-none-eabi-g-command-not-found/9458/11)

### Steps to Reproduce

1. Edit bug fix 2.0 to compile for LCP1768
2. Press Build

**Expected behavior:** [What you expect to happen]
Marlin to compile correctly

**Actual behavior:** [What actually happens]
`Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/maste

r.zip; framework: arduino; board: nxp_lpc1768)

--------------------------------------------------------------------------------

```
PackageManager: Installing toolchain-gccarmnoneeabi @ >=1.80201.181221

Downloading...

Unpacking...

toolchain-gccarmnoneeabi @ 1.80201.181221 has been successfully installed!

Verbose mode can be enabled via `-v, --verbose` option

CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.htm

l

PLATFORM: NXP Arduino LPC176x 0.0.2 > NXP LPC1768

HARDWARE: LPC1768 100MHz, 31.80KB RAM, 464KB Flash

DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)

PACKAGES: framework-arduino-lpc176x 0.1.3

Converting Marlin.ino

sh: arm-none-eabi-g++: command not found

*** Error 127

AssertionError: :

  File "/Volumes/Macintosh HDD/samuelread/.platformio/penv/lib/python2.7/site-packages/platfo

rmio/builder/main.py", line 126:

    env.SConscript("$BUILD_SCRIPT")

  File "/Volumes/Macintosh HDD/samuelread/.platformio/packages/tool-scons/script/../engine/SC

ons/Script/SConscript.py", line 541:

    return _SConscript(self.fs, *files, **subst_kw)

  File "/Volumes/Macintosh HDD/samuelread/.platformio/packages/tool-scons/script/../engine/SC

ons/Script/SConscript.py", line 250:

    exec _file_ in call_stack[-1].globals

  File "/Volumes/Macintosh HDD/samuelread/.platformio/platforms/nxplpc-arduino-lpc176x/builde

r/main.py", line 136:

    target_elf = env.BuildProgram()

  File "/Volumes/Macintosh HDD/samuelread/.platformio/packages/tool-scons/script/../engine/SC

ons/Environment.py", line 224:

    return self.method(*nargs, **kwargs)

  File "/Volumes/Macintosh HDD/samuelread/.platformio/penv/lib/python2.7/site-packages/platfo

rmio/builder/tools/platformio.py", line 110:

    env.BuildFrameworks(env.get("PIOFRAMEWORK"))

  File "/Volumes/Macintosh HDD/samuelread/.platformio/packages/tool-scons/script/../engine/SC

ons/Environment.py", line 224:

    return self.method(*nargs, **kwargs)

  File "/Volumes/Macintosh HDD/samuelread/.platformio/penv/lib/python2.7/site-packages/platfo

rmio/builder/tools/platformio.py", line 288:

    env.ConvertInoToCpp()

  File "/Volumes/Macintosh HDD/samuelread/.platformio/packages/tool-scons/script/../engine/SC

ons/Environment.py", line 224:

    return self.method(*nargs, **kwargs)

  File "/Volumes/Macintosh HDD/samuelread/.platformio/penv/lib/python2.7/site-packages/platfo

rmio/builder/tools/piomisc.py", line 198:
out_file = c.convert(ino_nodes)

  File "/Volumes/Macintosh HDD/samuelread/.platformio/penv/lib/python2.7/site-packages/platfo

rmio/builder/tools/piomisc.py", line 57:

    return self.process(contents)

  File "/Volumes/Macintosh HDD/samuelread/.platformio/penv/lib/python2.7/site-packages/platfo

rmio/builder/tools/piomisc.py", line 80:

    assert self._gcc_preprocess(contents, out_file)

========================= [FAILED] Took 154.12 seconds =========================
```
#### Additional Information
[Please see discussion here](https://community.platformio.org/t/can-not-compile-arm-none-eabi-g-command-not-found/9458/11)

[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/3621077/Config.Files.zip)

Please note: I have searched for a log of this issue on here but cannot find any indication of one. apologies if this is a repeat.

darknode, 2019-09-17:
You can install it with brew tap from here https://github.com/darknode/homebrew-stm32:
```
brew tap darknode/stm32
brew install arm-none-eabi-gcc
```

k-plan, 2019-09-18:
>You can install it with brew tap from here ...

Thanks for your answer @darknode 
But is there no other solution as install homebrew packages into your Mac OSX system?
If I remember correctly, it had been working with VS Code on Mac awhile ago.

darknode, 2019-09-18:
@k-plan It should work if have Arduino IDE installed and added arm-none-eabi into Arduino tool-chain. But if you don't have IDE or don't want to use it, you could use brew (for MacOS) or install gcc manually. You can find download link inside brew tap file: https://github.com/darknode/homebrew-stm32/blob/master/arm-none-eabi-gcc.rb

k-plan, 2019-09-18:
@darknode thanks for your response.
Problem is, `arm-none-eabi toolchain`  doesn’t reference the manifest.json on disk but link on the web.

> ~/.platformio/platforms/nxplpc-arduino-lpc176x/platform.json

refer to `https://raw.githubusercontent.com/p3p/pio-nxplpc-arduino-lpc176x/master/manifest.json` which work well on Windows and Linux, but not an MacOS.
We will need a either way link in `nxplpc-arduino-lpc176x/platform.json`

ReadieS, 2019-09-18:
> You can install it with brew tap from here https://github.com/darknode/homebrew-stm32:
> ```
> brew tap darknode/stm32
> brew install arm-none-eabi-gcc
> ```
Thanks for your reply @darknode 
I am not a very experienced user of Arduino or Platformio I only really use it for marlin installations.
I reported it as a bug as there is no mention of the issue in the marlin guides or anyway to get around.
I managed to follow the linked forum post from the original post and get my installation working but I thought either the links need addressing or the marlin guide needs to be updated to include any additional steps for mac users as it was quite hard to track down the solution.

Thanks @k-plan 
> > You can install it with brew tap from here ...
> 
> Thanks for your answer @darknode
> But is there no other solution as install homebrew packages into your Mac OSX system?
> If I remember correctly, it had been working with VS Code on Mac awhile ago.

I can confirm Vscode was not working on my mac either. As you mention above it would be great if homebrew installations weren't required so mac installations are as easy as for other OS's.

Thanks all,

p3p, 2019-09-18:
Fix one mildly annoying thing for most users and completely break it for the others, .. that does sound like something I would do. 

I replaced the officially distributed `arm-none-eabi-gcc` package because they have been building it with optimisations turned off, which at least doubles build times, not really a problem when doing one build but the Travis tests were taking too long.

I'l roll back the change and hope they fix the issue on their side one day.

ReadieS, 2019-09-18:
Hi @p3p .
Apologies I didn't mean any offence to your work. I have no real problem with it being left the same if the work around is in the guides my comment about avoiding home brew really comes from a complete lack of knowledge on the subject. Just took me a long time to figure out how to get it up and running on Mac. 
Didn't want to leave it causing any offence and I appreciate I have no knowledge of the subject other than I had this issue. Thanks to you and all who put time and effort into this.

k-plan, 2019-09-18:
Hi @p3p ,
thanks for all information and your fast help.
Sorry to hear, that  optimisations turned on break compatibility with Mac OS.
Yes, now with your roll back, it will work on Mac OSX. Thanks.
Is it not possible, to keep this enable for Linux and Windows by disable it only for Mac?
In this way, other user will not hate Mac users for doing this.

> Didn't want to leave it causing any offence and I appreciate I have no knowledge of the subject other than I had this issue. Thanks to you and all who put time and effort into this.

Same on my side. Sorry and no harm meant.

thinkyhead, 2019-09-18:
My entire workflow is Mac-based and I have never had any issues with builds for any target. Is it the default python install that is the problem? I use MacPorts and probably have a different python install than the default.

thinkyhead, 2019-09-18:
Note that if you want a decent `gcc` toolchain on the Mac, you must install XCode.

k-plan, 2019-09-18:
@thinkyhead 

First I have Mac default  installed Python 2.7 `.platformio/penv/lib/python2.7/` => didn't work.
Then [install Python 3.7](https://www.python.org/downloads/) but not via MacPorts, last Version, in the way platformIO suggest on their side `.platformio/penv/lib/python3.7/`=> didn't work.

p3p, 2019-09-18:
@ReadieS @k-plan Don't worry no offence taken ^^, (my irritation may have accidentally come though in my response but it wasn't directed at you) I have no OSX experience and don't have the time to research the correct solution at the moment so rolling back is the best I can do. It is only going to noticeably affect Travis tests so I don't think mac users are going to get any hate ;) .. no more than usual anyway.

ReadieS, 2019-09-21:
I'll mark this as closed due to the roll back.
Thank you all again for everything.

closed 2019-09-21 by ReadieS