MarlinFirmware/Marlin #15293: [BUG] (LCD2004 is hanging on SKR 1.3)

Decygnus, 2019-09-17:
Marlin 2.0
I have SKR v1.3 and LCD 2004. While printer is printing when I push on the encoder and screen is hanging without any information! Print is continuing, but screen doesn't answer for push ups or rotating of encoder. After some time it shows the info screen. All parameters are updating in real time, but when i push on encoder....the same hanging. 

This problem appears only when printing.
Files are attached
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3622953/Marlin.zip)

shitcreek, 2019-09-17:
LCD randomly hangs sometimes with the CR10 Stock Display and SKR 1.3 as well. This also happens sometimes during bed leveling.

Ke5han, 2019-09-17:
I am having exactly the same problem, SKR v1.3 with LCD2004
during a print I can't push the encoder, otherwise the screen will be blue, but the print will continue.
If I change it to 12864, the response from the screen is really slow, a lot of times it will be a timeout and jump back to the info page.

Prelo96, 2019-09-19:
Same problem,
lcd became unreactive during printing.
I can do something while traveling only
skr1.3 
delta 200 segment
tmc2130 spi 64microstep
linear advance
reprapdiscount controller

Ke5han, 2019-09-21:
I am using a LCD2004 from micromake and increase the BLOCK_BUFFER_SIZE to 128 solves my problem as I posted in #15169, BLOCK_BUFFER_SIZE 64 only allows me to play with menus after solid layers.

arbitur, 2019-09-25:
I also have this issue. It used to not hang like this in a build from commit #2971b48a128662ab2f50a0a238aedfb64ec785d9 branch bugfix-2.0.x but after I merged my config branch to commit 07eb2130467c40f4dbe4da969dc1a04746420d91 it started hanging while printing curves. So somewhere inbetween these commits there mustve been a change that made it hang like this Im guessing its some queue thing hanging up the UI thread or something. Really annoying.

boelle, 2019-10-12:
@Decygnus still having issues?

stefi01, 2019-10-14:
same problem with Rambo v1.3

boelle, 2019-10-24:
can it be rules out that this is not a hardware issue? ie bad made china board?

tried what @Ke5han did?

where do you get marlin from?

thinkyhead, 2019-10-24:
@boelle — We are pretty certain there's a bug, but it's hard to track down because of the spurious behavior.

boelle, 2019-10-24:
i only asked because there is a lot of issues where the common is boards from china
a little harsh on my side but i just find it a bit to strange that there are so many issues left with china made hardware

and in a few cases i discovered that people in fact did use a fork and not direct from us, hence the last q

ZhenyaKa, 2019-10-26:
The same problem. But I have summer build of Marlin 2.0.bugfix and all okay in it.

boelle, 2019-10-26:
do you remember the commit of that build?

ZhenyaKa, 2019-10-26:
> do you remember the commit of that build?

no, but i can upload source for comparing files

boelle, 2019-10-26:
let see what @thinkyhead says to that idea

ZhenyaKa, 2019-10-26:
> let see what @thinkyhead says to that idea

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3775256/Marlin.zip)

thinkyhead, 2019-10-26:
The STM32 platforms are constantly in flux, so that may be contributing. We've set `platformio.ini` to use ststm32@5.6.0. But possibly an earlier version would be better. We'll need more feedback.

arbitur, 2019-10-26:
@Boelle see my answer of working commit. (https://github.com/MarlinFirmware/Marlin/commit/2971b48a128662ab2f50a0a238aedfb64ec785d9)

thinkyhead, 2019-10-26:
So, around August 8.

runningz, 2019-10-27:
Same problem here with skr 1.3 and fystec 12864 rgb. With a Marlin from 13/07/2019 no problem...

istyszy, 2019-10-31:
Same problem with SKR V1.3 + rep-rap smart display controller 2004 with the included external card reader.(Anycubic Kossel delta printer)

hearnia2k, 2019-10-31:
I think this issue is the same as https://github.com/MarlinFirmware/Marlin/issues/15169

I also have it on my Anycubic Predator with SKR 1.3 using FYSETC MINI12864 RGB v2.1.

boelle, 2019-11-03:
@Decygnus are your issue the same as in https://github.com/MarlinFirmware/Marlin/issues/15169  ?

thinkyhead, 2019-11-06:
Duplicate of #15169

closed 2019-11-06 by thinkyhead