MarlinFirmware/Marlin #15294: [BUG] Enabling #define E0_AUTO_FAN_PIN causes compilation fail on Fystec Cheetah

salacpavel, 2019-09-17:
### Description

Latest Marlin buxfix-2.0.x build fails, when #define E0_AUTO_FAN_PIN is set from -1 to an actual mainboard E0 fan pin (PB0) with following error message:
Compiling .pio\build\STM32F103R_fysetc\src\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:31:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1849:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1849:5: error: the value of 'PIN_MAP' is not usable in a constant expression
In file included from C:\users\salac\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54:0,
                 from C:\users\salac\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\../shared/Marduino.h:36,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.h:32,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:30:
C:\users\salac\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:59:29: note: 'PIN_MAP' was not declared 'constexpr'
 extern const stm32_pin_info PIN_MAP[];
                             ^~~~~~~
*** [.pio\build\STM32F103R_fysetc\src\src\HAL\HAL_STM32F1\HAL.cpp.o] Error 1
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL_SPI.cpp:35:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1849:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1849:5: error: the value of 'PIN_MAP' is not usable in a constant expression
In file included from C:\users\salac\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54:0,
                 from C:\users\salac\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from d:\marlin-bugfix-2.0.x-cheetah\marlin\src\hal\shared\marduino.h:36,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/./HAL_STM32F1/HAL.h:32,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\HAL_SPI.cpp:35:
C:\users\salac\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:59:29: note: 'PIN_MAP' was not declared 'constexpr'
 extern const stm32_pin_info PIN_MAP[];
                             ^~~~~~~
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\Servo.cpp:26:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1849:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:1849:5: error: the value of 'PIN_MAP' is not usable in a constant expression
In file included from C:\users\salac\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54:0,
                 from C:\users\salac\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from d:\marlin-bugfix-2.0.x-cheetah\marlin\src\hal\shared\marduino.h:36,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/./HAL_STM32F1/HAL.h:32,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\Servo.cpp:26:
C:\users\salac\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:59:29: note: 'PIN_MAP' was not declared 'constexpr'
 extern const stm32_pin_info PIN_MAP[];
                             ^~~~~~~
*** [.pio\build\STM32F103R_fysetc\src\src\HAL\HAL_STM32F1\HAL_SPI.cpp.o] Error 1
*** [.pio\build\STM32F103R_fysetc\src\src\HAL\HAL_STM32F1\Servo.cpp.o] Error 1
In file included from Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/SanityCheck.h:1849:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/SanityCheck.h:1849:5: error: the value of 'PIN_MAP' is not usable in a constant expression
In file included from C:\users\salac\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:54:0,
                 from C:\users\salac\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from d:\marlin-bugfix-2.0.x-cheetah\marlin\src\hal\shared\marduino.h:36,
                 from d:\marlin-bugfix-2.0.x-cheetah\marlin\src\hal\hal_stm32f1\hal.h:32,
                 from d:\marlin-bugfix-2.0.x-cheetah\marlin\src\hal\hal.h:26,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
C:\users\salac\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/boards.h:59:29: note: 'PIN_MAP' was not declared 'constexpr'
 extern const stm32_pin_info PIN_MAP[];
                             ^~~~~~~
*** [.pio\build\STM32F103R_fysetc\src\src\HAL\HAL_STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
================================================================ [FAILED] Took 14.35 seconds ================================================================

Environment             Status    Duration
----------------------  --------  ------------
megaatmega2560          IGNORED
megaatmega1280          IGNORED
at90usb1286_cdc         IGNORED
at90usb1286_dfu         IGNORED
DUE                     IGNORED
DUE_USB                 IGNORED
DUE_debug               IGNORED
LPC1768                 IGNORED
LPC1769                 IGNORED
melzi                   IGNORED
melzi_optiboot          IGNORED
rambo                   IGNORED
sanguino_atmega644p     IGNORED
sanguino_atmega1284p    IGNORED
STM32F103R              IGNORED
STM32F103R_fysetc       FAILED    00:00:14.352
STM32F103R_bigtree      IGNORED
STM32F4                 IGNORED
STM32F7                 IGNORED
ARMED                   IGNORED
STM32F103V_longer       IGNORED
mks_robin               IGNORED
mks_robin_lite          IGNORED
mks_robin_mini          IGNORED
mks_robin_nano          IGNORED
jgaurora_a5s_a1         IGNORED
STM32F407VE_black       IGNORED
BIGTREE_SKR_PRO         IGNORED
BIGTREE_BTT002          IGNORED
teensy31                IGNORED
teensy35                IGNORED
malyanm200              IGNORED
esp32                   IGNORED
fysetc_f6_13            IGNORED
linux_native            IGNORED
SAMD51_grandcentral_m4  IGNORED
include_tree            IGNORED
=========================================================== 1 failed, 0 succeeded in 00:00:14.352 ===========================================================  
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

### Steps to Reproduce 

1. Enable above option in Config_adv.h
2. hit build

**Expected behavior:**  should buold green :)

**Actual behavior:** build fails

#### Additional Information
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3623064/Configuration.zip)

sjasonsmith, 2019-09-18:
You say _Latest Marlin buxfix-2.0.x_, but both platformio.ini and the configuration files look pretty old.

I can't try your files directly, because they are not compatible with the current state of Marlin. Can you apply your configuration to the latest state of bugfix-2.0.x, to see if that solves your problem?

sjasonsmith, 2019-09-18:
I was able to compile a _similar_ Cheetah config with the latest Marlin, with E0_AUTO_FAN_PIN == PB0.

salacpavel, 2019-09-18:
> You say _Latest Marlin buxfix-2.0.x_, but both platformio.ini and the configuration files look pretty old.
>
> I can't try your files directly, because they are not compatible with the current state of [...]

closed 2019-09-18 by salacpavel