MarlinFirmware/Marlin #15296: [BUG] SKR V1.3 and TMC5160 v1.2 - driver error

040medien, 2019-09-17:
### Description

I'm using 5 TMC5160 V1.2 (BIGTREETECH) on my SKR v1.3.

At boot after a few seconds I get the "Driver Error PRINTER HALTED please reset" error message.

### Steps to Reproduce

1. Set up board with 5 TMC5160 according to standard guide (all red jumpers closed under the driver). According to the manufacturer pin 6 is now connected to ground with v1.2, so no change needed.
2. Set up marlin with the attached config
3. For testing connect a single stepper motor
4. right after booting, run M122 (output below).
5. wait a few seconds.

**Expected behavior:** [What you expect to happen]
Firmware working

**Actual behavior:** [What actually happens]
Firmware stops with "Driver error"

#### Additional Information

*[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3623612/Configs.zip)
*M122 output:

SENT: M105
READ: Error:No thermistors - no temperature
SENT: M122
READ: 		X	Y	Y2	Z	Z2
READ: Enabled		false	false	false	false	false
READ: Set current	800	800	800	800	800
READ: RMS current	792	792	792	792	792
READ: MAX current	1117	1117	1117	1117	1117
READ: Run current	15/31	15/31	15/31	15/31	15/31
READ: Hold current	7/31	7/31	7/31	7/31	7/31
READ: Global scaler	133/256	133/256	133/256	133/256	133/256
READ: CS actual	0/31	16/31	16/31	0/31	0/31
READ: PWM scale	3758096384	3221225472	4160749568	4026531840	4026531840
READ: stealthChop	false	false	false	false	false
READ: msteps		256	256	0	256	256
READ: tstep		536870912	1073741824	1073741824	2147483648	2147483648
READ: pwm
READ: threshold					
READ: [mm/s]						
READ: OT prewarn	true	false	true	true	false
READ: OT prewarn has
READ: been triggered	true	true	true	true	true
READ: off time	0	0	0	0	0
READ: blank time	16	16	16	16	16
READ: hysteresis
READ: -end		-3	-3	-3	-3	-3
READ: -start		1	1	1	1	1
READ: Stallguard thrs	0	0	0	0	0
READ: DRVSTATUS	X	Y	Y2	Z	Z2
READ: stallguard		*	*	*	*	*
READ: sg_result	0	0	0	0	0
READ: fsactive						
READ: stst		*	*	*	*	*
READ: olb		*	*	*	*	*
READ: ola		*	*	*	*	*
READ: s2gb				*		*
READ: s2ga		*			*	
READ: otpw		*		*	*	*
READ: ot		*	*	*	*	*
READ: Driver registers:
READ: 		X	0xFB:F0:00:00
READ: 		Y	0xEF:E0:00:00
READ: 		Y2	0xF7:C0:00:00
READ: 		Z	0xFB:F0:00:00
READ: 		Z2	0xF7:E0:00:00
READ: 
READ: 
READ: Testing X connection... OK
  Connected to machine!
READ: Testing Y connection... OK
READ: Testing Y2 connection... OK
READ: Testing Z connection... OK
READ: Testing Z2 connection... OK
READ: ok P15 B3
SENT: T0
READ: ok P15 B3
SENT: M105

tvixen, 2019-09-18:
Which Marlin version do you use ?
Did you try latest version ?
Did the board work before this (with other drivers) ?
Do you get a compiler error when using 4988 drivers ?

vaka1983, 2019-09-18:
I cut clk pin

extesy, 2019-09-18:
@040medien Check out https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/issues/2

040medien, 2019-09-18:
Tak tvixen. 

I've was on the latest 2.0 bugfix as of yesterday (c7acd5c) but also tried Bigtreetech's [latest Marlin 2.0 fork](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/firmware/Marlin-bugfix-2.0.x) for the SKR with the same result.

The board runs fine with DRV8825 drivers. No compiler errors with DRV8825 or 4988.
I will try to cut the clk pin.

040medien, 2019-09-18:
Thank you @vaka1983 and @extesy - it was the clk pin. You don't need to pull it to gnd anymore, but it needs to be disconnected from the board.

closed 2019-09-18 by 040medien

modwadd, 2019-10-08:
Hello @040medien ,
could you describe in more detail whre to disconnect from?
Thanks

040medien, 2019-10-10:
@modwadd, you need to clip or remove this pin:
[image]

I simply heated the solder and pushed the pin up so it no longer makes connection with the board.

modwadd, 2019-10-10:
OK Thanks.
I did it for my SKR Pro too.
Together with the switch to SW SPI it now works for me too. (no TCM error)