MarlinFirmware/Marlin #15299: Arduino DUE compilation error

Roman-RV78, 2019-09-18:
Hello. I get an error while compiling

_arm-none-eabi-gcc: error: CreateProcess: No such file or directory
exit status 1
Error compiling for board Arduino Due (Programming Port)._
[Conf.zip](https://github.com/MarlinFirmware/Marlin/files/3625210/Conf.zip)

On early bug fixes, compilation went through.

tvixen, 2019-09-18:
Please fill out the Questions in the description!
It only takes a minimum of effort, but will make things a lot easier for everyone

Roman-RV78, 2019-09-19:
Marlin-bugfix-2.0.x from 08/25/2019 compilation runs fine, today's bugfix also does not compile. Arduino IDE 1.8.10, error:
  arm-none-eabi-gcc: error: CreateProcess: No such file or directory

sjasonsmith, 2019-09-20:
I think it has to do with some of the command line arguments being too long for Windows. I found references in other issues to it, but didn't happen across an easy solution. I encounter the same issue when I build your configs for DUE from the Arduino IDE.

The good news is that it builds fine in Visual Studio Code using Platform IO. Maybe you should set that up and build from there?

ZS1FJK, 2019-10-09:
Hello
I also have this issue.
I have a archim 2 board and can only use Arduino to compile my marlin unfortunately. I am using today's bug fix and with Arduino 1.9.0 beta .
@sjasonsmith I found the same post about the arguments being too long for Windows.

boelle, 2019-10-12:
@Roman-RV78 still having issues when using latest 2.0.x ?

boelle, 2019-10-12:
@Roman-RV78 think the problem is you are not using ide 1.9 beta

Roman-RV78, 2019-10-12:
I think the problem is in the Arduino IDE. Today's build also does not compile. In PlatformIO, everything compiles.

boelle, 2019-10-12:
what version of arduino ide do you use?

Roman-RV78, 2019-10-12:
1.8.10

boelle, 2019-10-12:
1.9.0 beta is the recommended version

ZS1FJK, 2019-10-13:
I have tried 1.9.0 beta and 1.8.10 both don't work for archim same error .

boelle, 2019-10-13:
also if using vscode+plaform.io ?

boelle, 2019-10-24:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-24 by boelle