MarlinFirmware/Marlin #15302: [BUG] Unable to compile AT90USB due to PIO cust scripts + TMC2208 dvr error

fiveangle, 2019-09-18:
### Description

Unable to compile for Printrboard RevF for 2 reasons (assuming 1 is toolchain change, as this worked March 26).

1) Suspect @Bob-the-Kuhn 's custom `create_custom_upload_command_DFU.py` uploader script either never was compliant, or has become non-compilant with PIO, but my Python-foo is not good enough to track down the issue, although disabling `extra_scripts` section for `at90usb1286_dfu` ENV moves beyond this error (disabling auto-upload functionality)
2) TMC2208 driver compile failure due to `TMC2208Stepper.cpp:202:16: error: 'class SoftwareSerial' has no member named 'stopListening'` but configs do not include the driver

### Steps to Reproduce

1. Install current PIO tool chain
2. Attempt to compile working-as-of March 26 configs
https://github.com/fiveangle/Marlin/blob/bf2_printrbot_simple_metal_config/Marlin/Configuration.h
https://github.com/fiveangle/Marlin/blob/bf2_printrbot_simple_metal_config/Marlin/Configuration_adv.h
3. Observe failure with PIO default environment naming warnings + compile failure

```
Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Can not remove temporary directory `C:\dev\git\Marlin\.pioenvs`. Please remove it manually to avoid build issues
Processing at90usb1286_dfu (platform: teensy; framework: arduino; board: at90usb1286; build_flags: -fmax-errors=5, -g, -ggdb; lib_deps: https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip, LiquidCrystal@1.3.4, TMCStepper@<1.0.0, Adafruit NeoPixel@1.1.3, https://github.com/lincomatic/LiquidTWI2/archive/30aa480.zip, https://github.com/ameyer/Arduino-L6470/archive/master.zip, https://github.com/trinamic/TMC26XStepper/archive/c1921b4.zip; src_filter: +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared> +<src/HAL/HAL_AVR>; extra_scripts: pre:buildroot/share/atom/create_custom_upload_command_DFU.py; monitor_speed: 250000)
-------------------------------------------------------------------------------
Mkdir("C:\dev\git\Marlin\.pioenvs\at90usb1286_dfu")

TypeError: unsupported operand type(s) for +: 'NoneType' and 'str':

  File "C:\python27\lib\site-packages\platformio\builder\main.py", line 123:

    env.SConscript(item, exports="env")

  File "C:\Users\speedster\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:

    return _SConscript(self.fs, *files, **subst_kw)

  File "C:\Users\speedster\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:

    exec _file_ in call_stack[-1].globals

  File "C:\dev\git\Marlin\buildroot\share\atom\create_custom_upload_command_DFU.py", line 26:

    avrdude_conf_path =  env.get("PIOHOME_DIR") + '\\packages\\toolchain-atmelavr\\etc\\avrdude.conf'
```

4. Rebase working-as-of March 26 configs onto current BF2 branch:
https://github.com/fiveangle/Marlin/blob/2019-09-18-pbsm/Marlin/Configuration.h
https://github.com/fiveangle/Marlin/blob/2019-09-18-pbsm/Marlin/Configuration_adv.h

5. Observe warnings resolved, but compile failure persists:
 
```
Processing at90usb1286_dfu (platform: teensy; framework: arduino; board: at90usb1286; build_flags: -fmax-errors=5 -g -D__MARLIN_FIRMWARE__; lib_deps: U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip, LiquidCrystal@1.3.4, TMCStepper@<1.0.0, Adafruit NeoPixel@1.1.3, LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip, Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/dev.zip, SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip, SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip, SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip, TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip; src_filter: +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared> +<src/HAL/HAL_AVR>; extra_scripts: pre:buildroot/share/atom/create_custom_upload_command_DFU.py; monitor_speed: 250000)
-------------------------------------------------------------------------------
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str':

  File "C:\python27\lib\site-packages\platformio\builder\main.py", line 123:

    env.SConscript(item, exports="env")

  File "C:\Users\speedster\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:

    return _SConscript(self.fs, *files, **subst_kw)

  File "C:\Users\speedster\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:

    exec _file_ in call_stack[-1].globals

  File "C:\dev\git\Marlin\buildroot\share\atom\create_custom_upload_command_DFU.py", line 26:

    avrdude_conf_path =  env.get("PIOHOME_DIR") + '\\packages\\toolchain-atmelavr\\etc\\avrdude.conf'

========================= [FAILED] Took 0.45 seconds =========================
```

6. Comment out `extra_scripts` in `at90usb1286_dfu `env in platformio.ini
7. Observe goes beyond toolchain incompatibility with create_custom_upload_command_DFU.py, but fails with downstream Marlin failure:

e\TMC2660Stepper.cpp.o -c -fno-exceptions -felide-constructors -std=gnu++11 -fpermissive -fsingle-precision-constant -fmax-errors=5 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DPLATFORMIO=40100 -DTEENSY2PP -D__MARLIN_FIRMWARE__ -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=147 -DF_CPU=16000000L -DLAYOUT_US_ENGLISH -I.pio\libdeps\at90usb1.pio\libdeps\at90usb1286_dfu\TMCStepper_ID5513\src\source\TMC2208Stepper.cpp:202:16: error: 'class SoftwareSerial' has no member named 'stopListening'

```...
avr-g++ -o .pio\build\at90usb1286_dfu\lib01d\TMCStepper_ID5513\source\TMC2209Stepper.cpp.o -c -fno-exceptions -felide-constructors -std=gnu++11 -fpermissive -fsingle-precision-constant -fmax-errors=5 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DPLATFORMIO=40100 -DTEENSY2PP -D__MARLIN_FIRMWARE__ -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=147 -DF_CPU=16000000L -DLAYOUT_US_ENGLISH -I.pio\libdeps\at90usb1286_dfu\TMCStepper_ID5513\src -IC:\Users\speedster\.platformio\packages\framework-arduinoteensy\libraries\SPI -IC:\Users\speedster\.platformio\packages\framework-arduinoteensy\libraries\SoftwareSerial -IC:\Users\speedster\.platformio\packages\framework-arduinoteensy\cores\teensy -IC:\Users\speedster\.platformio\packages\framework-arduinoteensy\cores .pio\libdeps\at90usb1286_dfu\TMCStepper_ID5513\src\source\TMC2209Stepper.cpp

.pio\libdeps\at90usb1286_dfu\TMCStepper_ID5513\src\source\TMC2208Stepper.cpp: In member function 'virtual uint32_t TMC2208Stepper::read(uint8_t)':

avr-g++ -o .pio\build\at90usb1286_dfu\lib01d\TMCStepper_ID5513\source\TMC2660Stepper.cpp.o -c -fno-exceptions -felide-constructors -std=gnu++11 -fpermissive -fsingle-precision-constant -fmax-errors=5 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DPLATFORMIO=40100 -DTEENSY2PP -D__MARLIN_FIRMWARE__ -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=147 -DF_CPU=16000000L -DLAYOUT_US_ENGLISH -I.pio\libdeps\at90usb1.pio\libdeps\at90usb1286_dfu\TMCStepper_ID5513\src\source\TMC2208Stepper.cpp:202:16: error: 'class SoftwareSerial' has no member named 'stopListening'

286_dfu\TMCSt      SWSerial->stopListening();

epper_ID5                ^

513\src -IC:\Users\speedster\.platformio\packages\framework-arduinoteensy\libraries\SPI -IC:\Users\speedster\.platformio\packages\framework-arduinoteensy\libraries\So*** [.pio\build\at90usb1286_dfu\lib01d\TMCStepper_ID5513\source\TMC2208Stepper.cpp.o] Error 1

ftwareSerial -IC:\Users\speedster\.platformio\packages\framework-ard
```

Thanks,

-=dave

closed 2019-09-25 by thinkyhead

fiveangle, 2019-09-27:
Compile issue still exists. The minor issue with bob’s script is mitigated by disabling, but downstream compile error in TMC driver remains. 

I couldn’t come up with any way to fix. Fear it will take someone like p3p to give insight on what’s going on.

thinkyhead, 2019-10-24:
Seeing similar issues getting TMCStepper to build with the SKR Pro v1.1 board. See #15324

fiveangle, 2019-11-09:
FYI, issue obv still exists. We need to figure out the Trinamic driver issue to unblock all these unrelated platforms. With #15324 closed, is there a master issue that's tracking this ?

drphil3d, 2019-12-26:
Just ran into this issue myself with the AT90USB1286 - Printrboard

sjasonsmith, 2019-12-26:
@fiveangle & @drphil3d, do you build in PlatformIO or in Arduino?

I believe your boards have permanently installed drivers, so there is no need for TMC driver support, right?

If so, you can add the following in PlatformIO environment to eliminate TMCStepper completely. In the Arduino IDE you should just need to ensure it is not installed.

`lib_ignore = TMCStepper`

Of course if that is an adequate solution the change needs to be submitted back in a PR, either by one of you, or myself if you let me know it solves your issue.

drphil3d, 2019-12-26:
That worked!

This is what my environment looked like to start with

#
# AT90USB1286 boards using DFU bootloader
# - PrintrBoard
# - PrintrBoard Rev.F
# - ? 5DPRINT ?
#
[env:at90usb1286_dfu]
platform      = teensy
board         = at90usb1286
lib_deps      = ${common.lib_deps}
  TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_AVR>
monitor_speed = 250000

Modified Environment that worked

#
# AT90USB1286 boards using DFU bootloader
# - PrintrBoard
# - PrintrBoard Rev.F
# - ? 5DPRINT ?
#
[env:at90usb1286_dfu]
platform      = teensy
board         = at90usb1286
lib_ignore = TMCStepper
lib_deps      = ${common.lib_deps}
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_AVR>
monitor_speed = 250000

sjasonsmith, 2019-12-26:
> That worked!

@drphil3d Great! I posted a PR to change the INI file for that environment. I'm sure other environments could benefit from it, but I'm not sure which have boards where TMC drivers might be installed. I guess we'll wait until people complain about build issues on other environments.

drphil3d, 2019-12-26:
More than likely anyone using one of these old boards isn't going to be using tmc drivers. This should also be added to the CDC version.

sjasonsmith, 2019-12-26:
> More than likely anyone using one of these old boards isn't going to be using tmc drivers. This should also be added to the CDC version.

I didn't add it to the CDC version because at least one of those boards has interchangeable drivers.