MarlinFirmware/Marlin #15304: [FR] Linear Deceleration/Accelleration

rflulling, 2019-09-19:
### Description 
Linear Deceleration/Acceleration relative to Circular diameter or continuous segment length.
As apposed to fixed constant speed set by the slicer.

### Steps to Reproduce

I would like a function that can take whatever you have as a top speed, or the speed specified for a move and dynamically reduce the speed at a constant value, perhaps logarithmic. That is if you are running a spiral, the outer rim is largest segment, the speed is 50, that's fine. But as segments gets smaller, the machine shakes and might even trip up. A Linear Reduction in speed, would insure that as the head gets closer to the end of the line, it will also move slower and cleaner, in tighter motions.

Another good example. I make complex mechanical parts. S3D makes very narrow spaces. At 50 the machine is tearing itself apart. I manually reduce to 20%. For larger areas, the 50 works fine. 

I not want this to conflict with other functions designed to apply smaller reductions in speed relative to sudden changes in direction (junction deviation), or those designed to reduce surface noise due to extruders or XY axis deviation. 

#### Additional Information

I will look at adding a video or two of sample places where this function would be helpful. 

I will also look at refining the request as time goes forward. I hope that this behavior is understood and catches the interest of some one that would mind helping implement.

DerAndere1, 2019-09-19:
Does it not help to reduce the value of JUNCTION_DEVIATION_MM or enable S_CURVE_ACCELERATION ? In any case I like the idea. Be aware that this would increase the amount of calculation and may not be doable for 8-bit AVRs. For that reason, similar feature requests were sent to slicer software developers (I know of Slicer3D and Cura). They could implement a feature similar to Makerbot Desktop's "doDynamicSpeed". There is this Python tool which you could use to post-process Slicer3D output: https://github.com/quotetherevan/slowDownFeatures

Grogyan, 2019-09-20:
Slow down on arcs should be on [G2 and G3 moves](https://reprap.org/wiki/G-code#G2_.26_G3:_Controlled_Arc_Move), which are moves that take into account curves without segments.

However, this would first need to generated by the slicer

A firmware cannot naively accept segmented curves as arcs, as not all segmented curves will be arcs, and no way to identify them as such.

However, this would first need to generated by the slicer

A firmware cannot naively accept segmented curves as arcs, as not all segmented curves will be arcs, and no way to identify them as such.

rflulling, 2019-09-20:
*DerAndere:*
I agree completely that this and many other features in Marlin regarding
motion control, or extrusion that are not specific to primary operation of
the hardware, should be part of the slicer. I would love to see more robust
tools in our slicers. I also agree that unless the code were very clever in
how it determine a motion rate change, or know when to ignore the change
need, it probably would require a more robust engine. My current card is a
MKS 1.4. I also have Duet Maestro. The MKS, I have considered replacing to
get 6 driver slots. So the next card may well be 32bit anyway.

May test again to night. Have both junction deviation and S Curve
operational. I think the last time I ran a circular test with them, the
head struggled to move near the center. Fighting itself? Having a stroke?
Not sure but did not look good. Did not report as bug, due to may well be a
settings value issue or conflicting tools.

*Regarding slicers:*

CURA has become very ego eccentric focused on what they think we need,
ignoring issues, and bugs redefining them as feature requests. S3D has done
a similar thing but more or less stopped doing any development at all. I
don't use slicer. No idea why by its always been unstable when I tried to
test it. Got sick of it crashing my computer, and removed it. Tested with
CURA used it to export some code for use on M3D. But found a laundry list
of issues there too. Later controllers from Wanhao and MKS where unstable
in CURA, as they refuse to fix the USB connectivity. S3D has its issues,
but USB is generally rock solid and its maybe the one thing that's gained
stability and connectivity features. That said, being that this is a USB
based printer, S3D is really the only stable tool I trust, even if there
much development to be done to make it suitable for more than tinkering.

*Wishlist:*

At this point I would be more hopeful of a stand alone tool.

   - To tweak a Gcode file for speed reductions.
   - *Would love something to convert segments into arcs too.
   - *Would love some good robust gCode editing software. Could gain many
   super skills with that setup right.

rflulling, 2019-09-20:
Grogyan, 

While I would enjoy using Arcs, this is not something that FDM slicers generate by default.  Even to use it in the Marlin firmware requires a couple switches flipped on. Not that I am against it. Opinion is that a couple arc commands would be cleaner than 100 segment commands. 

That said, yes the system can make some determinations. Infact it already does if using Junction Deviation of S_Curve_Acceleration. However a decent buffer, enough ram and a rock solid look ahead function would yield best results on the fly. 

How can the Firmware KNOW, when this function is needed? Variant 1: Any time threshold for minimum segment is crossed, and there is a 180 return (linear infill), more than 3 consecutive returns, then apply function in logarithmic scale based on speed set for segment and % deviation from threshold. Set speed as manual override until end of section. Variant 2: Any time segments are not linear, and not forming right angles, and delta length is greater or less than itself consecutively for 4 segments or more, aka making arcs.

If I was more a math guru, I imagine the logic is not that complex.

DerAndere1, 2019-09-21:
@rflulling I meant Simplify3D when I said Slicer3D. You can try the tool https://github.com/quotetherevan/slowDownFeatures  for reducing speed.

rflulling, 2019-09-22:
Thanks for that link. I am not really sure how to use the Python script.

DerAndere1, 2019-09-24:
@rflulling I did not test. If in doubt, ask the author of smallFeatures.py. Considering the README I would try  
1. Install Python 3.7. 
2. put a copy of your gcode file (called test.gcode in this example) next to the downloaded smallFeatures.py file. 
3. Open the command interpreter or shell (cmd.exe on Windows). 
4. Change directory to the directory of the downloaded file smallFeatures.py. For MS Windows that would be command `cd` like `cd Path\to\the\tool`. 
5. Execute a command with the following syntax: `python3 smallFeatures.py filename slowDownSpeed LengthThreshold` where you replace `filename` with the name of your gcode file and replace `slowDownSpeed` with an integer (e.g. 300) and `LengthThreshold` with another integer (e.g. 30). A full example would be the following command: `python3 smallFeatures.py test.gcode 300 20` . 
Close the command interpreter / shell. You should find the processed g-code file in the same directory as the input file (e.g. Path\to\the\tool\test_processed.gcode)

rflulling, 2019-09-26:
Thanks DerAndere1, 
I will check this out. It would be great if it offered exactly what I needed in one swipe.

closed 2020-07-06 by thinkyhead