MarlinFirmware/Marlin #15305: UBL build mesh failing at probe point 43 every time with a beep, and just stops

ktgrach, 2019-09-19:
While building a cold mesh either from the display or using G29 P1 from terminal it fails at probe point 43 with a beep and just stops. That's curious in itself as I have speaker turned off in marlin.

I am using a SKR 1.3 board, TMC 2208 in UART mode, One extruder, reprap discount full graphics controller with an induction probe. 

The terminal debug output
giving both point 42 as reference and the odd output at 43 when it just stops.
curious is that at probe point 43 it tries to do a blocking move to the point its already at, then does a move to 111,111,6.75  once printer moves to that point it beeps and ends

<-Probing mesh point 42/100.
<-
<->>> probe_pt(172.11, 147.67, raise, 0, probe_relative)
<-  current_position=(201.11, 173.22, 6.77) : 
<->>> do_blocking_move_to(201.11, 197.67, 6.77)
<-<<< do_blocking_move_to
<-  current_position=(201.11, 197.67, 6.77) : set_probe_deployed
<-deploy: 1
<-  current_position=(201.11, 197.67, 6.77) : >>> run_z_probe
<-  current_position=(201.11, 197.67, 6.77) : >>> do_probe_move
<->>> do_blocking_move_to(201.11, 197.67, -4.00)
<-<<< do_blocking_move_to
<-echo:busy: processing
<-  current_position=(201.11, 197.67, 1.74) : sync_plan_position
<-  current_position=(201.11, 197.67, 1.74) : <<< do_probe_move
<-  current_position=(201.11, 197.67, 1.74) : <<< run_z_probe
<->>> do_blocking_move_to(201.11, 197.67, 6.74)
<-<<< do_blocking_move_to
->N1073 M105*18

<-echo:busy: processing
<-<<< probe_pt
<-
<-Probing mesh point 43/100.
<-
<-  current_position=(201.11, 197.67, 6.74) : set_probe_deployed
<-deploy: 0
<->>> do_blocking_move_to(201.11, 197.67, 6.74)
<-<<< do_blocking_move_to
<->>> do_blocking_move_to(111.00, 111.00, 6.74)
<-<<< do_blocking_move_to
<-X:111.00 Y:111.00 Z:6.74 E:0.00 Count X:11100 Y:11100 Z:2697
<-ok
->N1074 M105*21

<-X:111.00 Y:111.00 Z:6.74 E:0.00 Count X:11100 Y:11100 Z:2697
<-ok

[config files.tar.gz](https://github.com/MarlinFirmware/Marlin/files/3629343/config.files.tar.gz)
[print_log.txt.txt.tar.gz](https://github.com/MarlinFirmware/Marlin/files/3629344/print_log.txt.txt.tar.gz)

Roxy-3D, 2019-09-19:
Can you take a quick video of it?   I looked at the print_log.txt file and I didn't see anything crazy.
It might help to do a M111 S255 to get a little bit more information when you do the video.

ktgrach, 2019-09-19:
Thank you for the assistance.
I did what you asked. Let me know if the video is good enough as it was dark and a slight bit out of focus. Also if its not obvious 111,111 is dead center on bed.

https://drive.google.com/file/d/1-24dqPqg-oYO-KvAleTP4JEAfMUiox2M/view?usp=sharing
[print_log17470919.txt.tar.gz](https://github.com/MarlinFirmware/Marlin/files/3633391/print_log17470919.txt.tar.gz)

Roxy-3D, 2019-09-19:
Are you building the firmware with Arduino?    Can you tell me how much free memory Arduino reports at the end of the firmware build?

ktgrach, 2019-09-19:
using VSC with platformIO can compile with m100 if that helps.
<-echo:N2656 M100*21
<-
<-bss_end               : 0x10003750
<-__heaplimit           : 0x10003F50
<-free_memory_start     : 0x10003F50
<-__stacklimit          : 0x10007400
<-free_memory_end       : 0x10007200
<-MEMORY_END_CORRECTION: 512
<-Stack Pointer         : 0x10007FAF
<-Initializing free memory block.
<-
<-
<-12726 bytes of memory initialized.
<-
<-ok

ktgrach, 2019-09-19:
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  42.8% (used 13928 bytes from 32568 bytes)
PROGRAM: [=====     ]  46.0% (used 218724 bytes from 475136 bytes)

Roxy-3D, 2019-09-19:
Yeah...   I just wanted to make sure you had plenty of memory.   And you most certainly do have memory to spare.

ktgrach, 2019-09-19:
Yes I just tried it with tmc in standalone mode, advanced pause/ filament sensor disabled, SD support disabled TMC debug disabled in an attempt to eliminated those as a culprit and to free as much memory as possible. Same thing in same place

ktgrach, 2019-09-19:
Better video with the above settings
https://drive.google.com/file/d/1-AEveYEn7iwb-HANsTXOVu6SfOom9dFo/view?usp=sharing

Roxy-3D, 2019-09-19:
Can you do a log with the DEBUG_LEVELING set at M111 S255 ?
That will give us much more verbose information on what is happening.

ktgrach, 2019-09-19:
[print_log18570919.txt](https://github.com/MarlinFirmware/Marlin/files/3633494/print_log18570919.txt)

ktgrach, 2019-09-19:
did it will octoprint terminal in case, my terminal software was not recording everything. would the g29 verbose help and what level?

[octoprint terminal.tar.gz](https://github.com/MarlinFirmware/Marlin/files/3633512/octoprint.terminal.tar.gz)

Roxy-3D, 2019-09-19:
This is interesting....
```
<-
<-Probing mesh point 42/100.
<-
<->>> probe_pt(172.11, 147.67, raise, 0, probe_relative)
<-  current_position= X201.11 Y173.22 Z6.76 : 
<->>> do_blocking_move_to X201.11 Y197.67 Z6.76
<-<<< do_blocking_move_to
<-  current_position= X201.11 Y197.67 Z6.76 : set_probe_deployed
<-deploy: 1
<-  current_position= X201.11 Y197.67 Z6.76 : >>> run_z_probe
<-  current_position= X201.11 Y197.67 Z6.76 : >>> do_probe_move
<->>> do_blocking_move_to X201.11 Y197.67 Z-4.00
<-<<< do_blocking_move_to
<-echo:busy: processing
<-  current_position= X201.11 Y197.67 Z1.80 : sync_plan_position
<-  current_position= X201.11 Y197.67 Z1.80 : <<< do_probe_move
<-  current_position= X201.11 Y197.67 Z1.80 : <<< run_z_probe
<->>> do_blocking_move_to X201.11 Y197.67 Z6.80
<-<<< do_blocking_move_to
->N69 M105*24

<-echo:busy: processing
<-<<< probe_pt
<-
<-Probing mesh point 43/100.
<-
<-  current_position= X201.11 Y197.67 Z6.80 : set_probe_deployed
<-deploy: 0
<->>> do_blocking_move_to X201.11 Y197.67 Z6.80
<-<<< do_blocking_move_to
<->>> do_blocking_move_to X111.00 Y111.00 Z6.80
<-<<< do_blocking_move_to
<-X:111.00 Y:111.00 Z:6.80 E:0.00 Count X:11100 Y:11100 Z:2720
<-ok
->N70 M105*16
```

deploy: 1  happens everywhere until the last time when it fails...
For some reason...   It is acting like the probe is not in a deployed state.

ktgrach, 2019-09-19:
marlin is configured as a fixed mount probe.

ktgrach, 2019-09-19:
gave it a G0 X0 Y0 before performing a G29 P1 V4 (tried the verbose option to see if it gave us more info )
Still malfunctioned at 43 but different behavior. AT 43 it hit the endstops of the X and Y 
[print_log19310919.txt.tar.gz](https://github.com/MarlinFirmware/Marlin/files/3633558/print_log19310919.txt.tar.gz)

ktgrach, 2019-09-21:
hmm a const bool should never change, the fact that it does shows something is running amok.
Tried a bunch of different thing always the same at 43 it "fails to deploy" a fixed probe. 
Maybe I can help (no guarantee My C++ is many, many years out of date) its been awhile but I do have over 2 decades of asm and C experience. Is there a Jtag you recommend to use with VSC and Platform IO both didn't exist when I was programming.

Roxy-3D, 2019-09-21:
The debugging is pretty limited on the AVR's.   The debugging is much better on the 32-bit platforms if you go to all the trouble to get it setup and learn how to use the GCC Debugger.

But most likely that doesn't matter.     Adding some SERIAL_ECHO() statements to print out status every couple of lines in the G29 P1 code will help us find where it is going off the rails.   It kind of looks like the probe code thinks the probe is not deployed.    

Anyway...   Probably the thing to do is put the SERIAL_ECHO() statements in the top level G29 P1 code, and find or prove it is the call to the probe deploy that is failing.   And then add some more SERIAL_ECHO() lines in the probe deploy code and figure out where it is going crazy.

ktgrach, 2019-09-21:
Thank you, using your advice I figured it out.  It's not a BUG just me not knowing how the system worked and UGH with no status message telling us what happened we had no idea it was finished probing all the points it could reach. That needs to be adjusted with documentation explaining that and a message on the display and terminal notifying the end user. 
So sorry for wasting your time.

closed 2019-09-21 by ktgrach

cinealfa, 2019-11-09:
What is the behavior you discover mine is doing the same thing had to go back to abl :) can't figure it out used to work flawlessly with the same settings.

Roxy-3D, 2019-11-09:
I believe this is fixed in BugFix v2.0.0     Grab the latest code and I believe the problem will be gone.