MarlinFirmware/Marlin #15306: [BUG] STM32F103R_bigtree_USB not detected

reloxx13, 2019-09-19:
https://github.com/MarlinFirmware/Marlin/commit/6282655d77d2f9811651843a547b61ca43bd61a9

this is failing somehow.
i get this:
[image]

and USB dir is empty. i removed both folders and tried and also cleaned anything.

looks like auto build script is not detecting the right env.

`default_envs = STM32F103R_bigtree_USB`

```
Processing STM32F103R_bigtree (platform: ststm32; framework: arduino; board: genericSTM32F103RC)

Verbose mode can be enabled via `-v, --verbose` option
Mkdir("D:\xampp\htdocs\Marlin\.pio\build\STM32F103R_bigtree")
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.6.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: tool-stm32duino 1.0.1, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.5.0
|-- <LiquidTWI2>

........

STM32F103R              IGNORED
STM32F103R_fysetc       IGNORED
STM32F103R_bigtree      SUCCESS   00:01:11.661
STM32F103R_bigtree_USB  IGNORED
STM32F4                 IGNORED
STM32F7                 IGNORED
ARMED                   IGNORED
```

cinealfa, 2019-09-20:
Im having the same issue. I had to manually put the bin file on the sd card.

anm8tor, 2019-09-22:
What is the intention behind the two different  environments? (bigtree vs bigtree_USB)

reloxx13, 2019-10-04:
mounting sd card over usb (allowing to flash by usb (move fw on sd over usb and trigger a reboot)).

still bugged.

boelle, 2019-10-24:
@reloxx13  it seems you tried to make a PR 20 days ago? am i right?

reloxx13, 2019-10-25:
heya,
nope, i dont have a fix for it and its still bugged.
i made a workaround by removing the non-usb version in my platformio.ini.

boelle, 2019-11-06:
@thinkyhead @shitcreek time to have a look here before the issue gets lost and not fixed

PRSRVFRSHNSS, 2019-11-08:
Same issue. At this point even when I try to put the firmware.bin on the sd and move it over to the printer the LCD is just blank and unresponsive. Now the only firmware I can get to run on it is the one off the Bigtreetech skr mini e3 master off the github. Can't edit anything and update it without the screen going blank.

reloxx13, 2019-11-08:
@PRSRVFRSHNSS this is another issue, try reinstalling pio core.
follow this:
https://github.com/MarlinFirmware/Marlin/issues/15767#issuecomment-549156764

reloxx13, 2019-11-16:
looks fixed with 0f57818f2decca8340fd5fc86f9ea7b792d9d802

closed 2019-11-16 by reloxx13