MarlinFirmware/Marlin #15308: [BUG] Compile error with FILAMENT_RUNOUT_SENSOR enabled

mdelecate, 2019-09-19:
### Description
Compile error when `ADVANCED_PAUSE_FEATURE` disabled and `FILAMENT_RUNOUT_SENSOR` is enabled with a custom `FILAMENT_RUNOUT_SCRIPT` other than the default `M600`. 

### Steps to Reproduce
1. In Configuration.h, enable `FILAMENT_RUNOUT_SENSOR`
2. A few lines below, change `FILAMENT_RUNOUT_SCRIPT` from the default `M600` to another custom command, e.g. `M118 A1 action:filamentrunout`
2. In Configuration_adv.h, confirm that `ADVANCED_PAUSE_FEATURE` is disabled 
3. Verify/Compile

**Expected behavior:** 
Successful compile

**Actual behavior:** 
Error compiling: 
```
In file included from /var/folders/b2/dzd0fn7x7ns16c7lxwkcgj7r0000gr/T/arduino_build_589548/sketch/src/Marlin.cpp:149:0:
/var/folders/b2/dzd0fn7x7ns16c7lxwkcgj7r0000gr/T/arduino_build_589548/sketch/src/feature/runout.h: In static member function 'static void TFilamentMonitor<RESPONSE_T, SENSOR_T>::run()':
/var/folders/b2/dzd0fn7x7ns16c7lxwkcgj7r0000gr/T/arduino_build_589548/sketch/src/feature/runout.h:101:95: error: 'did_pause_print' was not declared in this scope
       if (enabled && !filament_ran_out && (IS_SD_PRINTING() || print_job_timer.isRunning() || did_pause_print)) {
```

#### Additional Information
ADVANCED_PAUSE_FEATURE is currently required for FILAMENT_RUNOUT_SENSOR's default action M600. But if the default M600 action is not in use, then ADVANCED_PAUSE_FEATURE should not be required.

sjasonsmith, 2019-09-19:
Can you attach your config files?

mdelecate, 2019-09-20:
Sure. Configuration_adv.h remains unchanged from the latest bugfix-2.0.x, Configuration.h contains the two edits. 

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3633863/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3633864/Configuration.h.txt)

sjasonsmith, 2019-09-20:
Can you try the latest state of bugfix-2.0.x? My change just went in which should fix this for you.

mdelecate, 2019-09-20:
Excellent, just tested it and it works! Thanks a lot for the quick fix!

closed 2019-09-20 by mdelecate