MarlinFirmware/Marlin #15311: Adding a 3rd Tool on a Dual X Carriage Printer

audeophilic, 2019-09-19:
Hello, I have a peculiar challenge. I have a Makergear M3 ID running an older version of marlin. I have modified the firmware to include a 3rd extruder and remapped the second X Carriage to the next set of extruder pins, or E3_(function)_PIN in pins_RAMBO.h.

I am, for lack of a better analogy, trying to upgrade the printer with what is essentially an E3D cyclops. So ideally, on the left carriage, I will have E0 and its heater/thermistor, and on the right I will have E1 and its heater/thermistor. I would like to incorporate my 'Cyclops' on the left carriage.

So everything works, except when I send Gcode commands to change between T0 and T1 and T2, when I try to make the X carriage move with G1 X(distance) it moves the carriage for whatever tool I am using. So the following happens:
`
T0 selected:
    -T0 temperature reading from M105 for left hotend
    -G1 X(positive distance) leads to the left carriage moving to the right
    -G1 E(distance) leads to the left extruder moving

T1 selected:
    -T1 temperature reading from M105 for right hotend
    -G1 X(positive distance) leads to the right carriage moving to the left
    -G1 E(distance) leads to the right extruder moving

T2 selected (this is the problem child)
    -T2 temperature reading from M105 for left hotend (same as T0) // This is correct!!
    -G1 X(positive distance) leads to the right carriage moving to the left (same as T1) // THIS IS INCORRECT!!!
    -G1 E(distance) leads to the experimental third extruder moving // This works as anticipated.
`
I want to, when T2 is selected, move the left carriage where the 3rd tool will be mounted, but I am unsure about how to go about this.

TLDR: Is there a way in Marlin to assign different tools to different X axis carriages?

Roxy-3D, 2019-09-19:
> TLDR: Is there a way in Marlin to assign different tools to different X axis carriages?

Too Long.  Didn't Read ???????     I'm hoping that was a summary you were giving for people that didn't want to read the whole thing.   

The short answer is "No."    For example, in the standard 2 Tool IDEX (Independent Dual Extruder) configurations,  T0 is the Left Extruder.   And T1 is the Right extruder.     But it certainly can be added.

We have configurations to park a tool, and pick up a new tool.   If that feature set was combined with the IDEX configurations...    You may have what you want.

audeophilic, 2019-09-19:
Oh, no. It's a makergear M3 ID. It is a typical IDEX machine with two carriages. Each, stock, has its own single extruder. There is one X carriage, and another that the firmware designates as X2. Think of it in terms of this: I'm trying to stick something like an E3d Cyclops (2 materials, one hotend/thermistor) on the left X carriage. So 3 tools, but 2 x carriages.

So it links T0 to X, and T1/T2 to X2. I would like to have two extruders on X and one on X2 though. Am I making sense or does this sound like pure garbage?

InsanityAutomation, 2019-09-19:
This is actually a duplicate FR, this was requested before and kindve overlaps with some planned changes we have for the future (mixing on idex) but has not been done yet. Prior was 4 extruders with a chimera on each side.

InsanityAutomation, 2019-09-19:
Also may be worth noting im familiar with the machine and some of the odd electrical choices blocking Graphical LCD pins. They use Octoprint as the exclusive display. Ive met Sam there at Rapid and IMTS and have seen a Mosquito hotend changeover on one from Slice Eng but I dont own one.

audeophilic, 2019-09-19:
That's peculiar. I was completely oblivious to this other request on posting. That's basically what I want to do, except with just one double-extruder on one side. It is for a research project and is quite the long story.

I actually have my 3rd extruder set up with pins from the EXT2 section that would normally be used on the LCD display. I have been using the built in Octoprint to communicate with it to test my implementation of the motor via Send: commands, and can confirm it moves as expected. But that is besides the point. I suppose I could install it on the second X carriage, but that would be less than ideal for a few reasons I don't care to get into.

InsanityAutomation, 2019-09-19:
Well, at least for now no its not supported. Neither of us who have done the recent work on IDEX have a machine with enough drivers (or pins to add them) to perform the task thus no way to debug. If either of us get some hardware, itll likely happen quickly. Weve been working with a board designer on a 12 stepstick board and when its done we will finally have a way to do what we want (3 on each side, so youll be covered!). Baring someone suddenly coming up to donate hardware before that though itll be on the lower half of the list for another month or two.

InsanityAutomation, 2019-09-19:
See #13897

Roxy-3D, 2019-09-19:
> I'm trying to stick something like an E3d Cyclops (2 materials, one hotend/thermistor) on the left X carriage. So 3 tools, but 2 x carriages.   So it links T0 to X, and T1/T2 to X2. 

T0 is typically the Left Extruder on an IDEX machine.    So you have conflicting information when you say you want T1/T2 on the second X carriage.

InsanityAutomation, 2019-09-19:
@Roxy-3D true, but for concept its irrelevant how theyre organized. We would likely end up with a extruder gantry assignment then a toggle for single nozzle or not. If not we would have some fun determining all the offsets, so mixing (cyclops) will be first which suits this thread then independent (chimera ect) after which will suit the older one.

Roxy-3D, 2019-09-19:
> Neither of us who have done the recent work on IDEX have a machine with enough drivers (or pins to add them) to perform the task thus no way to debug. If either of us get some hardware, itll likely [...]

What InsanityAutomation is almost saying, but didn't quit get there....

We want to evolve the Marlin Firmware to handle a Color Mixing extruder on both sides of an IDEX machine.   In our mindset, you would not end up with a T0, T1 and a T2.    You would have just a T0 and a T1.    But there would be a color pallet for each extruder and in your case, one of your two 
 extruders could mix two different colors (if you chose to do that).

InsanityAutomation, 2019-09-19:
@Roxy-3D I would think depending on mode we could have T0-T5 then the mixing mode would combine them to T0 and T1 virtual tools with mix ratios. Thats all up in the air how we implement until we do though.

Roxy-3D, 2019-09-19:
Yeah...   We can debate the actual implementation.    Right now I'm leaning towards the the simplest and most straight forward way to implement it.   And that is each of the tool numbers (T0, T1 and T2 on the Left extruder and T3, T4 and T5 on the Right extruder) do 100% their color if selected.   In other words...  If you select T1, the Left Carriage is active and the 2nd stepper on it contributes 100% of the requested filament.

But then there is an imaginary tool for both the Left and Right carraige.   If T6 or T7 is selected, the specified color pallet (or color mixing of the available filament on that carriage) happens.

That would keep things simple for people.   They would only fall into the more complicated case if they enable the imaginary tools T6 or T7.

audeophilic, 2019-09-19:
> T0 is typically the Left Extruder on an IDEX machine. So you have conflicting information when you say you want T1/T2 on the second X carriage.

That's not what I'm saying. I'm saying that right now, T0 is on the left extruder and T1/T2 are on the right extruder. I'm saying that I want T0/T2 on the left extruder, and T1 on the right. You're not understanding me because of my lack of clarity, my apologies.

InsanityAutomation, 2019-09-19:
Or whatever tool numbers they assign to the mix / gradient up to whatever anyway. Thatll limit confusion and allow easier switching in and out of mix vs individual modes than my initial thought anyway.

@audeophilic aside from some semantics that will be configurable we at least understand the requirement and as you may have guessed by now we have a pretty good mental map of it.... Just no viable hardware in hand! lol

audeophilic, 2019-09-19:
> Yeah... We can debate the actual implementation. Right now I'm leaning towards the simplest and most straight forward way to implement it. And that is each of the tool numbers (T0, T1 and T2 on the Left extruder and T3, T4 and T5 on the Right extruder) do 100% their color if selected. In other words... If you select T1, the Left Carriage is active and the 2nd stepper on it contributes 100% of the requested filament.

@InsanityAutomation I personally think that, much like you can assign which pins are in use for what in pins.h, you should be able to assign which tools are on which carriage in IDEX printers. If you could help me gain insight into how the firmware assigns the tools to individual carriages, I can try to implement it myself, I just don't know where to start. I've already had to modify the heck out of a few configuration files to get as far as I have. There is a cascade of #if EXTRUDERS > x statements in pins.h that dynamically assign the pins of extruder steppers if there are a certain number of extruders, so maybe could there be something like that where we could assign some variables that store the tools per axis and use them later? Hypothetical example:

`//(we would make a series of conditionals to allow the assignment of only as many tools as there are extruders)'

'#ifdef DUAL_X_CARRIAGE'

'#if EXTRUDERS > x'

'//Sets which tools are active on X axis '

'#define X_LEFT_TOOLS T0 T1 T2'

'//Sets which tools are active on X2 axis'

'#define X2_RIGHT_TOOLS T3, T4, T5`

or something? There would need to be a foundation for some of these statements but I am willing.

InsanityAutomation, 2019-09-19:
dual x carriage drastically changes alot of that, and you need a way for active tool to be different than carriage number. There are underlying changed required before that will function just like that. Quite a bit needs to be done in tool_change.cpp and stepper_indirection.cpp as well as the assignment logic.

Its just not that simple.

Roxy-3D, 2019-09-19:
@audeophilic    You can find the pin assignments for IDEX in pins.h:

```
/**
 * Auto-Assignment for Dual X, Dual Y, Multi-Z Steppers
 *
 * By default X2 is assigned to the next open E plug
 * on the board, then in order, Y2, Z2, Z3. These can be
 * overridden in Configuration.h or Configuration_adv.h.
 */

#define __EPIN(p,q) E##p##_##q##_PIN
#define _EPIN(p,q) __EPIN(p,q)

// The X2 axis, if any, should be the next open extruder port
#if EITHER(DUAL_X_CARRIAGE, X_DUAL_STEPPER_DRIVERS)
  #ifndef X2_STEP_PIN
    #define X2_STEP_PIN   _EPIN(E_STEPPERS, STEP)
    #define X2_DIR_PIN    _EPIN(E_STEPPERS, DIR)
    #define X2_ENABLE_PIN _EPIN(E_STEPPERS, ENABLE)
    #if E_STEPPERS >= MAX_EXTRUDERS || !PIN_EXISTS(X2_STEP)
      #error "No E stepper plug left for X2!"
    #endif
  #endif
  #ifndef X2_CS_PIN
    #define X2_CS_PIN     _EPIN(E_STEPPERS, CS)
  #endif
  #ifndef X2_MS1_PIN
    #define X2_MS1_PIN    _EPIN(E_STEPPERS, MS1)
  #endif
  #ifndef X2_MS2_PIN
    #define X2_MS2_PIN    _EPIN(E_STEPPERS, MS2)
  #endif
  #ifndef X2_MS3_PIN
    #define X2_MS3_PIN    _EPIN(E_STEPPERS, MS3)
  #endif
```

Feel free the change what ever you want.    But understand it is very likely we deviate from your view of the world when we go to implement things....

audeophilic, 2019-09-20:
@Roxy-3D and that is fine. You fail to understand that I already am past this point, and this does not answer my question.

audeophilic, 2019-09-20:
@InsanityAutomation I understand that. That is the point, and thank you for directing me to those files, this is what I've been getting at.

audeophilic, 2019-09-20:
@Roxy-3D 

> But understand it is very likely we deviate from your view of the world

I'm not entirely sure what you're getting at. It's great that we may have our differences, that is irrelevant though. I'm asking for advice, I am not looking for you to do it for me. I'm looking to do it myself and need to know where to start.

InsanityAutomation, 2019-09-20:
@audeophilic she meant dont be offended if we end up changing whatever you do drastically or going a totally different direction as we are trying to handle every use case instead of one particular one. May be more open dev time to point through things after ERRF and 2.0 hits RC1, but for the moment everyone is quite busy.

audeophilic, 2019-09-20:
@InsanityAutomation  I wouldn't be offended. I understand the intricacies of the development process, and I understand needing to account for the common and edge cases in a solution. I'm just hoping to learn from this experience and get this printer working as the specifications of my research project require.